projectacrn/acrn-hypervisor #4504: doc: broken link to AGL release image

dbkinder, 2020-03-18:
Our AGL-VMS tutorial (https://projectacrn.github.io/latest/tutorials/agl-vms.html) references an old AGL release image version 6.0.2 that's no longer available to download. 

Perhaps this 6.05 version will still work, but we need verification from ACRN engineering/QA: https://mirrors.edge.kernel.org/AGL/release/flounder/6.0.5/intel-corei7-64/deploy/images/intel-corei7-64/agl-demo-platform-crosssdk-intel-corei7-64-20200304031825.rootfs.wic.xz

fuzhongl, 2020-05-04:
@NanlinXie With Limited resource; Do we still need to maintain this kinds of Doc?
Thanks!

dbkinder, 2020-05-05:
@NanlinXie Should we drop this Automotive Grade Linux (AGL) documentation for ACRN?

gvancuts, 2020-05-05:
Let's drop it for now. If someone gets around to validating this again with a newer version, we can always resurrect it at that point.

gvancuts, 2020-05-05:
/cc @kiendinh

kiendinh, 2020-05-06:
Somehow the AGL folks has replaced the old 6.0.2 release image with the newer one created on March 2020. The new link for the "hopefully the same" image is:

https://mirrors.edge.kernel.org/AGL/release/flounder/6.0.2/intel-corei7-64/deploy/images/intel-corei7-64/agl-demo-platform-crosssdk-intel-corei7-64-20200318141526.rootfs.wic.xz

dbkinder, 2020-06-19:
Fixed in #4973

closed 2020-06-19 by dbkinder