projectacrn/acrn-hypervisor #4553: Unable to launch sample Zephyr application

christophert, 2020-03-30:
I was following the tutorial on deploying Zephyr as a UOS on ACRN but have been unable to get it to boot. I've compiled Zephyr on both Ubuntu 18.04 and Clear Linux without success.

My NUC7i7DNHE NUC is running BIOS Version DNKBLi7v.86A.0067.2019.1101.1743 and is running Clear Linux 32650. I've also tested this with 31990 without success.
```
sudo ./launch_zephyr.sh
Password:
cpu1 online=1
cpu2 online=1
cpu3 online=1
SW_LOAD: get ovmf path /usr/share/acrn/bios/OVMF.fd, size 0x200000
vm_create: zephyr_vm1
VHM api version 1.0
vm_setup_memory: size=0x8000000
open hugetlbfs file /run/hugepage/acrn/huge_lv1/zephyr_vm1/D279543825D611E8864ECB7A18B34643
open hugetlbfs file /run/hugepage/acrn/huge_lv2/zephyr_vm1/D279543825D611E8864ECB7A18B34643
level 0 free/need pages:0/65 page size:0x200000
level 1 free/need pages:0/0 page size:0x40000000
to reserve more free pages:
to reserve pages (+orig 0): echo 65 > /sys/kernel/mm/hugepages/hugepages-2048kB/nr_hugepages
now enough free pages are reserved!

try to setup hugepage with:
        level 0 - lowmem 0x8000000, biosmem 0x200000, highmem 0x0
        level 1 - lowmem 0x0, biosmem 0x0, highmem 0x0
total_size 0x140200000

mmap ptr 0x0x7f0846ff1000 -> baseaddr 0x0x7f0847000000
mmap 0x8000000@0x7f0847000000
touch 64 pages with pagesz 0x200000
mmap 0x200000@0x7f0946e00000
touch 1 pages with pagesz 0x200000

really setup hugepage with:
        level 0 - lowmem 0x8000000, biosmem 0x200000, highmem 0x0
        level 1 - lowmem 0x0, biosmem 0x0, highmem 0x0
vm_init_vdevs
No correct pm notify channel given
pci init hostbridge
pci init lpc
pci init virtio-blk
pci init virtio-console
***********************************************
                                               virt-console backend redirected to /dev/pts/3
                                                                                            ***********************************************
                                           tpm: init_vtpm2:Invalid socket path!
                                                                               polling 34...
                                                                                            Listening 34...
           /tmp/dm.Xw34cpo     24:     Device (PCI0)
                                                    Warning  3073 -        Multiple types ^  (Device object requires either a _HID or _ADR, but not both)

                                                         /tmp/dm.Xw34cpo    785:     Processor (CPU0, 0x00, 0x00000000, 0x00) {}
                                Warning  3168 -                          ^ Legacy Processor() keyword detected. Use Device() keyword instead.

                                             acrn_sw_load
                                                         SW_LOAD: partition blob /usr/share/acrn/bios/OVMF.fd size 2097152 copy to guest 0xffe00000
                                                   SW_LOAD: build e820 9 entries to addr: 0x7f08470ef008
SW_LOAD: entry[0]: addr 0x0000000000000000, size 0x00000000000a0000,  type 0x1
SW_LOAD: entry[1]: addr 0x00000000000a0000, size 0x0000000000060000,  type 0x2
SW_LOAD: entry[2]: addr 0x0000000000100000, size 0x0000000007f00000,  type 0x1
SW_LOAD: entry[3]: addr 0x0000000008000000, size 0x0000000078000000,  type 0x2
SW_LOAD: entry[4]: addr 0x0000000080000000, size 0x0000000008000000,  type 0x2
SW_LOAD: entry[5]: addr 0x00000000db000000, size 0x0000000004000000,  type 0x2
SW_LOAD: entry[6]: addr 0x00000000df000000, size 0x0000000001000000,  type 0x2
SW_LOAD: entry[7]: addr 0x00000000e0000000, size 0x0000000020000000,  type 0x2
SW_LOAD: entry[8]: addr 0x0000000100000000, size 0x0000000040000000,  type 0x2
SW_LOAD: ovmf_entry 0xfffffff0
                              add_cpu
                                     out instr on NMI port (0x61) not supported
                                                                               out instr on NMI port (0x61) not supported
                         out instr on NMI port (0x61) not supported
```

gvancuts, 2020-03-31:
It looks like there might be a mismatch between your version of ACRN and `/usr/sbin/iasl`. Can you report which versions are installed and also whether you rebuilt/installed a different `acrn.efi` manually?

christophert, 2020-03-31:
I have just reinstalled the entire platform to ensure there weren't any incompatibilities. I am able to launch the Preempt RT VM with no issues.

My versions are the following:
Clear Linux 32030

```
$ iasl -v

Intel ACPI Component Architecture
ASL+ Optimizing Compiler/Disassembler version 20191213
Copyright (c) 2000 - 2019 Intel Corporation

$ acrn-dm -v
DM version is: 1.5-unstable-”2020w01.1.140000p_260” (daily tag:”2020w01.1.140000p”), build by mockbuild@2020-01-05 11:33:12
```

fuzhongl, 2020-04-08:
@christophert  
You are using **INDUSTRY** scenario, right?
If yes, please run **vm_console 1**(suppose Zephyr UOS is VM1) in HV console; then you will see:
**Hello World! acrn**

Following is the full log:
ACRN:\>vm_list

VM_UUID                          VM_ID VM_NAME                          VM_STATE
================================ ===== ================================ ========
dbbbd4347a574216a12c2201f1ab0240   0   ACRN SOS VM                      Running
d279543825d611e8864ecb7a18b34643   1   ACRN VM_1                        Running
ACRN:\>vm_console 1

----- Entering VM 1 Shell -----

Welcome to GRUB!

                           GNU GRUB  version 2.04~rc1

 /----------------------------------------------------------------------------\
 |*Zephyr Kernel                                                              |
 |                                                                            |
 |                                                                            |
 |                                                                            |
 |                                                                            |
 |                                                                            |
 |                                                                            |
 |                                                                            |
 |                                                                            |
 |                                                                            |
 |                                                                            |
 |                                                                            |
 |                                                                            |
 \----------------------------------------------------------------------------/

      Use the ^ and v keys to select which entry is highlighted.
      Press enter to boot the selected OS, `e' to edit the commands
      before booting or `c' for a command-line.
   The highlighted entry will be executed automatically in 0s.
  Booting `Zephyr Kernel'

WARNING: no console will be available to OS
error: no suitable video mode found.
***** Booting Zephyr OS build zephyr-v2.0.0-593-g85c1d90fd6e5 *****
Hello World! acrn

christophert, 2020-04-20:
@fuzhongl I have redeployed the NUC with the newest supported version, 32680 industry scenario and it still does not work. How do I access the hypervisor console? I only have access to the SOS.

fuzhongl, 2020-04-21:
@christophert  Do you connect  serial port cable on your NUC7i7DNHE NUC? 
Serial Port via Internal Header is needed.

christophert, 2020-04-22:
@fuzhongl It is not noted in the tutorial for running Zephyr that this cable is required to see the UOS but we will attempt to obtain a serial port cable for the NUC. Should the documentation be updated to reflect this requirement?

fuzhongl, 2020-04-23:
@christophert  The documentation is base on SDC scenario; with this scenario, you can see the log at the current console:
https://projectacrn.github.io/latest/tutorials/using_zephyr_as_uos.html
But you are using INDUSTRY scenario, Could you please try SDC scenario before you get serial cable?
Thanks!

fuzhongl, 2020-05-04:
@christophert Can this git issue be closed now?
Thanks!

gvancuts, 2020-05-06:
@fuzhongl , I think we should update our documentation to reflect the fact that a serial port connection is needed. Maybe @dbkinder or @deb-intel can help with that?

gvancuts, 2020-05-06:
Just proposed #4767 to address the shortcoming in the documentation.

christophert, 2020-05-06:
@gvancuts Would this change also need to be made for documentation for #4554? @fuzhongl indicated I needed the serial cable for Celadon as well but that documentation also doesn't mention the requirement.

fuzhongl, 2020-05-07:
@christophert @gvancuts Currently, it depends on the scenario you used; the HV console isn't needed for SDC scenario.
Within v1.6.1, the dynamic parameters will be enabled for post- launched VMs, you can see the log of post-launched VMs(non-RTVM) without HV console.

gvancuts, 2020-05-07:
> @gvancuts Would this change also need to be made for documentation for #4554? @fuzhongl indicated I needed the serial cable for Celadon as well but that documentation also doesn't mention the requirement.

Would this help? https://github.com/projectacrn/acrn-hypervisor/pull/4774

closed 2020-06-28 by fuzhongl