projectacrn/acrn-hypervisor #4554: No shell console appears when launching Celadon UOS

christophert, 2020-03-30:
I am unable to get a shell console when launching the Celadon UOS using the guide. I am able to see Celadon running on the second monitor but I do not get a shell prompt.

This is the console output I receive:
```
cpu1 online=0
cpu2 online=0
cpu3 online=0
passed gvt-g optargs low_gm 64, high_gm 448, fence 8
SW_LOAD: get ovmf path /usr/share/acrn/bios/OVMF.fd, size 0x200000
vm_create: win_vm1
VHM api version 1.0
vm_setup_memory: size=0x100000000
open hugetlbfs file /run/hugepage/acrn/huge_lv1/win_vm1/D279543825D611E8864ECB7A18B34643
open hugetlbfs file /run/hugepage/acrn/huge_lv2/win_vm1/D279543825D611E8864ECB7A18B34643
level 0 free/need pages:2049/1 page size:0x200000
level 1 free/need pages:0/4 page size:0x40000000
to reserve more free pages:
to reserve pages (+orig 0): echo 4 > /sys/kernel/mm/hugepages/hugepages-1048576kB/nr_hugepages
now enough free pages are reserved!

try to setup hugepage with:
        level 0 - lowmem 0x0, biosmem 0x200000, highmem 0x0
        level 1 - lowmem 0x80000000, biosmem 0x0, highmem 0x80000000
total_size 0x200000000

mmap ptr 0x0x7f621ba83000 -> baseaddr 0x0x7f6240000000
mmap 0x80000000@0x7f6240000000
touch 1024 pages with pagesz 0x200000
mmap 0x80000000@0x7f6380000000
touch 1024 pages with pagesz 0x200000
mmap 0x200000@0x7f633fe00000
touch 1 pages with pagesz 0x200000

really setup hugepage with:
        level 0 - lowmem 0x80000000, biosmem 0x200000, highmem 0x80000000
        level 1 - lowmem 0x0, biosmem 0x0, highmem 0x0
vm_init_vdevs
No correct pm notify channel given
pci init hostbridge
pci init lpc
pci init pci-gvt
polling 34...
             Listening 34...
                            pci init virtio-blk
pci init virtio-net
tpm: init_vtpm2:Invalid socket path!
                                    /tmp/dm.XGAwmtn     24:     Device (PCI0)
                                                                             Warning  3073 -        Multiple types ^  (Device object requires either a _HID or _ADR, but not both)

                                                                                                                                                                                  /tmp/dm.XGAwmtn
    785:     Processor (CPU0, 0x00, 0x00000000, 0x00) {}
                                                        Warning  3168 -                          ^ Legacy Processor() keyword detected. Use Device() keyword instead.

                                                                                                                                                                     acrn_sw_load
                                                                                                                                                                                 SW_LOAD: partition blob /usr/share/acrn/bios/OVMF.fd size 2097152 copy to guest 0xffe00000
                                                                          SW_LOAD: build e820 9 entries to addr: 0x7f62400ef008
SW_LOAD: entry[0]: addr 0x0000000000000000, size 0x00000000000a0000,  type 0x1
SW_LOAD: entry[1]: addr 0x00000000000a0000, size 0x0000000000060000,  type 0x2
SW_LOAD: entry[2]: addr 0x0000000000100000, size 0x000000007ff00000,  type 0x1
SW_LOAD: entry[3]: addr 0x0000000080000000, size 0x0000000008000000,  type 0x2
SW_LOAD: entry[4]: addr 0x00000000db000000, size 0x0000000004000000,  type 0x2
SW_LOAD: entry[5]: addr 0x00000000df000000, size 0x0000000001000000,  type 0x2
SW_LOAD: entry[6]: addr 0x00000000e0000000, size 0x0000000020000000,  type 0x2
SW_LOAD: entry[7]: addr 0x0000000100000000, size 0x0000000040000000,  type 0x2
SW_LOAD: entry[8]: addr 0x0000000140000000, size 0x0000000080000000,  type 0x1
SW_LOAD: ovmf_entry 0xfffffff0
                              add_cpu
```

I am running Clear Linux 32650 SOS on a NUC7i7DNHE.

gvancuts, 2020-03-31:
Hi @christophert , is the guide you have followed [this guide](https://projectacrn.github.io/latest/tutorials/using_celadon_as_uos.html)?

Can you confirm the following versions:
* `acrn-dm`
* `/usr/sbin iasl`

Are you able to run a standard Linux-based User VM on the machine?

/cc @ttzeng

christophert, 2020-03-31:
@gvancuts That is the guide I followed.

I am now running Clear Linux 32030, the verified version on the documentation. 
```
$ iasl -v

Intel ACPI Component Architecture
ASL+ Optimizing Compiler/Disassembler version 20191213
Copyright (c) 2000 - 2019 Intel Corporation

$ acrn-dm -v
DM version is: 1.5-unstable-”2020w01.1.140000p_260” (daily tag:”2020w01.1.140000p”), build by mockbuild@2020-01-05 11:33:12
```
I am able to run the Preempt RT VM without issue.

fuzhongl, 2020-04-07:
@christophert 
Could you please assign USB keyboard/mouse to Celadon UOS in launch script?
[Examples:](https://projectacrn.github.io/latest/user-guides/acrn-dm-parameters.html)
-s 7,xhci,1-2,2-2
This configuration means the virtual xHCI will appear in PCI slot 7 in UOS. Any physical USB device attached on 1-2 (bus 1, port 2) or 2-2 (bus 2, port 2) will be detected by UOS and be used as expected. To determine which bus and port a USB device is attached, you could run lsusb -t in SOS.

Then, run **cat /proc/cmdline**  in Celadon UOS.
Please also share your launch script.
Thanks!

christophert, 2020-04-20:
@fuzhongl 
When I attach a keyboard/mouse I can utilize the android interface but there is no console for me to run that command in. The launch script follows the tutorial of running Celadon on ACRN (copying launch_win.sh and modifying it to load celadon)

```bash
#!/bin/bash
# Copyright (C) 2019 Intel Corporation.
# SPDX-License-Identifier: BSD-3-Clause

function launch_win()
{
vm_name=celadon_vm$1

#check if the vm is running or not
vm_ps=$(pgrep -a -f acrn-dm)
result=$(echo $vm_ps | grep "${vm_name}")
if [[ "$result" != "" ]]; then
  echo "$vm_name is running, can't create twice!"
  exit
fi

#for memsize setting
mem_size=4096M

acrn-dm -A -m $mem_size -s 0:0,hostbridge -s 1:0,lpc -l com1,stdio \
  -s 2,pci-gvt -G "$2" \
  -s 3,virtio-blk,./android.img \
  -s 4,virtio-net,tap0 \
  -s 7,xhci,1-2,2-2 \
  --ovmf /usr/share/acrn/bios/OVMF.fd \
  --windows \
  $vm_name
}

# offline SOS CPUs except BSP before launch UOS
for i in `ls -d /sys/devices/system/cpu/cpu[1-99]`; do
        online=`cat $i/online`
        idx=`echo $i | tr -cd "[1-99]"`
        echo cpu$idx online=$online
        if [ "$online" = "1" ]; then
                echo 0 > $i/online
                # during boot time, cpu hotplug may be disabled by pci_device_probe during a pci module insmod
                while [ "$online" = "1" ]; do
                        sleep 1
                        echo 0 > $i/online
                        online=`cat $i/online`
                done
                echo $idx > /sys/class/vhm/acrn_vhm/offline_cpu
        fi
done

launch_win 1 "64 448 8"
```

fuzhongl, 2020-04-21:
@christophert 
Do you connect serial port cable on your NUC7i7DNHE NUC?
Serial Port via Internal Header is needed. Please check following link for detail:
https://projectacrn.github.io/latest/getting-started/rt_industry.html?highlight=use%20serial%20port%20kbl%20nuc#troubleshooting

christophert, 2020-04-22:
@fuzhongl We do not. It is not stated in the prerequisites or the instructions that it is required to see the terminal. The celadon step from the docs indicate:
```
3. You will see the shell console from the terminal...
```

Is the step incorrect in informing me that I should see a console in the terminal and that a serial connection is required?

fuzhongl, 2020-04-23:
@christophert 
Following documentation is base on SDC scenario, could you please try it with sdc scenario?
https://projectacrn.github.io/latest/tutorials/using_celadon_as_uos.html
If you do need INDUSTRY scenario, please connect serial cable for Hypervisor console.
Thanks!

fuzhongl, 2020-05-04:
@christophert Can this git issue be closed?
Thanks!

christophert, 2020-05-06:
I'm still awaiting the arrival of my serial adapter, I will update when I get it.

fuzhongl, 2020-05-07:
@christophert If you want to use INDUSTRY scenario before the arrival of  serial adapter; please make following change: 
With these change, the log of post-launched VMs isn't redirection to HV console any more.

diff --git a/hypervisor/scenarios/industry/vm_configurations.c b/hypervisor/scenarios/industry/vm_configurations.c
index 1f2cbaad..62963ce5 100644
--- a/hypervisor/scenarios/industry/vm_configurations.c
+++ b/hypervisor/scenarios/industry/vm_configurations.c
@@ -42,8 +42,7 @@ struct acrn_vm_config vm_configs[CONFIG_MAX_VM_NUM] = {
                .cpu_affinity_bitmap = VM1_CONFIG_CPU_AFFINITY,
                .vuart[0] = {
                        .type = VUART_LEGACY_PIO,
-                       .addr.port_base = COM1_BASE,
-                       .irq = COM1_IRQ,
+                       .addr.port_base = INVALID_COM_BASE,
                },
                .vuart[1] = {
                        .type = VUART_LEGACY_PIO,

closed 2020-06-28 by fuzhongl