projectacrn/acrn-hypervisor #4558: [WHL][Performance][WaaG] WaaG reboot automatically during run 3D directX12 with 7212 gfx driver

fuzhongl, 2020-03-31:
1.Environment
[Board]:WHLNUC i7
SOS +Win + RTVM
Installed version: 32610
Version URL: [http://linux-ftp.sh.intel.com/pub/mirrors/clearlinux/update]
Content URL: [http://linux-ftp.sh.intel.com/pub/mirrors/clearlinux/update]

SOS info:

root@nvme09 ~ # uname -a
Linux nvme09 4.19.106-PKT-200203T060100Z #1 SMP PREEMPT Tue Mar 17 02:35:16 UTC 2020 x86_64 GNU/Linux
root@nvme09 ~ # acrn-dm -v
DM version is: 1.6-unstable-a6c50702 (daily tag:acrn-2020w12.1-140000p), build by root@2020-03-17 02:37:53

Release Image:
Industry Scenario: SOS(clear linux) + WaaG(Win10) + RTVM(linux+ CAT patches)

CPU Allocation: 1 for SOS, 1 for WaaG, 2 for RTVM (offline the 2 CPUs)

Memory Allocation: 2G for SoS, 4G for WaaG, 2G for RTVM

Disk Allocation: 120G SATA SSD for SOS/WaaG Image, 128G M.2 SSD for WaaG I/O Partition, 256G NVMe for RT

Summary:

WaaG reboot automatically during run 3D directX12

 

Steps:

launch SOS+ RTVM+WaaG
Copy Passmark9.0 workload to waag desktop
Launch passmark9.0 and click "test-> 3D Graphics-> directX9"
Expect result:

3D directX9 should run normally

 

Actual result:

WaaG reboot automatically during run 3D directX12

 

Rate: 5/5

fuzhongl, 2020-03-31:
[External_System_ID] ACRN-6156

xinyunliu, 2020-04-09:
Need the same fix as https://github.com/projectacrn/acrn-hypervisor/issues/4558

closed 2020-06-28 by fuzhongl