projectacrn/acrn-hypervisor #4560: [WHL][SIT][HV]build HV fail in docker

Leo2Chang, 2020-03-31:
1.Environment
 build docker env follow wiki: https://projectacrn.github.io/latest/tutorials/building_acrn_in_docker.html#build-the-docker-image-from-dockerfile
 gcc version 9.3.1 20200325 releases/gcc-9.3.0-55-gdff885cdc0 (Clear Linux OS for Intel Architecture)
 docker CL version: 32700

 acrn-hypervisor branch: release_1.6
 HV commit: 0aa2c237d52da7243d9c7158a8c5600b9b98bec7
 board and scenario: BOARD=whl-ipc-i5 SCENARIO=industry

 build ACRN HV with: docker run -u`id -u`:`id -g` --rm -v $PWD:/workspace clearlinux-acrn-builder:latest bash -c "make BOARD=whl-ipc-i5 SCENARIO=industry FIRMWARE=uefi RELEASE=0"

2. Reproduce Steps
 build docker compile env with link: https://projectacrn.github.io/latest/tutorials/building_acrn_in_docker.html#build-the-docker-image-from-dockerfile 
 download HV code: git clone https://github.com/projectacrn/acrn-hypervisor 
 cd acrn-hypervisor
 git checkout release_1.6
 "git branch" to check branch
 "git log" to check commit id
 build whl-ipc-i5 industry image with cmd: docker run -u`id -u`:`id -g` --rm -v $PWD:/workspace clearlinux-acrn-builder:latest bash -c "make BOARD=whl-ipc-i5 SCENARIO=industry FIRMWARE=uefi RELEASE=0" 

3. Expected result:
 HV and acrn tools can be built successfully

4. Current result:
 HV built fail.

5. debug info
 in same compile env, can build default board and scenario config with: docker run -u`id -u`:`id -g` --rm -v $PWD:/workspace clearlinux-acrn-builder:latest bash -c "make clean && make"
 in same compile env, can build sos kernel successfully

6. repetition rate
 4/4

Leo2Chang, 2020-03-31:
[External_System_ID] ACRN-6180

NanlinXie, 2020-04-01:
@Leo2Chang , Could you also update the error log? Thanks!

Leo2Chang, 2020-04-01:
@NanlinXie , attach detail log during building HV for whl-ipc-i5 industry uefi release version
[buildDockerImage_buildACRNfail2.txt](https://github.com/projectacrn/acrn-hypervisor/files/4413993/buildDockerImage_buildACRNfail2.txt)
error: 
   cc1: all warnings being treated as errors
   make[3]: *** [Makefile:22: /workspace/build/misc/tools/acrn-crashlog/acrnprobe/obj/loop.o] Error 1

closed 2020-06-28 by fuzhongl