projectacrn/acrn-hypervisor #4585: No UEFI variable write access in SOS when running ACRN

hbuchsbaum, 2020-04-04:
I am running ACRN on a Debian Buster SOS based installation and trying to add the Debian packaging process to be able to simply install ACRN on our Debian based systems. This install  process also requires to adapt the UEFI boot variables. While there is no problem when installing my ACRN packages the first time on a virgin Debian installation, I cannot update ACRN, since it obviously blocks UEFI variable write access. This blocks removing old and stale boot entries as well as creating new ones. You can verify this by creating a dummy boot entry while running ACRN:

I first verify ACRN is running:
`root@debian:~# dmesg | grep 'Hypervisor detected'`
`[    0.000000] Hypervisor detected: ACRN`

And this is how my boot variable look like before I triy to change them:
`root@debian:~# efibootmgr -v`
`BootCurrent: 0000`
`Timeout: 5 seconds`
`BootOrder: 0000,0006,0008,0009`
`Boot0000* ACRN Hypervisor (Debug)       HD(1,GPT,9ae80608-9e6c-417b-9190-4f72d463da9a,0x1000,0x96000)/File(\EFI\ACRN\ACRN.EFI)b.o.o.t.l.o.a.d.e.r.=.\.E.F.I.\.D.e.b.i.a.n.\.g.r.u.b.x.6.4...e.f.i. .u.a.r.t.=.p.o.r.t.@.0.x.2.F.8. .`
`Boot0006* Debian        HD(1,GPT,9ae80608-9e6c-417b-9190-4f72d463da9a,0x1000,0x96000)/File(\EFI\DEBIAN\SHIMX64.EFI)`
`Boot0008* UEFI: Built-in EFI Shell      VenMedia(5023b95c-db26-429b-a648-bd47664c8012)..BO`
`Boot0009* debian        HD(1,GPT,9ae80608-9e6c-417b-9190-4f72d463da9a,0x1000,0x96000)/File(\EFI\DEBIAN\GRUBX64.EFI)..BO`

Then try to create a dummy boot entry:
`root@debian:~# efibootmgr -c -L test`
`Could not prepare Boot variable: No space left on device` **<-- !!!**

Even verbose read of the variable fails now:
`root@debian:~# efibootmgr -v`
`BootCurrent: 0000`
`Timeout: 5 seconds`
`BootOrder: 0000,0006,0008,0009`
`Boot0000* ACRN Hypervisor (Debug)       HD(1,GPT,9ae80608-9e6c-417b-9190-4f72d463da9a,0x1000,0x96000)/File(\EFI\ACRN\ACRN.EFI)b.o.o.t.l.o.a.d.e.r.=.\.E.F.I.\.D.e.b.i.a.n.\.g.r.u.b.x.6.4...e.f.i. .u.a.r.t.=.p.o.r.t.@.0.x.2.F.8. .`
`Boot0001  Could not parse device path: No such file or directory` **<-- !!!**

A look at the efivars filesystem now result in:
`root@debian:~# ls -al /sys/firmware/efi/efivars/Boot0*`
`-rw-r--r-- 1 root root 278 Apr  4 14:33 /sys/firmware/efi/efivars/Boot0000-8be4df61-93ca-11d2-aa0d-00e098032b8c`
`-rw-r--r-- 1 root root   0 Apr  4 14:35 /sys/firmware/efi/efivars/Boot0001-8be4df61-93ca-11d2-aa0d-00e098032b8c` **<-- !!!**
`-rw-r--r-- 1 root root 122 Apr  4 14:33 /sys/firmware/efi/efivars/Boot0006-8be4df61-93ca-11d2-aa0d-00e098032b8c`
`-rw-r--r-- 1 root root  88 Apr  4 14:33 /sys/firmware/efi/efivars/Boot0008-8be4df61-93ca-11d2-aa0d-00e098032b8c`
`-rw-r--r-- 1 root root 126 Apr  4 14:33 /sys/firmware/efi/efivars/Boot0009-8be4df61-93ca-11d2-aa0d-00e098032b8c`

So we have an empty Boot0001 entry (which makes efibootmgr -v bail out)! Creating the variable is allowed, but writing to it is obviously blocked!

In my opinion the ServiceOS should be allowed to write the UEFI NVRAM, otherwise we won't be able to do any package update unless (regardless of the distribution used) rebooting without ACRN.

salerio-gs, 2020-04-04:
I’d agree with that comment. I am working with a Yocto based SOS and we use EFI variables for handling our built in upgrade, so we need to be able to write them.

zhujun74, 2020-04-05:
ServiceOS should be allowed to write the UEFI NVRAM ? does that meaning IN SOS could upate bootmgr? I think it should be the device package.

zhujun74, 2020-04-05:
software program that is loaded  in SOS  should executed by Open Firmware (or a secondary boot program). (The client program may use services provided by the Open Firmware client interface.)

salerio-gs, 2020-04-05:
WE need to be able to write EFI variables from SOS user space, does that help?

zhujun74, 2020-04-05:
It seems Boot0001 not in the boot order.

gvancuts, 2020-04-06:
On most (*) of my systems, I can do that, i.e. update the EFI variables even when running under ACRN. Are you seeing the same problem when creating a valid boot entry?

What kernel are you running in the Service VM?

*: I say most because I don't remember testing this thoroughly on all systems. I should also say that my UP2 is the most capricious one when it comes to manipulating its EFI variables... although doing that from a native environment is not yielding to any better results.

fuzhongl, 2020-04-07:
@hbuchsbaum 
Does this **invalid boot entry** work on your native setup?
Thanks!

hbuchsbaum, 2020-04-13:
This invalid boot entry was only used to demonstrate the inability to write a new boot entry. Of course this also happens when trying to create a valid entry, e.g. during updating ACRN!
@gvancuts 
This is interesting! I'm running ACRN v1.6 with ClearLinux SOS Kernel linux-iot-lts2018-sos-4.19.97-104.

fuzhongl, 2020-04-14:
Can‘t reproduce this issue with Ubuntu/ClearLinux as SOS on our side, will try Debian as SOS later.

gvancuts, 2020-04-14:
A couple of quick questions @hbuchsbaum :
- Does it work as expected in a native Debian?
- Does it work as expected if you run the same kernel in Debian but *without* ACRN?

Can you also confirm what HW (and Bios version) this is seen on?

fuzhongl, 2020-04-16:
@hbuchsbaum  We can't reproduce this issue with Debian SOS on our side:

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Thu Apr 16 04:27:01 2020
root@debian:~# **dmesg |grep ACRN**
[    0.000000] Hypervisor detected: ACRN
[    2.471508] ACRNTrace: Initialized acrn trace module with 4 cpu
[    2.471835] ACRN HVLog: Failed to init last hvlog devs, errno -19
[    2.471835] ACRN HVLog: Initialized hvlog module with 4 cpu

root@debian:~# **efibootmgr -c -L test**
BootCurrent: 0008
Timeout: 1 seconds
BootOrder: 0009,0006,0008,0007,0001,0003,0004,0002,0005,0000
Boot0000* Windows Boot Manager
Boot0001* ubuntu
Boot0002* ACRN
Boot0003* UEFI : LAN : PXE IP4 Intel(R) Ethernet Connection I219-LM
Boot0004* UEFI : LAN : PXE IP6 Intel(R) Ethernet Connection I219-LM
Boot0005* Linux bootloader
Boot0006* ACRN Hypervisor Debian SOS
Boot0007* UEFI : Built-in EFI Shell
Boot0008* ACRN Hypervisor
Boot0009* test
root@debian:~# **ls -al /sys/firmware/efi/efivars/Boot0***
-rw-r--r-- 1 root root 328 Apr 16 04:26 /sys/firmware/efi/efivars/Boot0000-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root 122 Apr 16 04:26 /sys/firmware/efi/efivars/Boot0001-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root 132 Apr 16 04:26 /sys/firmware/efi/efivars/Boot0002-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root 330 Apr 16 04:26 /sys/firmware/efi/efivars/Boot0003-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root 363 Apr 16 04:26 /sys/firmware/efi/efivars/Boot0004-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root 194 Apr 16 04:26 /sys/firmware/efi/efivars/Boot0005-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root 222 Apr 16 04:28 /sys/firmware/efi/efivars/Boot0006-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root  90 Apr 16 04:26 /sys/firmware/efi/efivars/Boot0007-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root 200 Apr 16 04:26 /sys/firmware/efi/efivars/Boot0008-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root 112 Apr 16 05:59 /sys/firmware/efi/efivars/Boot0009-8be4df61-93ca-11d2-aa0d-00e098032b8c
root@debian:~#**efibootmgr -v**
BootCurrent: 0008
Timeout: 1 seconds
BootOrder: 0009,0006,0008,0007,0001,0003,0004,0002,0005,0000
Boot0000* Windows Boot Manager  VenHw(99e275e7-75a0-4b37-a2e6-c5385e6c00cb)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...}...............
Boot0001* ubuntu        HD(1,GPT,e4e65964-8570-49c8-bc2e-373170230beb,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)
Boot0002* ACRN  VenHw(99e275e7-75a0-4b37-a2e6-c5385e6c00cb)
Boot0003* UEFI : LAN : PXE IP4 Intel(R) Ethernet Connection I219-LM     PciRoot(0x0)/Pci(0x1f,0x6)/MAC(54b2030f4cd2,0)/IPv4(0.0.0.00.0.0.0,0,0)..BO
Boot0004* UEFI : LAN : PXE IP6 Intel(R) Ethernet Connection I219-LM     PciRoot(0x0)/Pci(0x1f,0x6)/MAC(54b2030f4cd2,0)/IPv6([::]:<->[::]:,0,0)..BO
Boot0005* Linux bootloader      VenHw(99e275e7-75a0-4b37-a2e6-c5385e6c00cb)
Boot0006* ACRN Hypervisor Debian SOS    HD(1,GPT,be1c1b14-53bb-4b27-81dd-d87e13a976e3,0x800,0x100000)/File(\EFI\acrn\acrn.efi)b.o.o.t.l.o.a.d.e.r.=.\.E.F.I.\.d.e.b.i.a.n.\.g.r.u.b.x.6.4...e.f.i. .
Boot0007* UEFI : Built-in EFI Shell     VenMedia(5023b95c-db26-429b-a648-bd47664c8012)..BO
Boot0008* ACRN Hypervisor       HD(1,GPT,be1c1b14-53bb-4b27-81dd-d87e13a976e3,0x800,0x100000)/File(\EFI\acrn\acrn.efi)b.o.o.t.l.o.a.d.e.r.=.\.E.F.I.\.d.e.b.i.a.n.\.g.r.u.b.x.6.4...e.f.i. .
Boot0009* test  HD(1,GPT,be1c1b14-53bb-4b27-81dd-d87e13a976e3,0x800,0x100000)/File(\EFI\debian\grub.efi)
root@debian:~#root@debian:~# **cat /etc/os-release**
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
root@debian:~#**uname -a**
Linux debian 5.4.28-PKT-200203T060100Z-96550-g2680b519fd72 #1 SMP PREEMPT Thu Apr 9 07:59:16 UTC 2020 x86_64 GNU/Linux
root@debian:~# **cat /proc/cmdline**
BOOT_IMAGE=/boot/bzImage console=tty0 console=ttyS0 root=/dev/sda2 rw rootwait ignore_loglevel no_timer_check consoleblank=0 i915.nuclear_pageflip=1 i915.avail_planes_per_pipe=0x01010F i915.domain_plane_owners=0x011111110000 i915.enable_gvt=1 i915.enable_guc=0 hvlog=2M@0x1FE00000 memmap=2M$0x1FE00000
root@debian:~#

hbuchsbaum, 2020-04-21:
I just verified that changing the boot variable is possible on my NUC7i7DNH. I still experience this issue on my TTTECH MFN-100 which is a Kontron COMe-mAL10 Version 2.0.0 based industrial PC using BIOS revision 5.12. Thus it seems MFN's BIOS somehow interferes with the way ACRN accesses/mediates the access to the UEFI boot variables. Maybe someone on the list can verify this on this specific HW, or give my any hint how I can dig deeper into it to make ACRN coexist with such a BIOS, since this may not be the only BIOS with such a deficiency (given the fact it does no show it witout ACRN!).

fuzhongl, 2020-04-22:
@hbuchsbaum Could you please share the link about the power supply of TTTECH MFN-100?
We need it to reproduce this issue on TTTECH MFN-100.
Thanks!

gvancuts, 2020-04-22:
Hi @hbuchsbaum , while this access is not blocked by ACRN today, the plan is to block it in the future as it relies on using the SMM mode, and that can impact severly the real-time performance.

My understanding is that the requirement you are trying to fulfill here is to upgrade ACRN (and its components) in a system that's already running ACRN and part of this involves running `efibootmgr` to add a new boot entry. With that assumption in mind, I have been wondering if there would be another mechanism that would **not** require such access to be available, and I was thinking that perhaps you could simply always use `acrn.efi` and replace that file without touching the boot order. If there is a need to have the current (or last known good version) of ACRN available, perhaps another permanent boot entry using `acrn-lkg.efi` could be set permanently in the EFI firmware and your package simply moves/renames files as `acrn.efi` and `acrn-lkg.efi` during an update. As long as we're only manipulating files on the filesystem, we do not realy on this SMM mode and the real-time characteristics would be preserved.

What do you think?

salerio-gs, 2020-04-24:
This is good to know. We use **efibootmgr** in a Yocto based guest to switch the boot order after an upgrade (our boot entries are fixed after install)

gvancuts, 2020-04-24:
> This is good to know. We use **efibootmgr** in a Yocto based guest to switch the boot order after an upgrade (our boot entries are fixed after install)

When you say Yocto guest, do you mean with Yocto running as the Service VM operating system?

Based on what I have described, do you believe there is anything you would **not** be able to do?

salerio-gs, 2020-04-24:
Yes sorry, a Yocto SOS my bad, I’m sure there would be a way around it if we could not use efibootmgr

gvancuts, 2020-04-24:
> Yes sorry, a Yocto SOS my bad, I’m sure there would be a way around it if we could not use efibootmgr

Great, thanks for the confirmation! Do let us know though if you hit a wall so we can understand the requirement and help address it!

fuzhongl, 2020-05-15:
@hbuchsbaum This issue is fixed by multiboot2(booting with acrn.bin) on MFN-100:
root@debian:~# **dmidecode --type 2**
# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 3.0.0 present.
Handle 0x0011, DMI type 2, 15 bytes
Base Board Information
        **Manufacturer: Kontron Europe GmbH**
        Product Name: COMe-mAL10 E2
        Version: 2.0.1
        Serial Number: DWD3T0042
        Asset Tag: 34009-0800-16-7
        Features:
                Board is removable
                Board is replaceable
        Location In Chassis: Not Specified
        Chassis Handle: 0x0000
        Type: Processor Module
        Contained Object Handles: 0

root@debian:~# **cat /proc/cmdline**
root=/dev/sda2 console=ttyS1 idle=halt console=tty0 consoleblank=0 cryptomgr.notests intel_iommu=igfx_off no_timer_check noreplace-smp rcu_nocbs=0-64 rcupdate.rcu_expedited=1 tsc=reliable i915.nuclear_pageflip=1 i915.avail_planes_per_pipe=0x01010F i915.domain_plane_owners=0x011111110000 i915.enable_gvt=1 i915.enable_guc=0 hugepagesz=1G hugepages=4
root@debian:~# **uname -a**
Linux debian 4.19.119-acrn-sos #1 SMP PREEMPT Sun May 10 18:10:39 CEST 2020 x86_64 GNU/Linux
root@debian:~# **dmesg |grep ACRN**
[    0.000000] Hypervisor detected: ACRN
[    1.220031] ACRNTrace: Initialized acrn trace module with 4 cpu
[    1.222877] ACRN HVLog: no fixed memory reserve for hvlog.

root@debian:~# **efibootmgr -c -L test**
BootCurrent: 0003
Timeout: 4 seconds
BootOrder: 0004,0003,0001,0002,0000
Boot0000* ACRN
Boot0001* ACRN Debian SOS
Boot0002* debian
Boot0003* UEFI: Built-in EFI Shell
Boot0004* test
root@debian:~#

Could you please double check on your side?
Thanks!

closed 2020-06-28 by fuzhongl