projectacrn/acrn-hypervisor #4590: Race condition occurs, while building acrn tools with Yocto

saininav, 2020-04-07:
Makefile in acrnprobe, generates version.h header file. This header is required while compiling main.c

https://github.com/projectacrn/acrn-hypervisor/tree/master/misc/tools/acrn-crashlog/acrnprobe.

In my multi-core Yocto build environment, main.c gets compiled while version.h still being generated. 

Build Error logs:
| main.c: In function 'main':
| main.c:95:5: error: 'AP_MAJOR_VERSION' undeclared (first use in this function)
|    95 |     AP_MAJOR_VERSION, AP_MINOR_VERSION,
|       |     ^~~~~~~~~~~~~~~~
| main.c:95:5: note: each undeclared identifier is reported only once for each function it appears in
| main.c:95:23: error: 'AP_MINOR_VERSION' undeclared (first use in this function)
|    95 |     AP_MAJOR_VERSION, AP_MINOR_VERSION,
|       |                       ^~~~~~~~~~~~~~~~
| main.c:96:5: error: 'AP_BUILD_VERSION' undeclared (first use in this function)
|    96 |     AP_BUILD_VERSION, AP_BUILD_USER,
|       |     ^~~~~~~~~~~~~~~~
| main.c:96:23: error: 'AP_BUILD_USER' undeclared (first use in this function)
|    96 |     AP_BUILD_VERSION, AP_BUILD_USER,
|       |                       ^~~~~~~~~~~~~
| main.c:97:5: error: 'AP_BUILD_TIME' undeclared (first use in this function)
|    97 |     AP_BUILD_TIME);
|       |     ^~~~~~~~~~~~~
| Makefile:24: recipe for target '/poky/build-acrn/master-acrn-sos/work/corei7-64-oe-linux/acrn-tools/1.6-r0/build/misc/tools/acrn-crashlog/acrnprobe/obj/main.o' failed

wenlingz, 2020-04-08:
@chengangc could you check this issue and fix it, thanks

fuzhongl, 2020-04-16:
[External_System_ID] ACRN-6218

closed 2020-04-17 by fuzhongl