projectacrn/acrn-hypervisor #4706: Intel NUC6CAYH: device Passthru not working

ShibiManoharan, 2020-04-23:
getting 
"pci init passthru
modify_bar_registration: bypass for pci-passthru 0:4.0" when try to passthru any device (eg.wifi) from acrn-dm. Without passthru it is working fine.

Environment:
Device: intel nuc6cayh
SOS: Clear linux 32610
VT-d enabled in bios. But not showing with "cat /proc/cpuinfo | grep vmx".

gvancuts, 2020-04-24:
Can you share which version of ACRN (acrn.efi and acrn-dm) you are using (esp. if you are not using the ones that come with that version of Clear Linux)?

Can you also share the launch script you are trying with?

ShibiManoharan, 2020-04-26:
I am using acrn-2020w15.2-140000p acrn tag. 
acrn-dm tag : DM version is: 1.7-unstable-140077f2-dirty (daily tag:acrn-2020w15.2-140000p),
I used the acrn.efi (copied to the boot partition) build on this tag. 
one more thing, iwlwifi driver(wifi) is also not found in the sos.

[launch_uos_android.txt](https://github.com/projectacrn/acrn-hypervisor/files/4534863/launch_uos_android.txt)

Got the below error messages:
_pci init passthru
acrn-dm: MSI not supported for PCI 0/19/0
modify_bar_registration: bypass for pci-passthru 0:19.0

modify_bar_registration: bypass for pci-passthru 0:4.0
pci passthru init failed_

gvancuts, 2020-04-28:
Can you also paste the output of `lspci`? On my machine, PCI device 00:19.0 is a SPI controller.

ShibiManoharan, 2020-04-29:
Right now I am not able to access my device since it is in another location. The following is taken from acrn_hypervisor repo. I generated the launch script using app.py. Tried individually for wifi and bluetooth passthrough.

<PCI_DEVICE>
	00:00.0 Host bridge: Intel Corporation Celeron N3350/Pentium N4200/Atom E3900 Series Host Bridge (rev 0b)
	00:02.0 VGA compatible controller: Intel Corporation Device 5a85 (rev 0b)
	Region 0: Memory at 90000000 (64-bit, non-prefetchable) [size=16M]
	Region 2: Memory at 80000000 (64-bit, prefetchable) [size=256M]
	00:0e.0 Audio device: Intel Corporation Celeron N3350/Pentium N4200/Atom E3900 Series Audio Cluster (rev 0b)
	Region 0: Memory at 91410000 (64-bit, non-prefetchable) [size=16K]
	Region 4: Memory at 91000000 (64-bit, non-prefetchable) [size=1M]
	00:0f.0 Communication controller: Intel Corporation Celeron N3350/Pentium N4200/Atom E3900 Series Trusted Execution Engine (rev 0b)
	Region 0: Memory at 91427000 (64-bit, non-prefetchable) [size=4K]
	00:12.0 SATA controller: Intel Corporation Celeron N3350/Pentium N4200/Atom E3900 Series SATA AHCI Controller (rev 0b)
	Region 0: Memory at 91414000 (32-bit, non-prefetchable) [size=8K]
	Region 1: Memory at 91424000 (32-bit, non-prefetchable) [size=256]
	Region 5: Memory at 91423000 (32-bit, non-prefetchable) [size=2K]
	00:13.0 PCI bridge: Intel Corporation Celeron N3350/Pentium N4200/Atom E3900 Series PCI Express Port A #1 (rev fb)
	00:13.1 PCI bridge: Intel Corporation Celeron N3350/Pentium N4200/Atom E3900 Series PCI Express Port A #2 (rev fb)
	00:13.2 PCI bridge: Intel Corporation Celeron N3350/Pentium N4200/Atom E3900 Series PCI Express Port A #3 (rev fb)
	00:15.0 USB controller: Intel Corporation Celeron N3350/Pentium N4200/Atom E3900 Series USB xHCI (rev 0b)
	Region 0: Memory at 91400000 (64-bit, non-prefetchable) [size=64K]
	00:16.0 Signal processing controller: Intel Corporation Celeron N3350/Pentium N4200/Atom E3900 Series I2C Controller #1 (rev 0b)
	Region 0: Memory at 91422000 (64-bit, non-prefetchable) [size=4K]
	Region 2: Memory at 91421000 (64-bit, non-prefetchable) [size=4K]
	00:18.0 Signal processing controller: Intel Corporation Celeron N3350/Pentium N4200/Atom E3900 Series HSUART Controller #1 (rev 0b)
	Region 0: Memory at 91420000 (64-bit, non-prefetchable) [size=4K]
	Region 2: Memory at 9141f000 (64-bit, non-prefetchable) [size=4K]
	00:19.0 Signal processing controller: Intel Corporation Celeron N3350/Pentium N4200/Atom E3900 Series SPI Controller #1 (rev 0b)
	Region 0: Memory at 9141e000 (64-bit, non-prefetchable) [size=4K]
	Region 2: Memory at 9141d000 (64-bit, non-prefetchable) [size=4K]
	00:19.1 Signal processing controller: Intel Corporation Celeron N3350/Pentium N4200/Atom E3900 Series SPI Controller #2 (rev 0b)
	Region 0: Memory at 9141c000 (64-bit, non-prefetchable) [size=4K]
	Region 2: Memory at 9141b000 (64-bit, non-prefetchable) [size=4K]
	00:19.2 Signal processing controller: Intel Corporation Celeron N3350/Pentium N4200/Atom E3900 Series SPI Controller #3 (rev 0b)
	Region 0: Memory at 9141a000 (64-bit, non-prefetchable) [size=4K]
	Region 2: Memory at 91419000 (64-bit, non-prefetchable) [size=4K]
	00:1a.0 Serial bus controller [0c80]: Intel Corporation Celeron N3350/Pentium N4200/Atom E3900 Series PWM Pin Controller (rev 0b)
	Region 0: Memory at 91418000 (64-bit, non-prefetchable) [size=4K]
	Region 2: Memory at 91417000 (64-bit, non-prefetchable) [size=4K]
	00:1f.0 ISA bridge: Intel Corporation Celeron N3350/Pentium N4200/Atom E3900 Series Low Pin Count Interface (rev 0b)
	00:1f.1 SMBus: Intel Corporation Celeron N3350/Pentium N4200/Atom E3900 Series SMBus Controller (rev 0b)
	Region 0: Memory at 91416000 (64-bit, non-prefetchable) [size=256]
	01:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS5229 PCI Express Card Reader (rev 01)
	Region 0: Memory at 91300000 (32-bit, non-prefetchable) [size=4K]
	02:00.0 Network controller: Intel Corporation Device 24fb (rev 10)
	Region 0: Memory at 91200000 (64-bit, non-prefetchable) [size=8K]
	03:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 15)
	Region 2: Memory at 91104000 (64-bit, non-prefetchable) [size=4K]
	Region 4: Memory at 91100000 (64-bit, non-prefetchable) [size=16K]

gvancuts, 2020-04-29:
Can you remove all the Bluetooth (`00:19.0`) stuff from the launch script and try again?

ShibiManoharan, 2020-04-29:
Without bluetooth, i got the below error message.
_pci init passthru
modify_bar_registration: bypass for pci-passthru 0:4.0
pci passthru init failed_

gvancuts, 2020-04-29:
> Note: I even enabled VT-D in BIOS. But, from Clear linux, vmx is not found in /proc/cpuinfo

I have not checked with Clear Linux lately (will try to tomorrow), but I see this entry on my system. My Bios version is 0047, what is yours (`sudo dmidecode | grep Version`)

ShibiManoharan, 2020-04-30:
dmidecode | grep Version
        Version: AYAPLCEL.86A.0055.2018.0821.1152
        Version: J26843-408
        Version: J23203-407
        Version: 2.0
        Version: Intel(R) Celeron(R) CPU J3455 @ 1.50GHz

fuzhongl, 2020-07-29:
@ShibiManoharan  Currently, ACRN doesn't support passthrough WIFI/BT on NUC platform.

closed 2021-03-15 by fuzhongl