projectacrn/acrn-hypervisor #4716: VM fault occurs during Android UOS reboot into fastboot mode.

lonzoc, 2020-04-26:
ACRN hypervisor report VM fault during Android UOS reboot into fastboot mode.

Hardware infomation:
- A3920
[vm-fault-android-fastboot.txt](https://github.com/projectacrn/acrn-hypervisor/files/4534907/vm-fault-android-fastboot.txt)

Software stack infomation:
- ACRN hypervisor v1.6
- ACRN kernel v1.6
- Android UOS  GP2.0_0.5.3_PV

Reproduce steps：
1. use launch_uos.sh to lunch Android UOS;
2. connect board to PC with USB cable;
3. adb reboot bootloader

Our previous software stack are based on ACRN v1.2  and has no such issue.

lonzoc, 2020-04-26:
[Uploading vm-fault-android-fastboot.txt…]()

lonzoc, 2020-04-27:
Append some debug logs

[65767906us][cpu=1][vm1:vcpu0][sev=5][seq=97]:VM 1 VCPU 0 successfully launched
[66078446us][cpu=1][vm1:vcpu0][sev=3][seq=98]:decode opcode failed: opcode=247 vie->op.op_type == TYPE_NONE
[66089165us][cpu=1][vm1:vcpu0][sev=3][seq=99]:inst0: 41
[66094867us][cpu=1][vm1:vcpu0][sev=3][seq=100]:inst1: f7
[66100662us][cpu=1][vm1:vcpu0][sev=3][seq=101]:inst2: 45
[66106453us][cpu=1][vm1:vcpu0][sev=3][seq=102]:inst3: 0
[66112143us][cpu=1][vm1:vcpu0][sev=3][seq=103]:inst4: 0
[66117838us][cpu=1][vm1:vcpu0][sev=3][seq=104]:inst5: 0
[66123541us][cpu=1][vm1:vcpu0][sev=3][seq=105]:inst6: 0
[66129239us][cpu=1][vm1:vcpu0][sev=3][seq=106]:inst7: 40
[66135032us][cpu=1][vm1:vcpu0][sev=3][seq=107]:inst8: 0
[66140735us][cpu=1][vm1:vcpu0][sev=3][seq=108]:inst9: 0
[66146430us][cpu=1][vm1:vcpu0][sev=3][seq=109]:inst10: 0
[66152221us][cpu=1][vm1:vcpu0][sev=3][seq=110]:inst11: 0
[66158015us][cpu=1][vm1:vcpu0][sev=3][seq=111]:inst12: 0
[66163808us][cpu=1][vm1:vcpu0][sev=3][seq=112]:inst13: 0
[66169603us][cpu=1][vm1:vcpu0][sev=3][seq=113]:inst14: 0
[66175401us][cpu=1][vm1:vcpu0][sev=3][seq=114]:decode instruction failed @ 0x0000000012910530: ret=-2, cpumode=3
[66186691us][cpu=1][vm1:vcpu0][sev=5][seq=115]:page fault happen during decode_instruction
[66195831us][cpu=1][vm1:vcpu0][sev=2][seq=116]:Guest Linear Address: 0x000000008020c704
[66204670us][cpu=1][vm1:vcpu0][sev=2][seq=117]:Guest Physical Address address: 0x000000008020c704

terryzouhao, 2020-04-27:
@lonzoc already bisect to "dm: vPCI: add assign/deassign PCI device IC APIs", commit ID: aa38ed5b69999c9c26a7d447d11296c6cea7a248.  @lifeix to help check.

lonzoc, 2020-04-28:
git bisect result
aa38ed5b -- good
dafa3da6  -- compile error 
checkout the last commit of your patch set: 4adad73c  --- bad

terryzouhao, 2020-04-28:
@lonzoc just double checked our CI build log, there was one PR#4375: "Vpci: Pass-thru PCI device in DM-launched VM" that contained below 8 commits. We just verified 2ca01206 and 4adad73c, so dafa3da compile error was fixed by next commits:

- 4adad73c hv: mmio: refine mmio access handle lock granularity
- fbe57d9f hv: vpci: restrict SOS access assigned PCI device
- 9d3d9c3d dm: vpci: restrict SOS access assigned PCI device
- e8479f84 hv: vPCI: remove passthrough PCI device unuse code
- 9fa6eff3 dm: vPCI: remove passthrough PCI device unused code
- dafa3da6 vPCI: split passthrough PCI device from DM to HV
- aa38ed5b dm: vPCI: add assign/deassign PCI device IC APIs
- fe3182ea hv: vPCI: add assign/deassign PCI device HC APIs

For, 4adad73c commit, there was no Android Guest test on MRB. But take it easy, we will try it, and back to you with results. @fuzhongl 
@acrnsi:   CONF_TOOL/KBL/INDUSTRY/WaaG_PREEMPT_RT_Test — Passed
@acrnsi:   KBL/HYBRID/Prelaunch-Zephyr_SoS_LaaG_Test — Passed
@acrnsi:   KBL/LOGICAL_PARTITION/FUSA_Test — Passed
@acrnsi:   LOC_Test — Passed
@acrnsi:   MISRA-C_Check — Pass
@acrnsi:  SDC/INDUSTRY/LOGICAL_PARTITION/HYBRID Build — Build finished.
@acrnsi:  SECURE-CODING — Passed
@acrnsi:  Tracked-On/JIRA_GIT_ISSUE_Check — Pass
@acrnsi:  WHL/INDUSTRY/WaaG_PREEMPT_RT_Test — Passed

lifeix, 2020-04-29:
@lonzoc  On our board, the android will boot failed when probe WIFI PCI device. If we remove the WIFI and Buletooth device passthrough (-s 18,passthru,3/0/0,keep_gsi and -s 24,passthru,0/18/0), the android could boot. I will focus on this later. But we didn't meet your issue. I think you may trigger a instruction vmexit which ACRN doesn't support. You could try to add that instruction emulation support.

terryzouhao, 2020-04-29:
@lonzoc we had re-tested on below 2 commits, just before and after PR#4375. Logically, both could boot AaaG successfully (4adad73c need remove/update BT/WiFi passthrough, @lifeix can provide more detailed steps). 
So seems your "decode instruction failed @ 0x0000000012910530: ret=-2, cpumode=3" failure is specific to your board: A3920 is not in our opensource support HW list, could you re-try on APL UP2 (sbl support also) or APL NUC: https://projectacrn.github.io/latest/reference/hardware.html

- 4adad73c hv: mmio: refine mmio access handle lock granularity
//ACRN,SOS boot successfully, need update AaaG launch script(BT/Wifi) to launch Android.
- 2ca01206 Makefile: fix build issue on old gcc
// ACRN, SOS and AaaG boot successfully.

lonzoc, 2020-04-29:
@lifeix Did you try 'adb reboot bootloader' command?  lunch Android UOS is ok for our board, but will fault when entering Android fastboot mode.

lifeix, 2020-04-29:
@lonzoc Yes, I could reproduce your problem. This issue is before my patch, each EPT mapping for PCI device BAR is mapped already by DeviceModel. After my patch, until the guest (or BIOS) size the PCI BAR size, this mapping wasn't mapped. SO in this case, when android wants to access a PCI BAR, it will trigger the EPT violation. (I think Adb will access the usb_xdci (00:15.1) device before its BAR EPT was mapped). You could have a WA to fix this.

terryzouhao, 2020-04-29:
@lonzoc it is an issue/conflict of android guest with @lifeix 's patch, but only happen with Android fasterboot (access PCI BAR before EPT mapping). We will eventually fix it in next design, and welcome to discuss in mailinglist/TCM with us together. For short term, could you make a Work Around for your usage, based on @lifeix 's analysis above, or you need more detailed suggestion for WA ?

lifeix, 2020-04-30:
@lonzoc U could WA this issue by below change
diff --git a/hypervisor/dm/vpci/pci_pt.c b/hypervisor/dm/vpci/pci_pt.c
index 9335b9e4..90a953ad 100644
--- a/hypervisor/dm/vpci/pci_pt.c
+++ b/hypervisor/dm/vpci/pci_pt.c
@@ -113,7 +113,7 @@ static void vdev_pt_unmap_mem_vbar(struct pci_vdev *vdev, uint32_t idx)
  * @pre vdev->vpci != NULL
  * @pre vdev->vpci->vm != NULL
  */
-static void vdev_pt_map_mem_vbar(struct pci_vdev *vdev, uint32_t idx)
+void vdev_pt_map_mem_vbar(struct pci_vdev *vdev, uint32_t idx)
 {
        struct pci_vbar *vbar = &vdev->vbars[idx];

diff --git a/hypervisor/dm/vpci/vpci.c b/hypervisor/dm/vpci/vpci.c
index 8546cae8..bac100d4 100644
--- a/hypervisor/dm/vpci/vpci.c
+++ b/hypervisor/dm/vpci/vpci.c
@@ -779,6 +779,12 @@ int32_t vpci_assign_pcidev(struct acrn_vm *tgt_vm, struct acrn_assign_pcidev *pc
                                        }
                                }
                                pci_vdev_write_vbar(vdev, idx, pcidev->bar[idx]);
+                               if (vdev->vbars[idx].type == PCIBAR_MEM32) {
+                                       vdev_pt_map_mem_vbar(vdev, idx);
+                               } else if (vdev->vbars[idx].type == PCIBAR_MEM64HI) {
+                                       vdev_pt_map_mem_vbar(vdev, idx - 1U);
+                               }
+
                        }

                        vdev->flags |= pcidev->type;
diff --git a/hypervisor/dm/vpci/vpci_priv.h b/hypervisor/dm/vpci/vpci_priv.h
index 793ddfe2..5f759adc 100644
--- a/hypervisor/dm/vpci/vpci_priv.h
+++ b/hypervisor/dm/vpci/vpci_priv.h
@@ -138,4 +138,5 @@ void pci_vdev_write_vcfg(struct pci_vdev *vdev, uint32_t offset, uint32_t bytes,

 uint32_t pci_vdev_read_vbar(const struct pci_vdev *vdev, uint32_t idx);
 void pci_vdev_write_vbar(struct pci_vdev *vdev, uint32_t idx, uint32_t val);
+void vdev_pt_map_mem_vbar(struct pci_vdev *vdev, uint32_t idx);

lonzoc, 2020-04-30:
@lifeix @terryzouhao Thanks for your WA solution, i will try later.

lonzoc, 2020-05-26:
Hi, is this issue fixed in v1.6.1 release?@lifeix @terryzouhao

terryzouhao, 2020-05-26:
Hi @lonzoc , i don't think we have the fix in v1.6.1 release yet.  @lifeix may provide more details, thanks.

closed 2022-07-05 by fuzhongl