projectacrn/acrn-hypervisor #4810: Build failure with GCC-10 [-Werror=array-bounds]

saininav, 2020-05-15:
After moving gcc version to GCC-10, started getting build failures.

Error log:
arch/x86/page.c: In function 'init_ept_mem_ops':
arch/x86/page.c:240:48: error: array subscript <unknown> is outside array bounds of 'struct page[0][1]' [-Werror=array-bounds]
  240 |   ept_pages_info[vm_id].ept.nworld_pml4_base = pre_uos_nworld_pml4_pages[vm_id];
      |                                                ^~~~~~~~~~~~~~~~~~~~~~~~~
arch/x86/page.c:92:20: note: while referencing 'pre_uos_nworld_pml4_pages'
   92 | static struct page pre_uos_nworld_pml4_pages[PRE_VM_NUM][PML4_PAGE_NUM(PRE_VM_EPT_ADDRESS_SPACE(CONFIG_UOS_RAM_SIZE))];
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~~
arch/x86/page.c:241:48: error: array subscript <unknown> is outside array bounds of 'struct page[0][1]' [-Werror=array-bounds]
  241 |   ept_pages_info[vm_id].ept.nworld_pdpt_base = pre_uos_nworld_pdpt_pages[vm_id];
      |                                                ^~~~~~~~~~~~~~~~~~~~~~~~~
arch/x86/page.c:93:20: note: while referencing 'pre_uos_nworld_pdpt_pages'
   93 | static struct page pre_uos_nworld_pdpt_pages[PRE_VM_NUM][PDPT_PAGE_NUM(PRE_VM_EPT_ADDRESS_SPACE(CONFIG_UOS_RAM_SIZE))];
      |                    ^~~~~~~~~~~~~~~~~~~~~~~~~
arch/x86/page.c:242:46: error: array subscript <unknown> is outside array bounds of 'struct page[0][10]' [-Werror=array-bounds]
  242 |   ept_pages_info[vm_id].ept.nworld_pd_base = pre_uos_nworld_pd_pages[vm_id];
      |                                              ^~~~~~~~~~~~~~~~~~~~~~~
arch/x86/page.c:94:20: note: while referencing 'pre_uos_nworld_pd_pages'
   94 | static struct page pre_uos_nworld_pd_pages[PRE_VM_NUM][PD_PAGE_NUM(PRE_VM_EPT_ADDRESS_SPACE(CONFIG_UOS_RAM_SIZE))];
      |                    ^~~~~~~~~~~~~~~~~~~~~~~
arch/x86/page.c:244:46: error: array subscript <unknown> is outside array bounds of 'struct page[0][5120]' [-Werror=array-bounds]
  244 |   ept_pages_info[vm_id].ept.nworld_pt_base = pre_uos_nworld_pt_pages[vm_id];
      |                                              ^~~~~~~~~~~~~~~~~~~~~~~
arch/x86/page.c:137:20: note: while referencing 'pre_uos_nworld_pt_pages'
  137 | static struct page pre_uos_nworld_pt_pages[PRE_VM_NUM][PT_PAGE_NUM(PRE_VM_EPT_ADDRESS_SPACE(CONFIG_UOS_RAM_SIZE))];
      |                    ^~~~~~~~~~~~~~~~~~~~~~~

fuzhongl, 2020-05-17:
@fyin1 Could you please help to check this GCC-10 related issue?
Thanks!

fuzhongl, 2020-05-18:
[External_System_ID] ACRN-6361

closed 2020-06-28 by fuzhongl