projectacrn/acrn-hypervisor #4811: Using NUC6CAYH with hybrid or industry scenario

hidrol, 2020-05-15:
I'm kind of confused because the "getting started" guide for hybrid scenario says that a serial port is required. But on the other hand nuc6cayh is listed as verified platform. And what about industrial scenario? So if I want to connect to the ACRN console I need a serial port right? How can I get a pre-launched safety VM running on my NUC in the way that I am able to get its output without the serial port?

terryzouhao, 2020-05-18:
thanks @hidrol for your feedback. Yes, current GSG and support list (https://projectacrn.github.io/latest/reference/hardware.html) is a little 'out of date', actually we are going to update it very soon. Specific to your question of APL NUC(nuc6cayh), because there is no Serial Port on it, i don't think you can enable pre-launched safety VM on it, so  "V" of "IU with Safety VM" on nuc6cayh is typo/error, we will PR/update it soon @fuzhongl . thanks.

fuzhongl, 2020-05-18:
@terryzouhao Following PR is under review:
https://github.com/projectacrn/acrn-hypervisor/pull/4819

@hidrol 
About  industrial scenario; you can use NUC6CAYH.
By default, the guest console is redirected to HV console for  industrial scenario. Please refer following modification to disable redirect for  NUC6CAYH:
acrn-hypervisor\hypervisor\scenarios\industry\vm_configurations.c
	{	/* VM1 */
		CONFIG_POST_STD_VM(1),
		.cpu_affinity = VM1_CONFIG_CPU_AFFINITY,
		.vuart[0] = {
			.type = VUART_LEGACY_PIO,
			.addr.port_base = **`INVALID_COM_BASE`**,

Thanks!

hidrol, 2020-05-20:
Is it possible to use something like a usb to uart converter?

gvancuts, 2020-05-20:
> Is it possible to use something like a usb to uart converter?

I don't believe so, but I'll let the engineering team confirm this.

fuzhongl, 2020-05-21:
@gvancuts  You are right!

terryzouhao, 2020-05-21:
@hidrol Hypervisor console need onboard serial port to access ACRN before loading VM, USB-Uart converter could not be recognized:(  
If you want to try pre-launch VM and Logical partition, KBL NUC7i7DNH could be a good choice. 
BTW, APL UP2 also has onboard serial port, logically HV console and pre-launch VM are supported, but we did not fully verified the scenario on it, welcome to submit gitissue or question in mailinglist in case you have UP2 and try again. But if you get chance to refresh your HW, Kaby Lake and Whiskey Lake are preferred :) 
You can find more details in updated homepage: https://projectacrn.github.io/latest/reference/hardware.html

hidrol, 2020-05-22:
@terryzouhao thank you very much for your answer. We currently have in our project the NUC6CAYH with 8GB RAM and 240GB SSD. We currently want to gain some experience with the ACRN Hypervisor for future projects. Therefore we would not like to switch to other hardware now. 

For us it is important that we have 2 user VMs running beside the service VM. Later these User VMs should be realtime capable. But this is not so important at this point in time. 

So the most important question for us is now:
Is it possible to have 2 User VMs running in parallel on the NUC6CAYH? The scenario doesn't play a big role at first. Nevertheless, it would be nice if the industry scenario would run.

In the industry scenario we have some bugs at the moment which we unfortunately can't fix. Perhaps you can take a closer look at them:
https://github.com/projectacrn/acrn-hypervisor/issues/4820

Are there any other guides on how to run the industry scenario on the NUC6CAYH? The SDC scenario runs without problems, but we can only start 1 user VM. Therefore it is rather uninteresting for our work.

terryzouhao, 2020-05-23:
@hidrol very glad to see you have APL NUC and ACRN in current project, and looking forward to more real success with ACRN in your future projects :)
Basically ACRN was started from SDC (2 VMs: SOS + LaaG) on APL, and moving to Industrial scenario(SOS+RT Linux+Windows) with KBL/WHL platform. But that not mean NUC6CAYH can not support Industrial scenario, which also has 4 cores(4C4T). I think it is doable for 2 VMs besides SOS on APL NUC. 
So, you already started trying linux + windows on NUC6CAYH and observed kernel panic with #4820 right. let's follow up your issue in #4820 and we will also try industrial scenario on APL NUC later, keep you updated.
BTW, there is no existed guide of Industrial scenario on APL NUC, but most steps should be similar with current GSG on KBL/WHL, anyway let's try it on APL NUC, and welcome you contribute back to ACRN document and patches, in case any fix for specific issue on APL NUC, thanks in advance !

hidrol, 2020-05-24:
@terryzouhao There was no Windows involved in the setting we tried. Just Clear Linux as Service OS which did fail to boot. After that the goal will be to get two User VMs each with Clear Linux running.

terryzouhao, 2020-05-25:
@hidrol OK, we will try ACRN v1.6.1 release tag and build APL NUC release with default Industrial Scenario, target to Clear Linux as Service OS and 2 Clear Linux VMs as guest.

hidrol, 2020-05-26:
@terryzouhao Sounds good, and thx for your support so far. Let me know if there's something new.

closed 2020-06-28 by fuzhongl