projectacrn/acrn-hypervisor #4812: ERROR with building industry scenario on nuc6cayh

hidrol, 2020-05-16:
I followed the "getting started" guide for the industry scenario. 
After typing the comand: $ make all BOARD=nuc6cayh SCENARIO=industry RELEASE=0
I get this error message:

...
/usr/bin/ld: /home/clear/acrn-hypervisor/build/hypervisor/acrn.out section `.bss' will not fit in region `ram'
/usr/bin/ld: region `ram' overflowed by 4678160 bytes
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:480: /home/clear/acrn-hypervisor/build/hypervisor/acrn.out] Error 1
make[1]: Leaving directory '/home/clear/acrn-hypervisor/hypervisor'
make: *** [Makefile:164: hypervisor] Error 2

I have a nuc6cayh with 8GB ram and updated bios version AYAPLCEL.86A.0066.2020.0107.1027.
Clear Linux Version 31470.

gvancuts, 2020-05-18:
Confirmed on Clear Linux 32700 too.

Possibly the same root cause as #4319 ?

fuzhongl, 2020-06-28:
@hidrol Will fix it in master branch.
Before that, please using xml to build.

fuzhongl, 2020-06-28:
[External_System_ID] ACRN-6490

fuzhongl, 2021-01-12:
Fixed in v2.3

closed 2021-01-12 by fuzhongl