projectacrn/acrn-hypervisor #4822: SOS crashes while booting Vecow spc-5200 whl-i7

darshanraj-ka, 2020-05-19:
Platform : http://www.vecow.com/dispUploadBox/PJ-VECOW/Files/5305.pdf
	         
"Intel(R) Core(TM) i7-8665UE CPU @ 1.70GHz"

```
Version 2.20.1271. Copyright (C) 2020 American Megatrends, Inc.
BIOS Date: 02/06/2020 09:36:06 Ver: E3000XXPUF00008
Press <CTRL + P> to Enter MEBX setup menu
Press <DEL> to enter setup.detected 4 processes, 4 enabled
current on process 0
memory region (480000000, 86C000000) is truncated from region (100000000, 86C000000).[0us][cpu=0][(null)][sev=3][seq=2]:reloc: ~|~ hv start at 0x11000000
ACRN Hypervisor
calibrate_tsc, tsc_khz=2000000
[10783147us][cpu=0][(null)][sev=2][seq=3]:HV version 1.6-2020-05-19 06:39:02-3c64d59-dirty DBG (daily tag:acrn-2020w18.4-140000p) build by draj with acrn-config, start time 10770619us
[10802413us][cpu=0][(null)][sev=2][seq=4]:API version 1.0
[10808630us][cpu=0][(null)][sev=2][seq=5]:Detect processor: Intel(R) Core(TM) i7-8665UE CPU @ 1.70GHz
[10908215us][cpu=0][(null)][sev=2][seq=25]:Start VM id: 0 name: ACRN SOS VM
[10931760us][cpu=0][vm0:vcpu0][sev=3][seq=42]:vlapic: Start Secondary VCPU1 for VM[0]...
ACRN:\>[10951640us][cpu=0][vm0:vcpu0][sev=3][seq=55]:vlapic: Start Secondary VCPU2 for VM[0]...
[10961077us][cpu=2][vm0:vcpu2][sev=3][seq=58]:vlapic: WARNING:invalid vpic wire mode change
[10981304us][cpu=0][vm0:vcpu0][sev=3][seq=78]:vlapic: Start Secondary VCPU3 for VM[0]...
[10990740us][cpu=3][vm0:vcpu3][sev=3][seq=81]:vlapic: WARNING:invalid vpic wire mode change
[22517472us][cpu=0][vm0:vcpu0][sev=3][seq=10891]:error[-14] in GVA2GPA, err_code=0x10
[22526502us][cpu=0][vm0:vcpu0][sev=2][seq=10893]:Guest Linear Address: 0x00000000e00b0002
[22536033us][cpu=0][vm0:vcpu0][sev=2][seq=10894]:Guest Physical Address address: 0x00000000e00b0002
[41648621us][cpu=0][vm0:vcpu0][sev=1][seq=10895]:VM0: triple fault @ guest RIP 0x000000008ced2dc3, exit qualification: 0x0000000000000000

ACRN:\>version
HV version 1.6-2020-05-19 06:39:02-3c64d59-dirty DBG (daily tag: acrn-2020w18.4-140000p) build by draj
API version 1.0
ACRN:\>
ACRN:\>vm_list

VM_UUID                          VM_ID VM_NAME                          VM_STATE
================================ ===== ================================ ========
ACRN:\>
ACRN:\>vcpu_list

VM ID    PCPU ID    VCPU ID    VCPU ROLE    VCPU STATE    THREAD STATE
=====    =======    =======    =========    ==========    ==========
```

- kconfig : [5.4.28+gitAUTOINC+b8c82ba373_4726da6405-r0.config.txt](https://github.com/projectacrn/acrn-hypervisor/files/4648803/5.4.28%2BgitAUTOINC%2Bb8c82ba373_4726da6405-r0.config.txt)

- hvconfig: [acrn-hypervisor_1.6.1-r0.config.txt](https://github.com/projectacrn/acrn-hypervisor/files/4648805/acrn-hypervisor_1.6.1-r0.config.txt)

gvancuts, 2020-05-19:
Hi @darshanraj-ka , do you have access to the Service VM at all when this happens?

darshanraj-ka, 2020-05-19:
@gvancuts : No. 
only ACRN prompt and no VMs are listed
`vm_console` results in a hang.

gvancuts, 2020-05-19:
@NanlinXie , I don't know how to troubleshoot this efficiently, could someone from the engineering jump in and help?

fuzhongl, 2020-05-27:
@gvancuts Checking with @darshanraj-ka ...

fuzhongl, 2020-06-28:
Fixed with following PR:
https://github.com/projectacrn/acrn-hypervisor/pull/4629

closed 2020-06-28 by fuzhongl