projectacrn/acrn-hypervisor #4824: WHL CRB: Hybrid Scenario, SOS console not coming up

nitings, 2020-05-19:
Hi all,
I wanted to get the hybrid scenario up on WHL CRB. My setup is Pre-launch VM (Yocto) + Ubuntu SOS (The SOS kernel I am using is 4.19.106-PKT-200203T060100Z)
vm_list shows two VMs running VM0 and VM1.
The Pre-launch VM (VM0) is coming up fine.
When I launch the command "vm_console 1" from the ACRN shell the VM1 does not boot completely. It prints following log and remains in hang state.
----- Entering VM 1 Shell -----
early console in extract_kernel
input_data: 0x00000000034d63b1
input_len: 0x0000000000cbca96
output: 0x0000000002000000
output_len: 0x0000000002160cd4
kernel_total_size: 0x0000000001e2c000
trampoline_32bit: 0x000000000008d000
booted via startup_32()
Physical KASLR using RDRAND RDTSC...
Virtual KASLR using RDRAND RDTSC...

Decompressing Linux... Parsing ELF... Performing relocations... done.
Booting the kernel.
Memory KASLR using RDRAND RDTSC...

I am able to go back to the ACRN shell by pressing "ctrl+space"
Let me know if any other detail is required.

gvancuts, 2020-05-20:
@NanlinXie , I don't have that HW. Can you ask someone who does to take a look at this please?

fuzhongl, 2020-06-28:
@nitings  As synced, please help to try with acrn 2.0 release.
Thanks!

fuzhongl, 2020-07-07:
@nitings 
Please help to try with acrn 2.0 release and provide whole HV log also.
Thanks!

fuzhongl, 2020-07-29:
@nitings Please help to verify if there is the same issue with acrn v2.0

fuzhongl, 2021-01-10:
Fixed with following PR: #4629 missing in v1.6.1
commit 597f7658fc4b3c9611b07c373286741552a5a199

closed 2021-01-10 by fuzhongl