projectacrn/acrn-hypervisor #4855: Cannot start UOS with Clear Linux on NUC6CAYH

hidrol, 2020-05-28:
After setting up the Service I want to get two UOS both with Clear Linux running in industry scenario. I followed the section "Launch additional User VM" in https://projectacrn.github.io/latest/getting-started/rt_industry.html#set-up-the-acrn-hypervisor-for-industry-scenario. 
After running the script `launch_uos_id3.sh` I get  "tpm:  init_vtpm2:Invalid socket path!" and "Failed to start Network Service"
The launch script I had to modify looks like this.

```
#echo ${passthru_vpid["gpu"]} > /sys/bus/pci/drivers/pci-stub/new_id
#echo ${passthru_bdf["gpu"]} > /sys/bus/pci/devices/${passthru_bdf["gpu"]}/driver/unbind
#echo ${passthru_bdf["gpu"]} > /sys/bus/pci/drivers/pci-stub/bind
echo 100 > /sys/bus/usb/drivers/usb-storage/module/parameters/delay_use
mem_size=200M
#interrupt storm monitor for pass-through devices, params order:
#threshold/s,probe-period(s),intr-inject-delay-time(ms),delay-duration(ms)
intr_storm_monitor="--intr_monitor 10000,10,1,100"

#logger_setting, format: logger_name,level; like following
logger_setting="--logger_setting console,level=4;kmsg,level=3;disk,level=5"

acrn-dm -A -m $mem_size -s 0:0,hostbridge -U 615db82a-e189-4b4f-8dbb-d321343e4ab3 \
   --mac_seed $mac_seed \
   $logger_setting \
   -s 2,virtio-blk,./clearlinux.img \
   -s 3,virtio-console,@stdio:stdio_port \
   --ovmf /usr/share/acrn/bios/OVMF.fd \
   -s 9,virtio-hyper_dmabuf \
   $vm_name
}
if [ "$1" = "-C" ];then
    if [ $(hostname) = "runc" ]; then
            echo "Already in container exit!"
            exit
    fi
    echo "runc_container"
    run_container
else
    launch_clearlinux 3
fi
```
Where do I find the console log file for posting the whole error messages?
The Clear Linux Version is `33050`.

hidrol, 2020-05-28:
After launching the script I get this. 
```
logger: name=console, level=4
logger: name=kmsg, level=3
logger: name=disk, level=5
SW_LOAD: get ovmf path /usr/share/acrn/bios/OVMF.fd, size 0x200000
vm_create: post_vm_id3
VHM api version 1.0
vm_setup_memory: size=0xc800000
open hugetlbfs file /run/hugepage/acrn/huge_lv1/post_vm_id3/615DB82AE1894B4F8DBBD321343E4AB3
open hugetlbfs file /run/hugepage/acrn/huge_lv2/post_vm_id3/615DB82AE1894B4F8DBBD321343E4AB3
level 0 free/need pages:0/101 page size:0x200000
level 1 free/need pages:0/0 page size:0x40000000
to reserve more free pages:
to reserve pages (+orig 0): echo 101 > /sys/kernel/mm/hugepages/hugepages-2048kB/nr_hugepages
now enough free pages are reserved!

try to setup hugepage with:
	level 0 - lowmem 0xc800000, biosmem 0x200000, highmem 0x0
	level 1 - lowmem 0x0, biosmem 0x0, highmem 0x0
total_size 0x140200000

mmap ptr 0x0x7f6c565a8000 -> baseaddr 0x0x7f6c56600000
mmap 0xc800000@0x7f6c56600000
touch 100 pages with pagesz 0x200000
mmap 0x200000@0x7f6d56400000
touch 1 pages with pagesz 0x200000

really setup hugepage with:
	level 0 - lowmem 0xc800000, biosmem 0x200000, highmem 0x0
	level 1 - lowmem 0x0, biosmem 0x0, highmem 0x0
vm_init_vdevs
No correct pm notify channel given
Listening 0...
polling 0...
pci init hostbridge
pci init virtio-blk
pci init virtio-console
pci init virtio-hyper_dmabuf
tpm: init_vtpm2:Invalid socket path!
                                    /tmp/dm.XtMEIym     24:     Device (PCI0)
                                                                             Warning  3073 -        Multiple types ^  (Device object requires either a _HID or _ADR, but not both)

                                                   /tmp/dm.XtMEIym     91:       })
                                                                                   Remark   2095 -              ^ Effective AML package length is zero

                       /tmp/dm.XtMEIym    148:     Processor (CPU0, 0x00, 0x00000000, 0x00) {}
                                                                                              Warning  3168 -                          ^ Legacy Processor() keyword detected. Use Device() keyword instead.

                                                                            /tmp/dm.XtMEIym    149:     Processor (CPU1, 0x01, 0x00000000, 0x00) {}
                    Warning  3168 -                          ^ Legacy Processor() keyword detected. Use Device() keyword instead.

  acrn_sw_load
              SW_LOAD: partition blob /usr/share/acrn/bios/OVMF.fd size 2097152 copy to guest 0xffe00000
                                                                                                        SW_LOAD: build e820 9 entries to addr: 0x7f6c566ef008
SW_LOAD: entry[0]: addr 0x0000000000000000, size 0x00000000000a0000,  type 0x1
SW_LOAD: entry[1]: addr 0x00000000000a0000, size 0x0000000000060000,  type 0x2
SW_LOAD: entry[2]: addr 0x0000000000100000, size 0x000000000c700000,  type 0x1
SW_LOAD: entry[3]: addr 0x000000000c800000, size 0x0000000073800000,  type 0x2
SW_LOAD: entry[4]: addr 0x0000000080000000, size 0x0000000008000000,  type 0x2
SW_LOAD: entry[5]: addr 0x00000000db000000, size 0x0000000004000000,  type 0x2
SW_LOAD: entry[6]: addr 0x00000000df000000, size 0x0000000001000000,  type 0x2
SW_LOAD: entry[7]: addr 0x00000000e0000000, size 0x0000000020000000,  type 0x2
SW_LOAD: entry[8]: addr 0x0000000100000000, size 0x0000000040000000,  type 0x2
SW_LOAD: ovmf_entry 0xfffffff0
                              add_cpu

```

hidrol, 2020-05-28:
I generated the launch_uos_id3.sh script with: 
```
$ cd acrn-hypervisor
$ export board_file=$PWD/misc/acrn-config/xmls/board-xmls/nuc6cayh.xml
$ export scenario_file=$PWD/misc/acrn-config/xmls/config-xmls/nuc6cayh/industry.xml
$ export launch_file=$PWD/misc/acrn-config/xmls/config-xmls/nuc6cayh/industry_launch_6uos.xml
$ python misc/acrn-config/launch_config/launch_cfg_gen.py --board $board_file --scenario $scenario_file --launch $launch_file --uosid 0
```

fuzhongl, 2020-05-28:
@hidrol please run `acrnctl list` in sos console(you can ssh to sos), if uos is running
,please losetup your guest image(./clearlinux.img) and check if console=hvc0 is in UOS kernel cmdline:
$ losetup -f -P --show ./clearlinux.img
# It will output loopx; following is loop0 for example
$ mount /dev/loop0p1 /mnt
$ cat /mnt/loader/entries/*.conf  
#please replace with your .conf
Thanks!

fuzhongl, 2020-05-29:
@hidrol Please checkout to v1.6 branch and using misc/acrn-config/xmls/config-xmls/nuc6cayh/industry_launch_2uos.xml
Thanks!

hidrol, 2020-06-02:
@terryzouhao Do you already have any new details about running  2 CL VMs on APL NUC you could share?

terryzouhao, 2020-06-02:
hi @hidrol , for APL NUC, actually we target GSG to support launch your config, e.g., industrial (RT+windows) by default , for 2 CL VMs, you only need modify sample launch UOS scripts(launch_uos_id1.sh) with your real HW configs. 
So prefer you try it on v1.6.1 base, and for master branch, there is one known issue to be fixed in coming v2.0 release.  We will also support you in #4820 thread.
Btw, if you have more questions of device config/setup, UUID... we can refer more doc/guide to you,   thanks.

closed 2020-06-28 by fuzhongl