projectacrn/acrn-hypervisor #4856: VSBL report RSA verification fail

lonzoc, 2020-05-29:
Launch android uos on ACRN v1.6.1 will fail at IasImage RSA verification.

The ACRN v1.6.1 release have the cpu sharing enabled by default,  so i modify the launch_uos.sh 
 script of SDC scenario to not offline the sos cpu, after this modification, VSBL fail to boot android uos because of IasImage RSA verifiction error.

[acrn1.6.1-uos-boot-fail-in-cpu-sharing-mode.txt](https://github.com/projectacrn/acrn-hypervisor/files/4699143/acrn1.6.1-uos-boot-fail-in-cpu-sharing-mode.txt)

lonzoc, 2020-05-29:
Before the modification of launch_uos.sh, i can successfully boot android uos, and if i only offline sos cpu1 (the BSP of android uos), i can alse boot the android uos.

fuzhongl, 2022-07-05:
Please using v1.0 for android User vm.

closed 2022-07-05 by fuzhongl