projectacrn/acrn-hypervisor #4864: Hard-coded installation path is being used !!

saininav, 2020-06-01:
Instead of hard-coded installation path, it should be set by variables, which allows the user to set custom path.

For example:
$(DESTDIR)/usr/bin/$(PROGRAM)
should be
$(DESTDIR)$(bindir)/$(PROGRAM)

Where variable can be set, like:
DESTDIR ?=
prefix ?= /usr
bindir ?= $(prefix)/bin

saininav, 2020-06-01:
https://github.com/projectacrn/acrn-hypervisor/pull/4858

fuzhongl, 2020-06-01:
[External_System_ID] ACRN-6404

dbkinder, 2020-06-02:
Have you analyzed what will/might break if the default installation directories are changed by the user?

fuzhongl, 2020-06-05:
@dbkinder  
Following id feedback from @saininav :
**Currently we are using default values set by yocto bitbake.conf and we do not change them while installing.
https://git.yoctoproject.org/cgit/cgit.cgi/poky/tree/meta/conf/bitbake.conf#n18

You can keep below default values of variables in Makefiles.

DESTDIR ?=
prefix ?= /usr
bindir ?= $(prefix)/bin
libdir ?= $(prefix)/lib64
nonarchlibdir ?= $(prefix)/lib
datadir ?= $(prefix)/share
includedir ?= $(prefix)/include
systemd_unitdir ?= $(prefix)/lib/systemd 

We will not change them due to acrnlog.service or other limitations.

Regards,
Naveen**

Will use [this issue](https://github.com/projectacrn/acrn-hypervisor/issues/4892) to remind user to update relate service if he/she modify the default value of installation path.

closed 2020-06-28 by fuzhongl