projectacrn/acrn-hypervisor #4889: [WHL][QEMU][HV] With latest master branch HV, build ACRN for Qemu fail

Leo2Chang, 2020-06-04:
**Environment**
------------------------------------------------------------------------------------------------------
ssg-kvm@HP:acrn-hypervisor$ git status
 On branch master
 Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
 ssg-kvm@HP:acrn-hypervisor$ sync
 ssg-kvm@HP:acrn-hypervisor$ git log
 commit 7e5c2865d71406a0eb34f7648e0cdb036d321590 (HEAD -> master, origin/master, origin/HEAD)

**Steps**
-------------------------------------------------------------------------------------------------------
1. git clone https://github.com/projectacrn/acrn-hypervisor.git
 2. cd acrn-hypervisor
 3. make all BOARD_FILE=./misc/acrn-config/xmls/board-xmls/qemu.xml SCENARIO_FILE=./misc/acrn-config/xmls/config-xmls/qemu/sdc.xml

**Expected result**
-----------------------------------------------------------------------------------------------------
can build HV successfully

**Actual result**
----------------------------------------------------------------------------------------------------
find error: 
……
Error: Validate the scenario item failure
 Error: hv,FEATURES,RDT,RDT_ENABLED: RDT_ENABLED should not empty
 Error: hv,FEATURES,RDT,CDP_ENABLED: CDP_ENABLED should not empty
……
detail log as attachment
[qemuACRN2_buildFail.txt](https://github.com/projectacrn/acrn-hypervisor/files/4728534/qemuACRN2_buildFail.txt)

liuwei987, 2020-06-04:
[External_System_ID] ACRN-6411

Leo2Chang, 2020-06-05:
with latest master, can build ACRN for Qemu successfully. 
close as fixed. 
Thanks

closed 2020-06-05 by Leo2Chang