projectacrn/acrn-hypervisor #4898: acrnctl: Failed to check directory /usr/share/acrn/conf, err: -1

haribabug, 2020-06-05:
The 'acrnctl' commands start/stop/reset are not working as expected, and also the path "/usr/share/acrn/conf doesn't exist!".  Please check and provide the details to resolve it.

Looks like below commit has some change related to it.
https://github.com/projectacrn/acrn-hypervisor/commit/e50c0c88#diff-5cf3490167207036b05ef855d48a2d48L231

Here provided the logs for the ' acrnctl reset', observed continuous error logs on both SOS and UOS console.

root@intel-corei7-64:/# acrnctl list
/usr/share/acrn/conf doesn't exist!
_scan_added_vm: Failed to check directory /usr/share/acrn/conf, err: -1
preempt_rt_linux_vm1            started

//Logs from SOS console
root@intel-corei7-64:/# acrnctl reset preempt_rt_linux_vm1
/usr/share/acrn/conf doesn't exist!
_scan_added_vm: Failed to check directory /usr/share/acrn/conf, err: -1
/usr/share/acrn/conf doesn't exist!
...
...
_scan_added_vm: Failed to check directory /usr/share/acrn/conf, err: -1
/usr/share/acrn/conf doesn't exist!
_scan_added_vm: Failed to check directory /usr/share/acrn/conf, err: -1
Failed to stop preempt_rt_linux_vm1 in 30 sec, reset failed

//Logs from UOS console on triggering 'acrnctl reset preempt_rt_linux_vm1' at SOS 
Connected:41
            Disconnect(41)!
Connected:52
            pm-vuart stop handler called: node-index=1
                                                      Disconnect(52)!
Connected:41
            Disconnect(41)!
Connected:52
            Disconnect(52)!
genirq: Flags mismatch irq 3. 00000000 (ttyS1) vs. 00000000 (virtio0-control-i)
genirq: Flags mismatch irq 4. 00000000 (ttyS0) vs. 00000000 (virtio0-input)
genirq: Flags mismatch irq 3. 00000000 (ttyS1) vs. 00000000 (virtio0-control-i)
genirq: Flags mismatch irq 4. 00000000 (ttyS0) vs. 00000000 (virtio0-input)
genirq: Flags mismatch irq 3. 00000000 (ttyS1) vs. 00000000 (virtio0-control-i)
genirq: Flags mismatch irq 4. 00000000 (ttyS0) vs. 00000000 (virtio0-input)
genirq: Flags mismatch irq 4. 00000000 (ttyS0) vs. 00000000 (virtio0-input)
genirq: Flags mismatch irq 4. 00000000 (ttyS0) vs. 00000000 (virtio0-input)
genirq: Flags mismatch irq 3. 00000000 (ttyS1) vs. 00000000 (virtio0-control-i)
genirq: Flags mismatch irq 3. 00000000 (ttyS1) vs. 00000000 (virtio0-control-i)
Connected:41
            Disconnect(41)!
Connected:52
            Disconnect(52)!
Can not receive acked message from uos, have try 3 times
Connected:41
            Disconnect(41)!
Connected:52
            Disconnect(52)!

fuzhongl, 2020-06-11:
@haribabug Please help to try the direction of Wenling:
`Please add “SERIAL_CONSOLES_pn-systemd-serialgetty = "115200;ttyS0” when build SOS rootfs. This parameter is used for serial consoles there are not be occupied.`
Thanks!

vmurthy-intel, 2020-08-19:
@fuzhongl as you suggested used “SERIAL_CONSOLES_pn-systemd-serialgetty = "115200;ttyS0” when building SOS. But still no luck. I still see the same error:

root@intel-corei7-64:/# acrnctl list
/usr/share/acrn/conf doesn't exist!
_scan_added_vm: Failed to check directory /usr/share/acrn/conf, err: -1

gvancuts, 2020-08-19:
As a side note, I also see these errors on my system running Ubuntu 20.04 and acrn 2.0:
```
/usr/share/acrn/conf doesn't exist!
_scan_added_vm: Failed to check directory /usr/share/acrn/conf, err: -1
```
But if I have a VM running, it will be listed below. I'm therefore not too sure what to make of this error, is it fatal or can be ignored?

ionutnechita, 2021-03-22:
I noticed this:
/usr/share/acrn/conf doesn't exist!
_scan_added_vm: Failed to check directory /usr/share/acrn/conf, err: -1

Shouldn't the acrnd service do this folder? ( mkdir -p /usr/share/acrn/conf )
Or from installing hypervisor through makefiles.

gvancuts, 2021-03-22:
The `acrnctl` does create this when running (see [here](https://github.com/projectacrn/acrn-hypervisor/blob/master/misc/services/acrn_manager/acrnctl.c#L349-L355)... except, it only creates it when running `acrnctl add`, i.e. when first adding a VM (its launch script).

I think it would be better to move this directory creation to the `main` function ([here](https://github.com/projectacrn/acrn-hypervisor/blob/master/misc/services/acrn_manager/acrnctl.c#L765)).

gvancuts, 2021-03-23:
[External_System_ID] ACRN-7031

gvancuts, 2021-03-25:
@haribabug , can you tell us whether the merge of #5855 solves your problems or you see more issues?

gvancuts, 2022-04-25:
Patch was merged, closing.

closed 2022-04-25 by gvancuts