projectacrn/acrn-hypervisor #4900: ACRN Config tool: "acrn-dm" line is missing from the launch script when generated for 3 or more VMs

vmurthy-intel, 2020-06-05:
Issue: In ACRN Release_1.6; after running the “launch_cfg_gen.py” to generate launch scripts for 3 or more VMs, the generated launch script for VM3 and above will have “acrn-dm” line missing from it. Please check the attached [launch_uos_id3.txt](https://github.com/projectacrn/acrn-hypervisor/files/4738510/launch_uos_id3.txt) launch script file

My initial analysis:  
This issue is existing in the branch release_1.6 of ACRN.
However, the issue was fixed in the master branch with commit-ID: “8ebf89eaa813082e9df39f3e8a1764dd503b8bc9” on 15th of May 2020.
We need this commit in release_1.6 to fix this issue.

fuzhongl, 2020-06-28:
@yaomichael  Any direction about porting this patch to release_1.6?
Thanks!

fuzhongl, 2021-03-15:
fixed in master branch.

closed 2021-03-15 by fuzhongl