projectacrn/acrn-hypervisor #4904: can't find qemu.xml when  build acrn for qemu-kvm

duhongyu, 2020-06-08:
I build acn follew this guide 
[(https://projectacrn.github.io/latest/tutorials/acrn_on_qemu.html?u=6500542303&m=4512490987818461&cu=1321892844)]
$ git clone https://github.com/projectacrn/acrn-hypervisor.git
$ cd acrn-hypervisor/
$ git fetch --all --tags --prune
$ git checkout tags/acrn-2020w19.5-140000p -b acrn_on_qemu
but face some error 
acrn@ubuntu:~/acrn-hypervisor$ make all BOARD_FILE=./misc/acrn-config/xmls/board-xmls/qemu.xml  SCENARIO_FILE=./misc/acrn-config/xmls/config-xmls/qemu/sdc.xml
Makefile:57: *** BOARD_FILE: ./misc/acrn-config/xmls/board-xmls/qemu.xml does not exist.  Stop.

ttzeng, 2020-06-08:
@duhongyu , the qemu patch was merged after `acrn-2020w19.5-140000p` tag, so you may try the master branch or `acrn-2020w21.5-180000p` or newer code.

duhongyu, 2020-06-08:
@ttzeng thanks for your reply， I have checkout to acrn-2020w19.5-140000p tag
$ git checkout tags/acrn-2020w19.5-140000p -b acrn_on_qemu

fuzhongl, 2020-07-02:
@duhongyu please use release 2.0
Thanks!

closed 2020-07-02 by fuzhongl