projectacrn/acrn-hypervisor #4940: `make install` does not install `acrnctl` on the system

gvancuts, 2020-06-15:
If you build and install ACRN on a system, the `acrnctl` binary is not installed under `/usr/bin` as one would expect.

Steps to reproduce this issue: go through the steps described [here](https://projectacrn.github.io/latest/getting-started/rt_industry_ubuntu.html#build-and-install-acrn-on-ubuntu). Then verify that there is no `/usr/bin/acrnctl` file after running the `sudo make install` steps (`acrn-dm` is correctly installed however).

gvancuts, 2020-06-16:
I suspect the problem is the conditional statement around the `install` directive [here](https://github.com/projectacrn/acrn-hypervisor/blob/master/misc/acrn-manager/Makefile#L110-L112). `RELEASE` is no longer defined when you run `sudo make install`.

gvancuts, 2020-06-16:
@jsun26intel is this one for you?

/cc @NanlinXie

fuzhongl, 2020-06-29:
@gvancuts It works as design, install acrnctl for debug version only; not install acrnctl for release version.

gvancuts, 2020-06-29:
> @gvancuts It works as design, install acrnctl for debug version only; not install acrnctl for release version.

The problem is that the following sequence does not install `acrnctl` (taking the example straight from our Getting Started Guide):
```
$ make all BOARD_FILE=misc/acrn-config/xmls/board-xmls/whl-ipc-i7.xml SCENARIO_FILE=misc/acrn-config/xmls/config-xmls/whl-ipc-i7/industry.xml RELEASE=0
$ sudo make install
```
So although one is building the debug version of ACRN, `acrnctl` is built as expected during the build process. But the `RELEASE=0` is lost when you run `sudo make install` and hence `acrnctl` is not installed on the rootfs.

gvancuts, 2020-09-03:
Based on the conversation on the mailing list (see [here](https://lists.projectacrn.org/g/acrn-dev/message/28820?p=,,,20,0,0,0::Created,,acrnctl,20,2,0,75276176)), the right course of action is to build and install `acrnctl` for both release and debug mode. Kata Containers needs it in order to successfully spin off a Kata VM.

I will prepare a patch for this.

fuzhongl, 2020-09-04:
[External_System_ID]ACRN-6679

closed 2021-01-06 by fuzhongl