projectacrn/acrn-hypervisor #4945: Fix macvtap network interface in acrn-dm

vijaydhanraj, 2020-06-15:
There are 2 issues with enabling macvtap support in acrn-dm,

1. There is an explicit interface name check for tapX preventing any other interface name like kata_tap to be setup.
2. For macvtap interface, one needs to directly open the character device /dev/tapXX rather than /dev/net/tun as in case of tap/bridge interface.
3. Add support to identify macvtap vs tap interface and if it is macvtap, identify the character device (/dev/tapXX)

This issue is created to fix the above issues and enable macvtap network interface support in acrn-dm.

vijaydhanraj, 2020-06-15:
[External_System_ID] ACRN-6456

ttzeng, 2020-06-22:
Works for me, thanks @vijaydhanraj for the resolution!

fuzhongl, 2020-06-28:
Created a fix and this has been validated by the customer as well. The patch is in the mailing list for review.

closed 2022-07-05 by fuzhongl