projectacrn/acrn-hypervisor #4948: Starting a Kata Container on ACRN takes a very long time

gvancuts, 2020-06-16:
Starting a Kata Container (via Docker)* takes a rather long time, around 15 sec.

* Following this installation guide: https://projectacrn.github.io/latest/tutorials/run_kata_containers.html

fuzhongl, 2020-06-28:
@szhen11 Please help to check Kata Container starting time.
Thanks!

szhen11, 2020-06-28:
Kata VM boot up time is about 15 seconds. @gvancuts is this an issue for ACRN?

gvancuts, 2020-06-29:
It's a very long time for a container to start, even for a VM actually. We're usually looking at a sub-second start time for a container. Do we understand why it's taking so long?

In y our test system, how long does it take to boot a normal Linux User VM?

gvancuts, 2020-12-01:
I'm seeing even longer boot times lately, more like a minute and sometimes even longer:
```
$ date && sudo docker run -ti --runtime=kata-acrn busybox sh -c "echo SUCCESS && exit" && date
Di 01 Dez 2020 09:46:26 CET
[sudo] password for acrn: 
SUCCESS
Di 01 Dez 2020 09:47:11 CET
```

`DM version is: 2.3-unstable-79892f91-dirty (daily tag:acrn-2020w45.1-180000p), build by acrn@2020-11-30 12:28:20`

Are you seeing this too in your test environment?

@vijaydhanraj , FYI

gvancuts, 2021-01-29:
Still facing this issue with the latest code:

```
$ date && sudo docker run -ti --runtime=kata-acrn busybox sh -c "echo SUCCESS && exit" && date
Fr 29 Jan 2021 15:49:17 CET
[sudo] password for acrn: 
SUCCESS
Fr 29 Jan 2021 15:50:30 CET
```
```
$ acrn-dm --version
DM version is: 2.4-unstable-c9ca23d2 (daily tag:acrn-2021w04.3-180000p), build by acrn@2021-01-28 16:54:53
```

vijaydhanraj, 2021-01-29:
Adding @gvancuts old analysis here.
•	After a clean boot: starting a Kata Container (busybox) takes even more time than I reported earlier, about 45 sec. About 15 secs are spent by the acrn-dm to start the VM, and it takes close to 30 secs before I see the acrn-dm process started.
•	A second start of the same Kata Container (busybox, previously shut down using ‘exit’ from the prompt) takes about 13sec, with just one or two seconds before the acrn-dm is started

@gvancuts is this analysis closer to what you observe now or has it become even worse? Based on the above analysis, I see that Kata launch seems to be the bottleneck. Do you see the same issue even with `Kata+QEMU`

gvancuts, 2021-02-01:
> @gvancuts is this analysis closer to what you observe now or has it become even worse? Based on the above analysis, I see that Kata launch seems to be the bottleneck. Do you see the same issue even with `Kata+QEMU`

I do actually. It takes more than a minute the first time I run Kata and subsequent launches take much less time. Is this a known issue (or behaviour)?

```
acrn@acrn-ubuntu-svm:~$ date && sudo docker run -ti --runtime=kata-qemu busybox sh -c "echo SUCCESS && exit" && date
Mo 01 Feb 2021 19:44:19 CET
[sudo] password for acrn: 
SUCCESS
Mo 01 Feb 2021 19:45:41 CET
acrn@acrn-ubuntu-svm:~$ date && sudo docker run -ti --runtime=kata-qemu busybox sh -c "echo SUCCESS && exit" && date
Mo 01 Feb 2021 19:45:55 CET
SUCCESS
Mo 01 Feb 2021 19:45:56 CET
acrn@acrn-ubuntu-svm:~$ date && sudo docker run -ti --runtime=kata-qemu busybox sh -c "echo SUCCESS && exit" && date
Mo 01 Feb 2021 19:45:58 CET
SUCCESS
Mo 01 Feb 2021 19:45:59 CET
acrn@acrn-ubuntu-svm:~$ date && sudo docker run -ti --runtime=kata-qemu busybox sh -c "echo SUCCESS && exit" && date
Mo 01 Feb 2021 19:46:00 CET
SUCCESS
Mo 01 Feb 2021 19:46:02 CET
```

vijaydhanraj, 2021-02-01:
Thanks, @gvancuts for checking. 
A quick glance at kata issues brought this up, 
1. https://github.com/kata-containers/runtime/issues/1025 (this is little old)
2. https://github.com/kata-containers/runtime/issues/2692. But this is with virtio-fs.

It might be a good idea to raise another issue with kata team with the recent data since we do have some indicators pointing to Kata when using either hypervisor or even update issue  [1025](https://github.com/kata-containers/runtime/issues/1025) add our new data to get the attention of kata team.

gvancuts, 2022-06-28:
Where do we stand on bringing back support for Kata and ACRN? @vijaydhanraj

vijaydhanraj, 2022-06-29:
I have kata-2.3.0-rc0 working with ACRN2.7 + some patches. I have shared the patches with the PRC team for further evaluation. My next will be to,
1. Rebase to latest Kata, as currently, I am on 2.3.0-rc0
2. Rebase ACRN to latest master.
3. Implement the fix for container rootfs mount (currently I have a hacky implementation).
4. Validation and testing before final PR.