projectacrn/acrn-hypervisor #4975: Ubuntu `snap` not working correctly with ACRN Service VM (SOS) kernel

gvancuts, 2020-06-22:
Some snap SW packages are broken when running under the ACRN Service VM (SOS) kernel. They will fail throwing an error such as:
```
cmd_run.go:918: WARNING: cannot create user data directory: failed to verify SELinux context of /root/snap: exec: "matchpathcon": executable file not found in $PATH
```

Steps to reproduce:
* Follow the ACRN installation guide: https://projectacrn.github.io/latest/getting-started/rt_industry_ubuntu.html
* Install `microk8s` following the general instructions here: https://microk8s.io/ (essentially `sudo snap install microk8s --classic`)
* Then run `microk8s inspect`, you will see something like:
```
Inspecting Certificates
Inspecting services
  Service snap.microk8s.daemon-cluster-agent is running
  Service snap.microk8s.daemon-flanneld is running
  Service snap.microk8s.daemon-containerd is running
  Service snap.microk8s.daemon-apiserver is running
  Service snap.microk8s.daemon-apiserver-kicker is running
  Service snap.microk8s.daemon-proxy is running
ESC[31m FAIL: ESC[0m Service snap.microk8s.daemon-kubelet is not running
For more details look at: sudo journalctl -u snap.microk8s.daemon-kubelet
  Service snap.microk8s.daemon-scheduler is running
  Service snap.microk8s.daemon-controller-manager is running
  Service snap.microk8s.daemon-etcd is running
  Copy service arguments to the final report tarball
[snip]
```
* `sudo journalctl -u snap.microk8s.daemon-kubelet` returns:
```
- Logs begin at Thu 2020-06-11 16:18:53 CEST, end at Mon 2020-06-22 18:02:38 CEST. --
Jun 16 13:42:15 acrn-ubuntu-svm systemd[1]: Started Service for snap application microk8s.daemon-kubelet.
Jun 16 13:42:17 acrn-ubuntu-svm microk8s.daemon-kubelet[14407]: cmd_run.go:918: WARNING: cannot create user data directory: failed to verify SELinux context of /root/snap: exec: "matchpathcon": executab>
Jun 16 13:42:17 acrn-ubuntu-svm microk8s.daemon-kubelet[14407]: + export PATH=/snap/microk8s/1422/usr/sbin:/snap/microk8s/1422/usr/bin:/snap/microk8s/1422/sbin:/snap/microk8s/1422/bin:/usr/local/sbin:/u>
Jun 16 13:42:17 acrn-ubuntu-svm microk8s.daemon-kubelet[14407]: + PATH=/snap/microk8s/1422/usr/sbin:/snap/microk8s/1422/usr/bin:/snap/microk8s/1422/sbin:/snap/microk8s/1422/bin:/usr/local/sbin:/usr/loca>
Jun 16 13:42:17 acrn-ubuntu-svm microk8s.daemon-kubelet[14575]: ++ /snap/microk8s/1422/bin/uname -m
Jun 16 13:42:17 acrn-ubuntu-svm microk8s.daemon-kubelet[14407]: + ARCH=x86_64
Jun 16 13:42:17 acrn-ubuntu-svm microk8s.daemon-kubelet[14407]: + export LD_LIBRARY_PATH=:/snap/microk8s/1422/lib:/snap/microk8s/1422/usr/lib:/snap/microk8s/1422/lib/x86_64-linux-gnu:/snap/microk8s/1422>
Jun 16 13:42:17 acrn-ubuntu-svm microk8s.daemon-kubelet[14407]: + LD_LIBRARY_PATH=:/snap/microk8s/1422/lib:/snap/microk8s/1422/usr/lib:/snap/microk8s/1422/lib/x86_64-linux-gnu:/snap/microk8s/1422/usr/li>
Jun 16 13:42:17 acrn-ubuntu-svm microk8s.daemon-kubelet[14407]: + export LD_LIBRARY_PATH=/var/lib/snapd/lib/gl:/var/lib/snapd/lib/gl32:/var/lib/snapd/void::/snap/microk8s/1422/lib:/snap/microk8s/1422/us>
Jun 16 13:42:17 acrn-ubuntu-svm microk8s.daemon-kubelet[14407]: + LD_LIBRARY_PATH=/var/lib/snapd/lib/gl:/var/lib/snapd/lib/gl32:/var/lib/snapd/void::/snap/microk8s/1422/lib:/snap/microk8s/1422/usr/lib:/>
Jun 16 13:42:17 acrn-ubuntu-svm microk8s.daemon-kubelet[14407]: + source /snap/microk8s/1422/actions/common/utils.sh
Jun 16 13:42:17 acrn-ubuntu-svm microk8s.daemon-kubelet[14407]: + app=kubelet
Jun 16 13:42:17 acrn-ubuntu-svm microk8s.daemon-kubelet[14407]: + '[' -e /var/snap/microk8s/1422/var/lock/clustered.lock ']'
Jun 16 13:42:17 acrn-ubuntu-svm microk8s.daemon-kubelet[14407]: + '[' kubelet = kubelet ']'
Jun 16 13:42:17 acrn-ubuntu-svm microk8s.daemon-kubelet[14583]: ++ cat /var/snap/microk8s/1422/args/kubelet
[snip]
```

gvancuts, 2020-06-22:
Upon investigating this, it appears to be due to SELinux being configured in the ACRN Service VM (SOS) kernel and used by default.

I changed the `.config` file to use AppArmor instead, done via `make menuconfig`, leading to this diff:
```
6272d6271
< CONFIG_SECURITY_WRITABLE_HOOKS=y
6277,6278c6276
< # CONFIG_SECURITY_PATH is not set
< CONFIG_LSM_MMAP_MIN_ADDR=65536
---
> CONFIG_SECURITY_PATH=y
6285,6290c6283
< CONFIG_SECURITY_SELINUX=y
< CONFIG_SECURITY_SELINUX_BOOTPARAM=y
< CONFIG_SECURITY_SELINUX_DISABLE=y
< CONFIG_SECURITY_SELINUX_DEVELOP=y
< CONFIG_SECURITY_SELINUX_AVC_STATS=y
< CONFIG_SECURITY_SELINUX_CHECKREQPROT_VALUE=1
---
> # CONFIG_SECURITY_SELINUX is not set
6293c6286,6289
< # CONFIG_SECURITY_APPARMOR is not set
---
> CONFIG_SECURITY_APPARMOR=y
> CONFIG_SECURITY_APPARMOR_HASH=y
> CONFIG_SECURITY_APPARMOR_HASH_DEFAULT=y
> # CONFIG_SECURITY_APPARMOR_DEBUG is not set
6299c6295
< CONFIG_DEFAULT_SECURITY_SELINUX=y
---
> CONFIG_DEFAULT_SECURITY_APPARMOR=y
```

I then recompiled the kernel and installed it. The issue is **not** seen anymore.

fuzhongl, 2020-06-28:
@shuox  Could you please help to check the sos kernel config?
Thanks!

shuox, 2020-06-28:
@fuzhongl @gvancuts Is coverting from selinux to apparmor the expected fix for running microk8s in SOS?

gvancuts, 2020-06-30:
> @fuzhongl @gvancuts Is coverting from selinux to apparmor the expected fix for running microk8s in SOS?

Yes, but only for Ubuntu. Other Service VM OS may use/need different security mechanism. I can propose a patch to the `acrn-kernel` repo to help fix this.

wenlingz, 2020-07-10:
please add JIRA ID,thanks

wenlingz, 2020-07-21:
@gvancuts  please add JIRA ID,thanks

fuzhongl, 2020-07-21:
[External_System_ID] ACRN-6563

closed 2021-03-15 by fuzhongl