projectacrn/acrn-hypervisor #4980: graphics not working in UserVM on NUC6CAY

hidrol, 2020-06-23:
I followed this guide https://projectacrn.github.io/latest/tutorials/running_ubun_as_user_vm.html in order to run Ubuntu as UserVM. After launching the script launch_ubuntu.sh I get:

```
gvt: vgpu 1: Unsupport pipe 0, scaler 1 scaling
gvt: vgpu 1: Unexpected ME_SEMAPHORE_WAIT handler error
gvt: vgpu 1: ME_SEMAPHORE_WAIT HANDLER error
gvt: vgpz 1: cmd parser error
0x0
0x1c

gvt: vgpu 1: scan error 
GVT Internal error for the guest
Now vgpu 1 will enter failsafe mode.
gvt: vgpu 1: failed to submit desc 0
gvt: vgpu 1: fail submit workload on ring 0
gvt: vgpu 1: fail to emulate MMIO write 00002230 len4
gvt: failed to write mmio at addr 0xa0002230
```

gvancuts, 2020-06-23:
Can you confirm which version of ACRN you are on and the Service VM (SOS) kernel parameters in use? (from the service VM: `cat /proc/cmdline`)

hidrol, 2020-06-23:
I use ACRN version 1.6. The kernel parameters are:
`initrd=\EFI\org.clearlinux\freestanding-00-intel-ucode.cpio i915.enable_conformance_check=0 initrd=\EFI\org.clearlinux\freestanding-i915-firmware.cpio.xz root=PARTUUID=c8e3c781-10f9-4a8a-9923-3fc21f69d1c1 quiet console=tty0 console=ttyS0,115200n8 consoleblank=0 cryptomgr.notests hvlog=2M@0xE00000 i915.avail_planes_per_pipe=0x010101 i915.domain_plane_owners=0x011100001111 i915.enable_guc=0 i915.enable_gvt=1 i915.nuclear_pageflip=1 ignore_loglevel init=/usr/lib/systemd/systemd-bootchart intel_iommu=igfx_off memmap=2M$0x1FE00000 no_timer_check no_timer_check noreplace-smp rcu_nocbs=0-64 rcupdate.rcu_expedited=1 rootfstype=ext4,btrfs,xfs rootwait tsc=reliable rw`

ttzeng, 2020-06-24:
@hidrol can you check whether there is errors in `dmesg` output? As I recall, there is an exception in the kernel driver while running on NUC6CAYH, not sure that has been addressed.

hidrol, 2020-06-24:
When I filter dmesg errors I get:

```
[    1.406242] intel_pmc_ipc 0000:00:0d.1: can't enable device: BAR 0 [mem size 0x00002000 64bit] not assigned
[    3.709864] systemd[1]: Failed to find module 'autofs4'
[   10.119549] snd_soc_skl 0000:00:0e.0: No matching machine driver found
[  224.176079] gvt: vgpu 1: Unsupport pipe 0, scaler 1 scaling
[  224.183711] gvt: vgpu 1: Unsupport pipe 0, scaler 1 scaling
[  224.192865] gvt: vgpu 1: Unsupport pipe 0, scaler 1 scaling
[  224.266712] gvt: vgpu 1: Unexpected ME_SEMAPHORE_WAIT in command buffer!
[  224.274601] gvt: vgpu 1: ME_SEMAPHORE_WAIT handler error
[  224.283493] gvt: vgpu 1: cmd parser error
[  224.291663] 0x0 
[  224.291665] 0x1c 

[  224.314404] gvt: vgpu 1: scan workload error
[  224.322717] GVT Internal error  for the guest
[  224.330845] Now vgpu 1 will enter failsafe mode.
[  224.338886] gvt: vgpu 1: failed to submit desc 0
[  224.347133] gvt: vgpu 1: fail submit workload on ring 0
[  224.355510] gvt: vgpu 1: fail to emulate MMIO write 00002230 len 4
[  224.363858] gvt: failed to write mmio at addr 0xa0002230
```

fuzhongl, 2020-06-28:
@hidrol  What's difference with issue 4981?
Please use acrn-kernel as sos kernel.

hidrol, 2020-06-29:
@fuzhongl the differnce to issue 4981 is that in contrast to debian, gnome won't start, even though I use the acrn-kernel now.

gvancuts, 2020-06-30:
Could you try with this kernel parameter instead: `i915.domain_plane_owners=0x011111110000`?

hidrol, 2020-06-30:
> Could you try with this kernel parameter instead: `i915.domain_plane_owners=0x011111110000`?
@gvancuts I did try, unfortunately I get the same error messages and ubuntu won't boot.

fuzhongl, 2020-07-02:
@hidrol 
Which platform are you using? Any error log from HV console?
Could you please share your launch script? or remove gvt-g parameter to check if ubuntu boots up.

`-s 2,pci-gvt -G "$2" `

fuzhongl, 2020-07-29:
@hidrol Any progress ?

hidrol, 2020-07-30:
@fuzhongl I needed a graphical uos for demonstration purposes, so I was happy that I got debian working, and didn't further try with ubuntu.

fuzhongl, 2020-07-30:
@hidrol Thanks for feedback. Could you please close this issue?

closed 2020-07-30 by hidrol