projectacrn/acrn-hypervisor #4981: Invisible Cursor on Debian 10.4 as UserVM [NUC6CAY]

hidrol, 2020-06-24:
The mouse Cursor is working but invisible, besides that I get some minor graphical glitches on the desktop.
I use ACRN Version 1.6 and the kernel parameters are:

`initrd=\EFI\org.clearlinux\freestanding-00-intel-ucode.cpio i915.enable_conformance_check=0 initrd=\EFI\org.clearlinux\freestanding-i915-firmware.cpio.xz root=PARTUUID=c8e3c781-10f9-4a8a-9923-3fc21f69d1c1 quiet console=tty0 console=ttyS0,115200n8 consoleblank=0 cryptomgr.notests hvlog=2M@0xE00000 i915.avail_planes_per_pipe=0x010101 i915.domain_plane_owners=0x011100001111 i915.enable_guc=0 i915.enable_gvt=1 i915.nuclear_pageflip=1 ignore_loglevel init=/usr/lib/systemd/systemd-bootchart intel_iommu=igfx_off memmap=2M$0x1FE00000 no_timer_check no_timer_check noreplace-smp rcu_nocbs=0-64 rcupdate.rcu_expedited=1 rootfstype=ext4,btrfs,xfs rootwait tsc=reliable rw`

gvancuts, 2020-06-24:
@ttzeng , is this the same kernel crash you previously saw on APL?

ttzeng, 2020-06-25:
@gvancuts from the above log, it looks like @hidrol is using the stock kernel 4.19.120-108. Though it's different than my home-made 5.4 kernel, they are all trap at the driver `intel_display.c` but at different location.

hidrol, 2020-06-25:
@ttzeng is there a workaround to fix this issue?

gvancuts, 2020-06-26:
> @ttzeng is there a workaround to fix this issue?

Not sure if it will fix this but you can try to use the [acrn-kernel](https://github.com/projectacrn/acrn-kernel). The quick version of how to build it is:
```
$ git clone https://github.com/projectacrn/acrn-kernel
$ cd acrn-kernel
$ git checkout v1.6.1
$ cp kernel_config_uefi_sos .config
$ make olddefconfig
$ make all
$ sudo make modules_install # assuming you are building this kernel directly inside the Service VM
```
And then install the new kernel and update Grub to use it.

hidrol, 2020-06-26:
I forgot to mention that I use Clear Linux as Service VM, so I think I can't use the acrn-kernel.

gvancuts, 2020-06-26:
> I forgot to mention that I use Clear Linux as Service VM, does Clear Linux work with acrn-kernel?

Yes. but in that case you'll need to adjust the `acrn.conf` file from `systemd-boot` to boot this new kernel

fuzhongl, 2020-06-28:
@hidrol Just like @gvancuts said, please use acrn-kernel as sos kernel. It will fix mouse cursor issue.

hidrol, 2020-06-28:
Yes that did the job. Thanks to your support @gvancuts and @fuzhongl

closed 2020-06-28 by hidrol