projectacrn/acrn-hypervisor #4982: ivshmemTest transfer file failed after UOS shutdown or reboot

fuzhongl, 2020-06-24:
Environment:
------------
HW: WHL

HV: https://github.com/projectacrn/acrn-hypervisor/commit/e3fe3da2732682cbb53e79675851ab33c54b7d8b
KERNEL: https://github.com/projectacrn/acrn-kernel/commit/80ea97f638a5efc362ce828f8994d6293baac2c7

SOS ROOTFS:
http://sit-image.sh.intel.com/yocto/e2e/ww23.5/acrn-image-weston-intel-corei7-64.wic.acrn

UOS: http://10.239.147.147/yocto/e2e/ww24.2/

 

Bug detailed description:
-------------

ivshmemTest transfer file failed after one UOS shutdown or reboot

Reproduce steps:
-------------
1. Boot SOS and HV
2.Add the ivshmem device for acrn-dm boot parameters for both YaaG1 and YaaG2
   -s 8,ivshmem,shm-test,33554432 \ （mem-size range: 4k-128M , and should be the power of 2) \
3. in UOS lspci to check PCI device:
00:04.0 RAM memory: Red Hat, Inc. Inter-VM shared memory (rev 01)
4, check shm in SOS under /dev/shm
$ root@intel-corei7-64:~# ls /dev/shm/
shm-test
5. load UIO driver in UOS with cmd below:
$modprobe uio
$modprobe uio_pci_generic
$echo "1af4 1110" > /sys/bus/pci/drivers/uio_pci_generic/new_id
6. can find uio0 in UOS /dev/uio0
7.Using ivshmemTest for shared memory transfer verification, ivshmem location(http://10.238.145.49/000/inter-vm/ivshmemTest)
Input ivshmemTest uio0 tx a.img in UOS1
Input ivshmemTest uio0 rx b.img in UOS2
8. Check the shared memory creation/removal
Input df -h in SOS
Check “tmpfs 7.8G 4.3M 7.8G 1% /dev/shm”
9. shutdown YaaG1 and check shm-test disappear in SOS /dev/shm/
10, Boot up YaaG1 again and load uio driver again:
11. transfer file via ivshmem between UOS1 and UOS2

12. reboot YaaG1 and loader uio driver, thransfer file with UOS2

Current result:
--------------
Transfer file failed after reboot or shutdown and boot YaaG1 again;

YaaG2 can't receive file successfully

Expected result:
--------------
Transfer file successfully after shutdown and boot YaaG1 or reboot YaaG1 

console log:

first boot:

df -h
Filesystem Size Used Avail Use% Mounted on
tmpfs 3.7G 32M 3.7G 1% /dev/shm

after YaaG1 reboot:
df -h
Filesystem Size Used Avail Use% Mounted on
tmpfs 3.7G 64M 3.7G 2% /dev/shm

UOS1:

root@intel-corei7-64:~# ivshmemTest uio0 tx test.file
Mmap /sys/class/uio/uio0/device/resource2, the size is 33554432
Image test.file md5sum is 20616e51c7df0165df595ffd05460099
Prepare to send to test.file file, size 25000000
Source image md5sum is 20616e51c7df0165df595ffd05460099, len 32
Send done, total size 25000000, cost 7 secs

UOS2:

root@intel-corei7-64:~# ivshmemTest uio0 rx test.file
Mmap /sys/class/uio/uio0/device/resource2, the size is 33554432
Prepare to recv from test.file file, size 25000000
Target image md5sum is 20616e51c7df0165df595ffd05460099, len 32

fuzhongl, 2020-06-24:
This issue is fixed on the master branch with following PR:

https://github.com/projectacrn/acrn-hypervisor/pull/4955

Commit ID: c7e527f2efd7324644683619899a3a32d3c1ad67

 For acrn 2.0 release.

we have two limitations in here.

1. It can't support post-launched VMs reboot, it only supports the whole platform reboot.

2. If one Post-Launched VM shut down, another VM needs to  shut down too, or the whole platform shutdown.

fuzhongl, 2020-06-28:
dup to 4959

closed 2020-06-28 by fuzhongl

SIMING29, 2021-04-17:
Hi @fuzhongl  
> For acrn 2.0 release.
> 
> we have two limitations in here.
> 
> 1. It can't support post-launched VMs reboot, it only supports the whole platform reboot.
> 2. If one Post-Launched VM shut down, another VM needs to  shut down too, or the whole platform shutdown.

Are there these limitations still exist in ACRN 2.3?

> 4\. 7.Using ivshmemTest for shared memory transfer verification, ivshmem location(http://10.238.145.49/000/inter-vm/ivshmemTest)
>     Input ivshmemTest uio0 tx a.img in UOS1
>     Input ivshmemTest uio0 rx b.img in UOS2

Can you share the test application? Or give some clues about ivshmem test programming, like where can I find the user-space driver and api
Thanks.

fuzhongl, 2021-04-19:
@zhangshuai75 

This issue is fixed in v2.1.
Please refer this [link](https://projectacrn.github.io/latest/tutorials/enable_ivshmem.html) about ivshmemTest.

zhangshuai75, 2021-04-19:
@fuzhongl 
Thanks, I have set the acrn according to the link above, but I'm not sure how to write the application to test
please check this issue #5939

fuzhongl, 2021-04-19:
@NanlinXie 
Any comments about ivshmemTest?

zhangshuai75, 2021-04-23:
Hi @fuzhongl  Any update?