projectacrn/acrn-hypervisor #4990: can't acrn over qemu/kvm in lenovo pc(cpu I5-6300U)

duhongyu, 2020-06-29:
when I ruun acrn over qemu/kvm in lenovo pc(cpu I5-6300U), 
[cpu=2][vm0:vcpu2][sev=3][seq=46]:VMX ctrl 0x486 not fully enabled: request 0x4ab but get 0xab

gvancuts, 2020-06-29:
@vijaydhanraj is this something you can help with?

That CPU is from the Sky Lake family (SKL), we can run ACRN on this as far as I know.

vijaydhanraj, 2020-06-30:
The above doesn't seem like an error but an info message. What this means is that ACRN is trying to set pause-loop exiting VM-execution control but since the emulated processor doesn't that support “PAUSE-loop exiting” feature, it is not able to set it.

Even with this, you should be able to enter ACRN console if you hit "enter key" and from there should be able to launch guest VM using ``vm_console "vm_id"`` command. If not there is something else going wrong.

duhongyu, 2020-07-02:
thanks  @gvancuts @vijaydhanraj , I has entered ACRNSOS console by using `vm_console` .

closed 2021-03-15 by fuzhongl