projectacrn/acrn-hypervisor #4997: [Improvement] Add GDB server support for UOS debugging

tbursztyka, 2020-06-30:
ACRN hypervisor does not provide a way to get a gdb server on a UOS and that is a blocker to easily adapt OS as UOS for ACRN. For instance, there are issues within Zephyr on ACRN which are hard to debug with only a console output, not impossible but really tedious. (see https://github.com/zephyrproject-rtos/zephyr/issues/23802 and https://github.com/zephyrproject-rtos/zephyr/issues/23801)

I have no idea how difficult it is to add such feature in ACRN, but that would greatly help debugging UOS and in return get more OS running on ACRN.

gvancuts, 2020-07-03:
@NanlinXie I think this is a useful feature, can we assign someone to look at it? (and do we also need to replicate it in the internal DB)?

NanlinXie, 2020-07-06:
No need to replicate it in the internal DB yet, I will follow up this request as DX(Developer eXperience) improvement task by some discussion and investigation firstly.