projectacrn/acrn-hypervisor #5008: Slowdown in UOS (Zephyr)

starmachine24, 2020-07-02:
Hi

We have installed ACRN on our UP_squared board using Debian as our Service VM following these two guides:
https://projectacrn.github.io/latest/tutorials/running_deb_as_serv_vm.html
https://projectacrn.github.io/latest/tutorials/up2.html

We want to use Zephyr as our UOS so we followed this guide: https://projectacrn.github.io/latest/tutorials/using_zephyr_as_uos.html,
we have managed to get everything running except for UART for serial communication. When we measure the amount of iterations taken during one tick in the system, we noticed that compared to running Zephyr directly on the board there is a 10x slowdown when using ACRN. We can understand that using a hypervisor as a backend could result in a small performance loss, but 10x seems a bit excessive. 

Is this a normal behaviour for ACRN or is there something going on here? Currently we are running acrn-2019w35.1-140000p version, Zephyr version 2.2.99 and Debian version 10 (Buster). The build for the ACRN is that the UOS is running on a clock thats 19.2 MHz with a tick rate of 1000 ticks/s.

Here is the program that we are using to measure these results.
[Program.txt](https://github.com/projectacrn/acrn-hypervisor/files/4863948/Program.txt)

Result from running Zephyr as a UOS
```
Cycles taken for 1 second of busy waiting (zephyr's own) = 19280
Cycles taken for 1 second of busy waiting (our busy wait) = 193898
Average amount of iterations taken after 100 loops = 1292
```

Result from running directly on the board
```
Cycles taken for 1 second of busy waiting (zephyr's own) = 19205
Cycles taken for 1 second of busy waiting (our busy wait) = 19361
Average amount of iterations taken after 100 loops = 12849
```

These measurements are taken when each busy wait is supposed to be running for 1 tick, and the average amount of iterations is a very low amount when run as a UOS. We have no idea whats causing this issue and it seems odd given that its the same version of Zephyr being run directly on the board and as a UOS.

fuzhongl, 2020-07-08:
@starmachine24 Thanks for raise ticket for this issue!
We only verify Zephyr version 2.1.
Could you please try Zephyr version 2.1 with acrn 2.0 ? 
If it is related with Zephyr version 2.2, we will raise ticket to Zephyr team to fix it.
Thanks!

starmachine24, 2020-07-20:
Sorry that we haven't responded earlier, we have not managed to test it yet because we have extreme problems to install the hypervisor due to efibootmgr not saving our configuration specifically the boot order which means that the hypervisor never gets launched.

starmachine24, 2020-07-21:
Hi again, we tried to install the v2.0 of ACRN but we can't ever get it to save the boot order. But we found an option in our bios where we could change the boot order directly which seems to impact the uefi boot order as well. And for the older version of ACRN that is used in the guide this works flawlessly and makes it possible for us to just get the hypervisor up and running. But when we use the 2.0 version this does not work because it will never pass the state where it says something in the sense of "detecting 2 processes currently running process 0" screen. 

We are using the N3350 version of the UPsquared board, is there any flags or other parameters that need to be set inorder to build the hypervisor for this specific board? We are asking because we suspect that the problem might come from that the hypervisor is being built for the other board (N4200).

fuzhongl, 2020-07-22:
@starmachine24 Your N3350 UPsquared only has 2 cpu cores; 
please rebuild with following board xml:
 https://github.com/projectacrn/acrn-hypervisor/blob/master/misc/acrn-config/xmls/board-xmls/apl-up2-n3350.xml
 Thanks!

starmachine24, 2020-07-22:
Hi

When we try to build it using the XML files we just get errors, i don't know if we are doing something wrong or if there is something wrong with the scripts building the hypervisor. It complain about something being wrong in the python script which i have no idea he reason for.

We have also gotten an error message saying 
```
"Makefile:73: *** BOARD_FILE/SCENARIO_FILE parameter could not coexist with Kconfig file: /home/linus/acrn-hypervisor/hypervisor/build/.config.  Stop."
```
Which we have no idea what it means or what we have done incorrectly.

Whenever we try to build the defconfig file for the board it just gives us this error mesasage
```
"Using hypervisor configurations from source code directly.
/bin/sh: 1: [: unexpected operator"
```
We attached a file with the error message
[Error message.txt](https://github.com/projectacrn/acrn-hypervisor/files/4958819/Error.message.txt)

fuzhongl, 2020-07-22:
@starmachine24 Please share full command about build hypervisor.
Thanks!

fuzhongl, 2020-07-23:
@starmachine24 
For apl-up2-n3350; only logical_partition.xml was verified.
If you really want to try SDC; please copy it from misc\acrn-config\xmls\config-xmls\apl-up2 to misc\acrn-config\xmls\config-xmls\apl-up2-n3350
and update 
<acrn-config board="apl-up2" scenario="sdc">
to 
<acrn-config board="apl-up2-n3350" scenario="sdc">

Following is the example:
<acrn-config board="**apl-up2-n3350**" scenario="sdc">
  <hv>
    <DEBUG_OPTIONS desc="Debug options for ACRN hypervisor, only valid on debug version">
        <RELEASE desc="Release build. 'y' for Release, 'n' for Debug.">n</RELEASE>
        <SERIAL_CONSOLE desc="The serial device which is used for hypervisor debug, only valid on Debug version.">/dev/ttyS0</SERIAL_CONSOLE>
        <MEM_LOGLEVEL desc="Default loglevel in memory">5</MEM_LOGLEVEL>
        <NPK_LOGLEVEL desc="Default loglevel for the hypervisor NPK log">5</NPK_LOGLEVEL>
        <CONSOLE_LOGLEVEL desc="Default loglevel on the serial console">3</CONSOLE_LOGLEVEL>
        <LOG_DESTINATION desc="Bitmap of consoles where logs are printed.">7</LOG_DESTINATION>
        <LOG_BUF_SIZE desc="Capacity of logbuf for each physical cpu.">0x40000</LOG_BUF_SIZE>
    </DEBUG_OPTIONS>

    <FEATURES>
        <RELOC desc="Enable hypervisor relocation">y</RELOC>
        <SCHEDULER desc="The CPU scheduler to be used by the hypervisor.">SCHED_BVT</SCHEDULER>
        <MULTIBOOT2 desc="Support boot ACRN from multiboot2 protocol.">y</MULTIBOOT2>
	<RDT desc="Intel RDT (Resource Director Technology).">
            <RDT_ENABLED desc="Enable RDT">n</RDT_ENABLED>
            <CDP_ENABLED desc="CDP (Code and Data Prioritization). CDP is an extension of CAT.">n</CDP_ENABLED>
            <CLOS_MASK desc="Cache Capacity Bitmask">0xff</CLOS_MASK>
            <CLOS_MASK desc="Cache Capacity Bitmask">0xff</CLOS_MASK>
            <CLOS_MASK desc="Cache Capacity Bitmask">0xff</CLOS_MASK>
            <CLOS_MASK desc="Cache Capacity Bitmask">0xff</CLOS_MASK>
	</RDT>
        <HYPERV_ENABLED desc="Enable Hyper-V enlightenment">y</HYPERV_ENABLED>
        <IOMMU_ENFORCE_SNP desc="IOMMU enforce snoop behavior of DMA operation.">n</IOMMU_ENFORCE_SNP>
        <ACPI_PARSE_ENABLED desc="Enable ACPI runtime parsing.">y</ACPI_PARSE_ENABLED>
        <L1D_VMENTRY_ENABLED desc="Enable L1 cache flush before VM entry.">n</L1D_VMENTRY_ENABLED>
        <MCE_ON_PSC_DISABLED desc="Force to disable software workaround for Machine Check Error on Page Size Change.">n</MCE_ON_PSC_DISABLED>
    </FEATURES>

    <MEMORY>
        <STACK_SIZE desc="Capacity of one stack, in bytes.">0x2000</STACK_SIZE>
        <HV_RAM_SIZE desc="Size of the RAM region used by the hypervisor"></HV_RAM_SIZE>
        <HV_RAM_START desc="2M-aligned Start physical address of the RAM region used by the hypervisor."></HV_RAM_START>
        <LOW_RAM_SIZE desc="Size of the low RAM region">0x00010000</LOW_RAM_SIZE>
        <UOS_RAM_SIZE desc="Size of the User OS (UOS) RAM.">0x200000000</UOS_RAM_SIZE>
        <SOS_RAM_SIZE desc="Size of the Service OS (SOS) RAM.">0x400000000</SOS_RAM_SIZE>
        <PLATFORM_RAM_SIZE desc="Size of the physical platform RAM">0x400000000</PLATFORM_RAM_SIZE>
    </MEMORY>

    <CAPACITIES desc="Capacity limits for static assigned data struct or maximum supported resouce">
        <IOMMU_BUS_NUM desc="Highest PCI bus ID used during IOMMU initialization.">0x100</IOMMU_BUS_NUM>
        <MAX_IR_ENTRIES desc="Maximum number of Interrupt Remapping Entries.">256</MAX_IR_ENTRIES>
        <MAX_IOAPIC_NUM desc="Maximum number of IO-APICs.">1</MAX_IOAPIC_NUM>
        <MAX_PCI_DEV_NUM desc="Maximum number of PCI devices.">96</MAX_PCI_DEV_NUM>
        <MAX_IOAPIC_LINES desc="Maximum number of interrupt lines per IOAPIC.">120</MAX_IOAPIC_LINES>
        <MAX_PT_IRQ_ENTRIES desc="Maximum number of interrupt source for PT devices.">64</MAX_PT_IRQ_ENTRIES>
        <MAX_MSIX_TABLE_NUM desc="Maximum number of MSI-X tables per device. Please leave it blank if not sure.">16</MAX_MSIX_TABLE_NUM>
        <MAX_EMULATED_MMIO desc="Maximum number of emulated MMIO regions.">16</MAX_EMULATED_MMIO>
    </CAPACITIES>

    <MISC_CFG>
        <GPU_SBDF desc="Segment, Bus, Device, and function of the GPU.">0x00000010</GPU_SBDF>
        <UEFI_OS_LOADER_NAME desc="UEFI OS loader name.">\\EFI\\BOOT\\bootx64.efi</UEFI_OS_LOADER_NAME>
    </MISC_CFG>
  </hv>

  <vm id="0">
    <vm_type desc="Specify the VM type" readonly="true">SOS_VM</vm_type>
    <name desc="Specify the VM name which will be shown in hypervisor console command: vm_list.">ACRN SOS VM</name>
[... 73 lines cut ...]
    <vm_type readonly="true" desc="Specify the VM type">KATA_VM</vm_type>
    <cpu_affinity desc="List of pCPU: the guest VM is allowed to create vCPU from all or a subset of this list.">
        <pcpu_id>3</pcpu_id>
    </cpu_affinity>
    <clos desc="Class of Service for Cache Allocation Technology. Please refer SDM 17.19.2 for details and use with caution.">
        <vcpu_clos>0</vcpu_clos>
    </clos>
    <epc_section configurable="0" desc="epc section">
        <base desc="SGX EPC section base, must be page aligned">0</base>
        <size desc="SGX EPC section size in Bytes, must be page aligned">0</size>
    </epc_section>
    <vuart id="0">
        <type configurable="0" desc="vCOM1 type">VUART_LEGACY_PIO</type>
        <base configurable="0" desc="vUART0 (A.K.A COM1) enabling switch. Enable by exposing its base address, disable by returning invalid base address.">INVALID_COM_BASE</base>
[... cut ...]
        <base configurable="0" desc="vUART1 (A.K.A COM2) enabling switch. Enable by exposing its base address, disable by returning invalid base address.">INVALID_COM_BASE</base>
        <irq configurable="0" desc="vCOM2 irq">COM2_IRQ</irq>
        <target_vm_id desc="COM2 is used for VM communications. When it is enabled, please specify which target VM that current VM connect to.">0</target_vm_id>
        <target_uart_id configurable="0" desc="target vUART ID that vCOM2 connect to">0</target_uart_id>
    </vuart>
  </vm>
</acrn-config>

starmachine24, 2020-07-23:
With the new xml file you provided us with we can now build the hypervisor using the XML, but we still get stuck on the startup screen when it says it has detected two processes and is currently working on process 0. It might be that we need to change some of the settings in the XML file for when we build the hypervisor or some other parameters/settings.

fuzhongl, 2020-07-24:
@starmachine24 Any error form Hypervisor console?
Please try again with following sdc.xml

```
<acrn-config board="apl-up2-n3350" scenario="sdc">
  <hv>
    <DEBUG_OPTIONS desc="Debug options for ACRN hypervisor, only valid on debug version">
        <RELEASE desc="Release build. 'y' for Release, 'n' for Debug.">n</RELEASE>
        <SERIAL_CONSOLE desc="The serial device which is used for hypervisor debug, only valid on Debug version.">/dev/ttyS0</SERIAL_CONSOLE>
        <MEM_LOGLEVEL desc="Default loglevel in memory">5</MEM_LOGLEVEL>
        <NPK_LOGLEVEL desc="Default loglevel for the hypervisor NPK log">5</NPK_LOGLEVEL>
        <CONSOLE_LOGLEVEL desc="Default loglevel on the serial console">3</CONSOLE_LOGLEVEL>
        <LOG_DESTINATION desc="Bitmap of consoles where logs are printed.">7</LOG_DESTINATION>
        <LOG_BUF_SIZE desc="Capacity of logbuf for each physical cpu.">0x40000</LOG_BUF_SIZE>
    </DEBUG_OPTIONS>

    <FEATURES>
        <RELOC desc="Enable hypervisor relocation">y</RELOC>
        <SCHEDULER desc="The CPU scheduler to be used by the hypervisor.">SCHED_BVT</SCHEDULER>
        <MULTIBOOT2 desc="Support boot ACRN from multiboot2 protocol.">y</MULTIBOOT2>
	<RDT desc="Intel RDT (Resource Director Technology).">
            <RDT_ENABLED desc="Enable RDT">n</RDT_ENABLED>
            <CDP_ENABLED desc="CDP (Code and Data Prioritization). CDP is an extension of CAT.">n</CDP_ENABLED>
            <CLOS_MASK desc="Cache Capacity Bitmask">0xff</CLOS_MASK>
            <CLOS_MASK desc="Cache Capacity Bitmask">0xff</CLOS_MASK>
            <CLOS_MASK desc="Cache Capacity Bitmask">0xff</CLOS_MASK>
            <CLOS_MASK desc="Cache Capacity Bitmask">0xff</CLOS_MASK>
	</RDT>
        <HYPERV_ENABLED desc="Enable Hyper-V enlightenment">y</HYPERV_ENABLED>
        <IOMMU_ENFORCE_SNP desc="IOMMU enforce snoop behavior of DMA operation.">n</IOMMU_ENFORCE_SNP>
        <ACPI_PARSE_ENABLED desc="Enable ACPI runtime parsing.">y</ACPI_PARSE_ENABLED>
        <L1D_VMENTRY_ENABLED desc="Enable L1 cache flush before VM entry.">n</L1D_VMENTRY_ENABLED>
        <MCE_ON_PSC_DISABLED desc="Force to disable software workaround for Machine Check Error on Page Size Change.">n</MCE_ON_PSC_DISABLED>
    </FEATURES>

    <MEMORY>
        <STACK_SIZE desc="Capacity of one stack, in bytes.">0x2000</STACK_SIZE>
        <HV_RAM_SIZE desc="Size of the RAM region used by the hypervisor"></HV_RAM_SIZE>
        <HV_RAM_START desc="2M-aligned Start physical address of the RAM region used by the hypervisor."></HV_RAM_START>
        <LOW_RAM_SIZE desc="Size of the low RAM region">0x00010000</LOW_RAM_SIZE>
        <UOS_RAM_SIZE desc="Size of the User OS (UOS) RAM.">0x200000000</UOS_RAM_SIZE>
        <SOS_RAM_SIZE desc="Size of the Service OS (SOS) RAM.">0x400000000</SOS_RAM_SIZE>
        <PLATFORM_RAM_SIZE desc="Size of the physical platform RAM">0x400000000</PLATFORM_RAM_SIZE>
    </MEMORY>

    <CAPACITIES desc="Capacity limits for static assigned data struct or maximum supported resouce">
        <IOMMU_BUS_NUM desc="Highest PCI bus ID used during IOMMU initialization.">0x100</IOMMU_BUS_NUM>
        <MAX_IR_ENTRIES desc="Maximum number of Interrupt Remapping Entries.">256</MAX_IR_ENTRIES>
        <MAX_IOAPIC_NUM desc="Maximum number of IO-APICs.">1</MAX_IOAPIC_NUM>
        <MAX_PCI_DEV_NUM desc="Maximum number of PCI devices.">96</MAX_PCI_DEV_NUM>
        <MAX_IOAPIC_LINES desc="Maximum number of interrupt lines per IOAPIC.">120</MAX_IOAPIC_LINES>
        <MAX_PT_IRQ_ENTRIES desc="Maximum number of interrupt source for PT devices.">64</MAX_PT_IRQ_ENTRIES>
        <MAX_MSIX_TABLE_NUM desc="Maximum number of MSI-X tables per device. Please leave it blank if not sure.">16</MAX_MSIX_TABLE_NUM>
        <MAX_EMULATED_MMIO desc="Maximum number of emulated MMIO regions.">16</MAX_EMULATED_MMIO>
    </CAPACITIES>

    <MISC_CFG>
        <GPU_SBDF desc="Segment, Bus, Device, and function of the GPU.">0x00000010</GPU_SBDF>
        <UEFI_OS_LOADER_NAME desc="UEFI OS loader name.">\\EFI\\BOOT\\bootx64.efi</UEFI_OS_LOADER_NAME>
    </MISC_CFG>
  </hv>

  <vm id="0">
    <vm_type desc="Specify the VM type" readonly="true">SOS_VM</vm_type>
    <name desc="Specify the VM name which will be shown in hypervisor console command: vm_list.">ACRN SOS VM</name>
    <guest_flags desc="Select all applicable flags for the VM" multiselect="true">
        <guest_flag>0</guest_flag>
    </guest_flags>
    <clos configurable="0" desc="Class of Service for Cache Allocation Technology. Please refer SDM 17.19.2 for details and use with caution.">
        <vcpu_clos>0</vcpu_clos>
[... 70 lines cut ...]
        <pcpu_id>1</pcpu_id>
    </cpu_affinity>
    <clos desc="Class of Service for Cache Allocation Technology. Please refer SDM 17.19.2 for details and use with caution.">
        <vcpu_clos>0</vcpu_clos>
    </clos>
    <epc_section configurable="0" desc="epc section">
        <base desc="SGX EPC section base, must be page aligned">0</base>
        <size desc="SGX EPC section size in Bytes, must be page aligned">0</size>
    </epc_section>
    <vuart id="0">
        <type configurable="0" desc="vCOM1 type">VUART_LEGACY_PIO</type>
        <base configurable="0" desc="vUART0 (A.K.A COM1) enabling switch. Enable by exposing its base address, disable by returning invalid base address.">INVALID_COM_BASE</base>
        <irq configurable="0" desc="vCOM1 irq">COM1_IRQ</irq>
    </vuart>
    <vuart id="1">
[... cut ...]
        <base configurable="0" desc="vUART1 (A.K.A COM2) enabling switch. Enable by exposing its base address, disable by returning invalid base address.">INVALID_COM_BASE</base>
        <irq configurable="0" desc="vCOM2 irq">COM2_IRQ</irq>
        <target_vm_id desc="COM2 is used for VM communications. When it is enabled, please specify which target VM that current VM connect to.">0</target_vm_id>
        <target_uart_id configurable="0" desc="target vUART ID that vCOM2 connect to">0</target_uart_id>
    </vuart>
  </vm>
</acrn-config>

```

starmachine24, 2020-07-24:
Hi,

Thank you, now it works, we can build it and the hypervisor starts. When we run Zephyr as our User VM we want to get the output on the UART 1 port, but right now we can't seem to get that to work. Would you have any idea of why? 
We have configured the Zephyr OS to output the information on UART 1 and in the launch_zephyr.sh we changed the COM1 to COM2 but this didn't seem to have any effect.

fuzhongl, 2020-07-24:
@starmachine24 Do you get HV log form COM1 or COM2?
If yes, please share your launch_zephyr.sh.

starmachine24, 2020-07-25:
@fuzhongl What do you mean by HV log? 

If we have the UART cable connected to another computer when we start the board we get some information sent over the UARt which i have no idea what it is, but i would take a guess that it is information related to the hypervisor. But after we have booted into the service VM and then launch the user VM we aren't getting any output from the UART.

fuzhongl, 2020-07-27:
@starmachine24 Do you use the [sample launch_zephyr.sh](https://github.com/projectacrn/acrn-hypervisor/blob/master/devicemodel/samples/nuc/`launch_zephyr.sh)?
Can you see `Hello World! acrn` in the screen after running launch_zephyr.sh?
If you want to get zephyr output log from UART, please try following 2 steps:
1) update sdc xml
<acrn-config board="apl-up2-n3350" scenario="sdc">
  <hv>
    <DEBUG_OPTIONS desc="Debug options for ACRN hypervisor, only valid on debug version">
        <RELEASE desc="Release build. 'y' for Release, 'n' for Debug.">n</RELEASE>
        <SERIAL_CONSOLE desc="The serial device which is used for hypervisor debug, only valid on Debug version.">/dev/ttyS0</SERIAL_CONSOLE>
        <MEM_LOGLEVEL desc="Default loglevel in memory">5</MEM_LOGLEVEL>
        <NPK_LOGLEVEL desc="Default loglevel for the hypervisor NPK log">5</NPK_LOGLEVEL>
        <CONSOLE_LOGLEVEL desc="Default loglevel on the serial console">3</CONSOLE_LOGLEVEL>
        <LOG_DESTINATION desc="Bitmap of consoles where logs are printed.">7</LOG_DESTINATION>
        <LOG_BUF_SIZE desc="Capacity of logbuf for each physical cpu.">0x40000</LOG_BUF_SIZE>
    </DEBUG_OPTIONS>

    <FEATURES>
        <RELOC desc="Enable hypervisor relocation">y</RELOC>
        <SCHEDULER desc="The CPU scheduler to be used by the hypervisor.">SCHED_BVT</SCHEDULER>
        <MULTIBOOT2 desc="Support boot ACRN from multiboot2 protocol.">y</MULTIBOOT2>
	<RDT desc="Intel RDT (Resource Director Technology).">
            <RDT_ENABLED desc="Enable RDT">n</RDT_ENABLED>
            <CDP_ENABLED desc="CDP (Code and Data Prioritization). CDP is an extension of CAT.">n</CDP_ENABLED>
            <CLOS_MASK desc="Cache Capacity Bitmask">0xff</CLOS_MASK>
            <CLOS_MASK desc="Cache Capacity Bitmask">0xff</CLOS_MASK>
            <CLOS_MASK desc="Cache Capacity Bitmask">0xff</CLOS_MASK>
            <CLOS_MASK desc="Cache Capacity Bitmask">0xff</CLOS_MASK>
	</RDT>
        <HYPERV_ENABLED desc="Enable Hyper-V enlightenment">y</HYPERV_ENABLED>
        <IOMMU_ENFORCE_SNP desc="IOMMU enforce snoop behavior of DMA operation.">n</IOMMU_ENFORCE_SNP>
        <ACPI_PARSE_ENABLED desc="Enable ACPI runtime parsing.">y</ACPI_PARSE_ENABLED>
        <L1D_VMENTRY_ENABLED desc="Enable L1 cache flush before VM entry.">n</L1D_VMENTRY_ENABLED>
        <MCE_ON_PSC_DISABLED desc="Force to disable software workaround for Machine Check Error on Page Size Change.">n</MCE_ON_PSC_DISABLED>
    </FEATURES>

    <MEMORY>
        <STACK_SIZE desc="Capacity of one stack, in bytes.">0x2000</STACK_SIZE>
        <HV_RAM_SIZE desc="Size of the RAM region used by the hypervisor"></HV_RAM_SIZE>
        <HV_RAM_START desc="2M-aligned Start physical address of the RAM region used by the hypervisor."></HV_RAM_START>
        <LOW_RAM_SIZE desc="Size of the low RAM region">0x00010000</LOW_RAM_SIZE>
        <UOS_RAM_SIZE desc="Size of the User OS (UOS) RAM.">0x200000000</UOS_RAM_SIZE>
        <SOS_RAM_SIZE desc="Size of the Service OS (SOS) RAM.">0x400000000</SOS_RAM_SIZE>
        <PLATFORM_RAM_SIZE desc="Size of the physical platform RAM">0x400000000</PLATFORM_RAM_SIZE>
    </MEMORY>

    <CAPACITIES desc="Capacity limits for static assigned data struct or maximum supported resouce">
        <IOMMU_BUS_NUM desc="Highest PCI bus ID used during IOMMU initialization.">0x100</IOMMU_BUS_NUM>
        <MAX_IR_ENTRIES desc="Maximum number of Interrupt Remapping Entries.">256</MAX_IR_ENTRIES>
        <MAX_IOAPIC_NUM desc="Maximum number of IO-APICs.">1</MAX_IOAPIC_NUM>
        <MAX_PCI_DEV_NUM desc="Maximum number of PCI devices.">96</MAX_PCI_DEV_NUM>
        <MAX_IOAPIC_LINES desc="Maximum number of interrupt lines per IOAPIC.">120</MAX_IOAPIC_LINES>
        <MAX_PT_IRQ_ENTRIES desc="Maximum number of interrupt source for PT devices.">64</MAX_PT_IRQ_ENTRIES>
        <MAX_MSIX_TABLE_NUM desc="Maximum number of MSI-X tables per device. Please leave it blank if not sure.">16</MAX_MSIX_TABLE_NUM>
        <MAX_EMULATED_MMIO desc="Maximum number of emulated MMIO regions.">16</MAX_EMULATED_MMIO>
    </CAPACITIES>

    <MISC_CFG>
        <GPU_SBDF desc="Segment, Bus, Device, and function of the GPU.">0x00000010</GPU_SBDF>
        <UEFI_OS_LOADER_NAME desc="UEFI OS loader name.">\\EFI\\BOOT\\bootx64.efi</UEFI_OS_LOADER_NAME>
    </MISC_CFG>
  </hv>

  <vm id="0">
    <vm_type desc="Specify the VM type" readonly="true">SOS_VM</vm_type>
    <name desc="Specify the VM name which will be shown in hypervisor console command: vm_list.">ACRN SOS VM</name>
    <guest_flags desc="Select all applicable flags for the VM" multiselect="true">
        <guest_flag>0</guest_flag>
    </guest_flags>
    <clos configurable="0" desc="Class of Service for Cache Allocation Technology. Please refer SDM 17.19.2 for details and use with caution.">
        <vcpu_clos>0</vcpu_clos>
[... 70 lines cut ...]
        <pcpu_id>1</pcpu_id>
    </cpu_affinity>
    <clos desc="Class of Service for Cache Allocation Technology. Please refer SDM 17.19.2 for details and use with caution.">
        <vcpu_clos>0</vcpu_clos>
    </clos>
    <epc_section configurable="0" desc="epc section">
        <base desc="SGX EPC section base, must be page aligned">0</base>
        <size desc="SGX EPC section size in Bytes, must be page aligned">0</size>
    </epc_section>
    <vuart id="0">
        <type configurable="0" desc="vCOM1 type">VUART_LEGACY_PIO</type>
[... cut ...]
        <base configurable="0" desc="vUART1 (A.K.A COM2) enabling switch. Enable by exposing its base address, disable by returning invalid base address.">INVALID_COM_BASE</base>
        <irq configurable="0" desc="vCOM2 irq">COM2_IRQ</irq>
        <target_vm_id desc="COM2 is used for VM communications. When it is enabled, please specify which target VM that current VM connect to.">0</target_vm_id>
        <target_uart_id configurable="0" desc="target vUART ID that vCOM2 connect to">0</target_uart_id>
    </vuart>
  </vm>
</acrn-config>

2) After launch launch_zephyr.sh; input `vm_console 1` in Hypervisor console(with ACRN:\ character )

starmachine24, 2020-07-27:
Hi

When we try to build the hypervisor using the new sdc configuration we get this error:
```
scenarios/sdc/vm_configurations.c:42:23: error: invalid type argument of unary ‘*’ (have ‘unsigned int’)
    .addr.port_base = **COM1_BASE**,
scenarios/sdc/vm_configurations.c:42:35: error: expected expression before ‘,’ token
    .addr.port_base = **COM1_BASE**,
                                   ^
make[1]: *** [Makefile:541: /root/acrn-hypervisor/build/hypervisor/scenarios/sdc/vm_configurations.o] Error 1
make[1]: Leaving directory '/root/acrn-hypervisor/hypervisor'
make: *** [Makefile:163: hypervisor] Error 2
```

starmachine24, 2020-07-27:
Hi, 

We get this error when we try to build the hypervisor using the new sdc.xml configuration:
```
scenarios/sdc/vm_configurations.c:42:23: error: invalid type argument of unary ‘*’ (have ‘unsigned int’)
    .addr.port_base = **COM1_BASE**,
scenarios/sdc/vm_configurations.c:42:35: error: expected expression before ‘,’ token
    .addr.port_base = **COM1_BASE**,
                                   ^
make[1]: *** [Makefile:541: /root/acrn-hypervisor/build/hypervisor/scenarios/sdc/vm_configurations.o] Error 1
make[1]: Leaving directory '/root/acrn-hypervisor/hypervisor'
make: *** [Makefile:163: hypervisor] Error 2
```

fuzhongl, 2020-07-27:
Just want to highlight the modification.
Please change **COM1_BASE** to COM1_BASE.

starmachine24, 2020-07-27:
Oh okay, sorry we first thought you meant that we should have changed the parameter there or something.

starmachine24, 2020-07-27:
Now we managed to get the output we wanted, however it is printed out in the same console window that we launch the user VM from, while that is fine for now it would be more preferable if we could get the info via the uart.

fuzhongl, 2020-07-28:
@starmachine24 
Please input "vm_list" in Hypervisor console(with ACRN:\ character) to check vm id of Zephyr uos;
Then input vm_console 1 if vm1 is Zephyr uos.

fuzhongl, 2020-07-28:
@starmachine24
Do you connect serial cable for your up2?
If yes, can you get UART log for native boot?

starmachine24, 2020-07-28:
@fuzhongl 
Oh now we found it, didn't know it would be over uart the ACRN console would be shown. When i ran the command vm_list it said that ACRN VM_1 had ID 1. Which means that we should run vm_console 1 then?

fuzhongl, 2020-07-28:
Any log output after input "vm_console 1" in hypervisor console?
Could you please try with another uos? i.e. clearlinux uos

starmachine24, 2020-07-29:
We cannot verify if Zephyr 2.1 runs correctly due to that it doesn't seem to have support for UART when using CTF, but only support posix. And for our work we need to use ACRN and Zephyr, Right now our biggest concern is getting the trace output, and if it is printed in the same console window that the user VM is launched from that is fine. So right now the problem is what is wrong when Zephyr version 2.2.99 and 2.3.0 is run via ACRN hypervisor since the difference in run time is so extreme.

closed 2020-09-10 by starmachine24