projectacrn/acrn-hypervisor #5033: SOS decode instruction failed in hybrid mode

ChicLee0819, 2020-07-14:
**Describe the bug**
I tried to hybrid mode with my own device and pre-launched vm0 is ok but service VM is failed.
I also tried to industry mode and everything is OK.
The service VM kernel is the same in hybrid and industry.
I'm not sure where is the mistake.

**To Reproduce**
Steps to reproduce the behavior:
1. Create board info file(XML) for my own device and generate scenario file(hybrid) using configuration app
2. make BOARD_FILE=$(board info file path) SCENARIO_FILE=$(scenario file path) RELEASE=0
3. modify /etc/grub.d/40_custom

menuentry 'ACRN hypervisor Hybird Scenario' --class ubuntu --class gnu-linux --class gnu --class os $menuentry_id_option 'gnulinux-simple-e23c76ae-b06d-4a6e-ad42-46b8eedfd7d3' {
   recordfail
   load_video
   gfxmode $linux_gfx_mode
   insmod gzio
   insmod part_gpt
   insmod ext2
   echo 'Loading hypervisor Hybrid scenario ...'
   multiboot2 /boot/acrn/acrn.bin
   module2 /boot/acrn/bzImage_rt Linux_bzImage2 root=PARTUUID="xxxxxxxxxxxxxxxxxxxxxx" rw rootwait nohpet console=hvc0 console=ttyS0 no_timer_check ignore_loglevel log_buf_len=16M consoleblank=0
   module2 /boot/acrn/bzImage Linux_bzImage
}

4. select it in GRUB menu

**Expected behavior**
The pre-launched VM and service VM are OK.

**Additional context**

[0us][cpu=0][(null)][sev=5][seq=1]:Multiboot2 detected.
Detect bootloader: GRUB 2.02-2ubuntu8.14
ACRN Hypervisor
calibrate_tsc, tsc_khz=2900000
[24798530us][cpu=0][(null)][sev=2][seq=2]:HV version 2.0-2020-07-13 07:00:39-ae748704-dirty DBG (daily tag:acrn-2020w25.7-180000p) build by sos with acrn-config, start time 24783232us
[24817571us][cpu=0][(null)][sev=2][seq=3]:API version 1.0
[24823715us][cpu=0][(null)][sev=2][seq=4]:Detect processor: Intel(R) Core(TM) i7-7600U CPU @ 2.80GHz
[24834103us][cpu=0][(null)][sev=1][seq=5]:SECURITY WARNING!!!!!!
[24840769us][cpu=0][(null)][sev=1][seq=6]:Please apply the latest CPU uCode patch!
[24849448us][cpu=0][(null)][sev=4][seq=7]:DRHD with INCLUDE_PCI_ALL flag is NOT the last one!
[24916205us][cpu=0][(null)][sev=3][seq=8]:prepare the stack for AP trampline code
[24924712us][cpu=0][(null)][sev=3][seq=9]:prepare the stack for AP trampline code
[24933200us][cpu=0][(null)][sev=3][seq=10]:prepare the stack for AP trampline code
[24941697us][cpu=0][(null)][sev=5][seq=11]:profiling_setup: entering
[24941697us][cpu=2][(null)][sev=5][seq=12]:profiling_setup: entering
[24941697us][cpu=1][(null)][sev=5][seq=14]:profiling_setup: entering
[24941697us][cpu=3][(null)][sev=5][seq=13]:profiling_setup: entering
[24948721us][cpu=0][(null)][sev=5][seq=15]:profiling_setup: calling request_irq
[24955847us][cpu=2][(null)][sev=5][seq=16]:profiling_setup: exiting
[24962921us][cpu=1][(null)][sev=5][seq=17]:profiling_setup: exiting
[24970046us][cpu=3][(null)][sev=5][seq=18]:profiling_setup: exiting
[24978333us][cpu=0][(null)][sev=5][seq=19]:profiling_setup: exiting
[25000021us][cpu=3][(null)][sev=5][seq=20]:Creating VCPU working on PCPU3ACRN:\>
[25014568us][cpu=3][(null)][sev=5][seq=21]:Create VM0-VCPU0, Role: PRIMARY
[25015517us][cpu=0][(null)][sev=5][seq=22]:System S3/S5 is supported.
[25024040us][cpu=3][(null)][sev=5][seq=23]:prepare_loading_bzimage, RSI pointing to zero page for VM 0 at GPA FFF000
[25032417us][cpu=0][(null)][sev=5][seq=24]:Creating VCPU working on PCPU0
[25041444us][cpu=3][(null)][sev=5][seq=25]:direct_boot_sw_loader, VM 0 VCPU 0 Entry: 0x0000000001004200
[25048954us][cpu=0][(null)][sev=5][seq=26]:Create VM1-VCPU0, Role: PRIMARY
[25059602us][cpu=3][(null)][sev=2][seq=27]:Start VM id: 0 name: ACRN PRE-LAUNCHED VM0
[25067354us][cpu=0][(null)][sev=5][seq=28]:Creating VCPU working on PCPU1
[25076016us][cpu=3][vm0:vcpu0][sev=5][seq=29]:VM 0 Starting VCPU 0
[25083616us][cpu=0][(null)][sev=5][seq=30]:Create VM1-VCPU1, Role: SECONDARY
[25090481us][cpu=3][vm0:vcpu0][sev=5][seq=31]:VM 0 VCPU 0 successfully launched
[25098379us][cpu=0][(null)][sev=5][seq=32]:Creating VCPU working on PCPU2
[25114003us][cpu=0][(null)][sev=5][seq=33]:Create VM1-VCPU2, Role: SECONDARY
[25125054us][cpu=0][(null)][sev=5][seq=34]:prepare_loading_bzimage, RSI pointing to zero page for VM 1 at GPA FFF000
[25136824us][cpu=0][(null)][sev=5][seq=35]:direct_boot_sw_loader, VM 1 VCPU 0 Entry: 0x0000000001004600
[25147512us][cpu=0][(null)][sev=2][seq=36]:Start VM id: 1 name: ACRN SOS VM
[25155329us][cpu=0][vm1:vcpu0][sev=5][seq=37]:VM 1 Starting VCPU 0
[25162192us][cpu=0][vm1:vcpu0][sev=5][seq=38]:VM 1 VCPU 0 successfully launched
[25170572us][cpu=0][vm1:vcpu0][sev=3][seq=39]:decode instruction failed @ 0x0000000001dc4659:
[25180059us][cpu=0][vm1:vcpu0][sev=5][seq=40]:page fault happen during decode_instruction
[25189218us][cpu=0][vm1:vcpu0][sev=2][seq=41]:Guest Linear Address: 0x0000000011004200
[25198060us][cpu=0][vm1:vcpu0][sev=2][seq=42]:Guest Physical Address address: 0x0000000011004200
[25207954us][cpu=0][vm1:vcpu0][sev=1][seq=43]:VM1: triple fault @ guest RIP 0x0000000001dc4659, exit qualification: 0x0000000000000000

fuzhongl, 2020-08-14:
@ChicLee0819 This issue is fixed  in kernel v2.1 branch. Could you please try v2.1 kernel?

ChicLee0819, 2020-08-17:
@fuzhongl I tried v2.1 release and the issue is fixed. Thanks a lot.

closed 2020-08-17 by ChicLee0819