projectacrn/acrn-hypervisor #5044: Juniper Hill board build and run

ynahum, 2020-07-15:
**Describe the bug**
Using APL with Juniper Hill board. 
Tried to build w/ BOARD=generic but I'm not sure it will work.
What configuration should I use for this kind of board. didn't find any info on it.
Also, how to run it exactly once it is built?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Attach the following logs if available:

- native Linux boot kernel message log
- ACRN serial port log (when issues happens)
- Service VM kernel message when issue happens
- acrn-dm log: all files in /var/log/acrn-tm
- console log when running script to launch guest VM

fuzhongl, 2020-07-21:
@ynahum VT-D can't be enabled on Juniper Hill board.

ynahum, 2020-07-21:
Doesn't it depend on the BIOS version? would it not support VT-D on JuniperHill w/o relation to the BIOS/UEFI?

Thanks.
Yair

fuzhongl, 2020-07-21:
please update BIOS with VT-D enable option.

closed 2021-01-07 by fuzhongl