projectacrn/acrn-hypervisor #5047: The ACRN SOS dmesg log is flooded with DRM error messages

na3em, 2020-07-16:
**Describe the bug**

- After booting the ACRN SOS, the dmesg console is flooded with an error message.
- Error logs are coming from gfx/display pipes.
- We can see these error messages also if there is no eDP display connected
- The issue does not occur if we use the ACRN kernel (based on 4.19).
- The issue occurs with linux-intel-acrn kernel (Linux intel-corei7-64 5.4.46-linux-intel-acrn-sos).
- Issue resolved when SOS is not assigned to pipe A or B (i915.avail_planes_per_pipe=0x010000 and i915.domain_plane_owners=0x000022221111)
- The issue is also observed when changing the SOS display configuration on the fly.
- The issue is observed only for certain monitors
- The 12'' automotive display at the HDMI output.
- Two big displays HP EliteDisplay E232, using two HDMI cables, with one DP-to-HDMI adapter).
- Dell monitors seem to work fine.

**To Reproduce**
Steps to reproduce the behavior:
1. Build acrn with the industrial scenario.
2. Use Yocto based SOS with Linux Intel-corei7-64 5.4.46-linux-intel-acrn-sos
3. Change i915 configurations on the fly or use one of the mentioned monitor types.

**Expected behavior**
In a normal scenario, the dmesg shouldn't be flooded with this error.

**Additional context**
dmesg errors: 

[  278.508985] [drm:gen8_de_irq_handler] ERROR Fault errors on pipe B: 0x00000080
[  278.524843] [drm:gen8_de_irq_handler] ERROR Fault errors on pipe B: 0x00000080
[  278.525649] [drm:gen8_de_irq_handler] ERROR Fault errors on pipe B: 0x00000080

na3em, 2020-07-16:
[External_System_ID] ACRN-6554

fuzhongl, 2020-11-02:
@na3em  Do you still have issue with v2.2?

fuzhongl, 2021-05-26:
Closed since no feedback for long time.

closed 2021-05-26 by fuzhongl