projectacrn/acrn-hypervisor #5048: - iTCO_wdt issue: can't request region for resource -

ionutnechita, 2020-07-16:
**Describe the bug**
[ 0.760862] iTCO_wdt: Intel TCO WatchDog Timer Driver v1.11
[ 0.760937] iTCO_wdt iTCO_wdt: can't request region for resource [mem 0x00c5fffc-0x00c5ffff]
[ 0.760942] iTCO_wdt: probe of iTCO_wdt failed with error -16

**To Reproduce**
Ubuntu 20.04 SOS
ACRN master branch
32G RAM SOS

lscpu
Architecture: x86_64
CPU op-mode(s): 32-bit, 64-bit
Byte Order: Little Endian
CPU(s): 4
On-line CPU(s) list: 0-3
Thread(s) per core: 2
Core(s) per socket: 2
Socket(s): 1
Vendor ID: GenuineIntel
CPU family: 6
Model: 142
Model name: Intel(R) Core(TM) i7-7567U CPU @ 3.50GHz
Stepping: 9
CPU MHz: 3932.239
CPU max MHz: 4000.0000
CPU min MHz: 400.0000
BogoMIPS: 7008.00
L1d cache: 32K
L1i cache: 32K
L2 cache: 256K
L3 cache: 4096K
Flags: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq monitor est tm2 ssse3 fma cx16 xtpr sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand hypervisor lahf_lm abm 3dnowprefetch cpuid_fault epb pti ssbd ibrs ibpb stibp fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms rdseed adx smap clflushopt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp flush_l1d

lspci
00:00.0 Host bridge: Intel Corporation Celeron N3350/Pentium N4200/Atom E3900 Series Host Bridge (rev 0b)
00:02.0 VGA compatible controller: Intel Corporation Device 5927 (rev 06)
00:08.0 System peripheral: Intel Corporation Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th Gen Core Processor Gaussian Mixture Model
00:14.0 USB controller: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller (rev 21)
00:14.2 Signal processing controller: Intel Corporation Sunrise Point-LP Thermal subsystem (rev 21)
00:16.0 Communication controller: Intel Corporation Sunrise Point-LP CSME HECI #1 (rev 21)
00:17.0 SATA controller: Intel Corporation Sunrise Point-LP SATA Controller [AHCI mode] (rev 21)
00:1c.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #1 (rev f1)
00:1c.5 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #6 (rev f1)
00:1c.7 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #8 (rev f1)
00:1f.0 ISA bridge: Intel Corporation Intel(R) 100 Series Chipset Family LPC Controller/eSPI Controller - 9D4E (rev 21)
00:1f.2 Memory controller: Intel Corporation Sunrise Point-LP PMC (rev 21)
00:1f.3 Audio device: Intel Corporation Sunrise Point-LP HD Audio (rev 21)
00:1f.4 SMBus: Intel Corporation Sunrise Point-LP SMBus (rev 21)
00:1f.6 Ethernet controller: Intel Corporation Ethernet Connection (4) I219-V (rev 21)
3a:00.0 Network controller: Intel Corporation Wireless 8265 / 8275 (rev 78)
3b:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS5229 PCI Express Card Reader (rev 01)

NUC7i7BNH

Linux os1 4.19.128-109.iot-lts2018-sos #1 SMP PREEMPT Sat Jun 20 15:06:27 PDT 2020 x86_64 x86_64 x86_64 GNU/Linux

fuzhongl, 2020-07-21:
@jsun26intel  Could you please help to check this issue?
Thanks!

fuzhongl, 2020-07-22:
@ionutnechita Please share following info in sos console.
cat /proc/ioports

ionutnechita, 2020-07-22:
Hi @fuzhongl, @jsun26intel 

Output for cat /proc/ioports:
cat /proc/ioports
0000-0000 : PCI Bus 0000:00
  0000-0000 : dma1
  0000-0000 : pic1
  0000-0000 : timer0
  0000-0000 : timer1
  0000-0000 : keyboard
  0000-0000 : PNP0C09:00
    0000-0000 : EC data
  0000-0000 : keyboard
  0000-0000 : PNP0C09:00
    0000-0000 : EC cmd
  0000-0000 : rtc0
  0000-0000 : dma page reg
  0000-0000 : pic2
  0000-0000 : dma2
  0000-0000 : fpu
  0000-0000 : serial
  0000-0000 : iTCO_wdt
  0000-0000 : pnp 00:02
  0000-0000 : pnp 00:01
  0000-0000 : pnp 00:01
  0000-0000 : pnp 00:01
  0000-0000 : pnp 00:01
  0000-0000 : pnp 00:01
  0000-0000 : pnp 00:01
0000-0000 : PCI conf1
0000-0000 : PCI Bus 0000:00
  0000-0000 : pnp 00:02
  0000-0000 : pnp 00:02
    0000-0000 : ACPI PM1a_EVT_BLK
    0000-0000 : ACPI PM1a_CNT_BLK
    0000-0000 : ACPI PM_TMR
    0000-0000 : iTCO_wdt
    0000-0000 : ACPI PM2_CNT_BLK
    0000-0000 : pnp 00:04
    0000-0000 : ACPI GPE0_BLK
  0000-0000 : PCI Bus 0000:01
  0000-0000 : 0000:00:02.0
  0000-0000 : 0000:00:1f.4
    0000-0000 : i801_smbus
  0000-0000 : 0000:00:17.0
    0000-0000 : ahci
  0000-0000 : 0000:00:17.0
    0000-0000 : ahci
  0000-0000 : 0000:00:17.0
    0000-0000 : ahci
  0000-0000 : pnp 00:07
  0000-0000 : pnp 00:02
    0000-0000 : pnp 00:02
      0000-0000 : pnp 00:02

fuzhongl, 2020-07-22:
@ionutnechita Could you please share these info for native boot?
Thanks!

ionutnechita, 2020-07-23:
Hi @fuzhongl, @jsun26intel

Output for cat /proc/ioports with Native:
cat /proc/ioports 
0000-0cf7 : PCI Bus 0000:00
  0000-001f : dma1
  0020-0021 : pic1
  0040-0043 : timer0
  0050-0053 : timer1
  0060-0060 : keyboard
  0062-0062 : PNP0C09:00
    0062-0062 : EC data
  0064-0064 : keyboard
  0066-0066 : PNP0C09:00
    0066-0066 : EC cmd
  0070-0077 : rtc0
  0080-008f : dma page reg
  00a0-00a1 : pic2
  00c0-00df : dma2
  00f0-00ff : fpu
  0400-041f : iTCO_wdt
    0400-041f : iTCO_wdt
  0680-069f : pnp 00:02
  0a00-0a1f : pnp 00:01
  0a20-0a2f : pnp 00:01
  0a30-0a3f : pnp 00:01
  0a40-0a4f : pnp 00:01
  0a50-0a5f : pnp 00:01
  0a60-0a6f : pnp 00:01
0cf8-0cff : PCI conf1
0d00-ffff : PCI Bus 0000:00
  164e-164f : pnp 00:02
  1800-18fe : pnp 00:02
    1800-1803 : ACPI PM1a_EVT_BLK
    1804-1805 : ACPI PM1a_CNT_BLK
    1808-180b : ACPI PM_TMR
    1830-1833 : iTCO_wdt
      1830-1833 : iTCO_wdt
    1850-1850 : ACPI PM2_CNT_BLK
    1854-1857 : pnp 00:04
    1880-189f : ACPI GPE0_BLK
  2000-2fff : PCI Bus 0000:01
  f000-f03f : 0000:00:02.0
  f040-f05f : 0000:00:1f.4
    f040-f05f : i801_smbus
  f060-f07f : 0000:00:17.0
    f060-f07f : ahci
  f080-f083 : 0000:00:17.0
    f080-f083 : ahci
  f090-f097 : 0000:00:17.0
    f090-f097 : ahci
  ff00-fffe : pnp 00:07
  ffff-ffff : pnp 00:02
    ffff-ffff : pnp 00:02
      ffff-ffff : pnp 00:02

fuzhongl, 2020-07-27:
@ionutnechita Could you please share kernel info with uname -a for ACRN and native boot?

ionutnechita, 2020-07-27:
Hi @fuzhongl,

I used the kernel here:
https://github.com/clearlinux-pkgs/linux-iot-lts2018
Clear Linux SOS
I compiled it as it is there, then I unzipped the archive and I put the files exactly as in the documentation. 
Ubuntu SOS with ACRN hypervisor.
https://projectacrn.github.io/latest/tutorials/using_ubuntu_as_sos.html

4.19.128-109

fuzhongl, 2020-07-28:
@ionutnechita the issue is related with root login.
Please double check it on your side.
root@acrn-NUC7i7DNHE:/home/acrn# cat /proc/ioports
0000-0cf7 : PCI Bus 0000:00
  0000-001f : dma1
  0020-0021 : pic1
  0040-0043 : timer0
  0050-0053 : timer1
  0060-0060 : keyboard
  0062-0062 : PNP0C09:00
    0062-0062 : EC data
  0064-0064 : keyboard
  0066-0066 : PNP0C09:00
    0066-0066 : EC cmd
  0070-0077 : rtc0
  0080-008f : dma page reg
  00a0-00a1 : pic2
  00c0-00df : dma2
  00f0-00ff : fpu
  02f8-02ff : serial
  03f8-03ff : serial
  0400-041f : iTCO_wdt
  0680-069f : pnp 00:02
  0a00-0a1f : pnp 00:00
  0a20-0a2f : pnp 00:00
  0a30-0a3f : pnp 00:00
  0a40-0a4f : pnp 00:00
  0a50-0a5f : pnp 00:00
  0a60-0a6f : pnp 00:00
0cf8-0cff : PCI conf1
0d00-ffff : PCI Bus 0000:00
  164e-164f : pnp 00:02
  1800-18fe : pnp 00:02
    1800-1803 : ACPI PM1a_EVT_BLK
    1804-1805 : ACPI PM1a_CNT_BLK
    1808-180b : ACPI PM_TMR
    1850-1850 : ACPI PM2_CNT_BLK
    1854-1857 : pnp 00:04
    1880-189f : ACPI GPE0_BLK
  2000-2fff : PCI Bus 0000:01
  f000-f03f : 0000:00:02.0
  f040-f05f : 0000:00:1f.4
    f040-f05f : i801_smbus
  f060-f07f : 0000:00:17.0
    f060-f07f : ahci
  f080-f083 : 0000:00:17.0
    f080-f083 : ahci
  f090-f097 : 0000:00:17.0
    f090-f097 : ahci
  ff00-fffe : pnp 00:07
  ffff-ffff : pnp 00:02
    ffff-ffff : pnp 00:02
      ffff-ffff : pnp 00:02
root@acrn-NUC7i7DNHE:/home/acrn# dmesg |grep ACRN
[    0.000000] Hypervisor detected: ACRN
[    1.168600] ACRNTrace: Initialized acrn trace module with 4 cpu
[    1.168726] ACRN HVLog: Failed to init last hvlog devs, errno -19
[    1.168727] ACRN HVLog: Initialized hvlog module with 4 cpu
root@acrn-NUC7i7DNHE:/home/acrn#

fuzhongl, 2020-07-29:
@ionutnechita Could you double check root login and close this issue if verified?

ionutnechita, 2020-07-29:
Not yet.
My setup is broken, I have to redo it.
I used kernel version 4.19.
Do i have to migrate i to the new kernel version too?

How did you create acrn.efi for NUC?
I used like this (NUC7i7BNH):
make BOARD=nuc7i7dnb FIRMWARE=uefi
make install

I will come back with an update, after my setup is ready.
Thanks.

fuzhongl, 2020-07-30:
@ionutnechita  It is fine for your method.
You can also use the script(install_uSoS.py) to build ACRN/kernel and install them automatically.
Please check the doc for detail: https://projectacrn.github.io/latest/misc/packaging/README.html
After reboot, acrn will boot up by default.

fuzhongl, 2020-08-14:
Will close this ticket, you can reopen it if still met this issue.

closed 2020-08-14 by fuzhongl

ionutnechita, 2021-02-02:
Hi @fuzhongl,

This issue is reproduced with:
Linux  5.4.90-PKT-200203T060100Z-g42a5e48c3e11 #1 SMP PREEMPT Tue Feb 2 20:08:45 EET 2021 x86_64 x86_64 x86_64 GNU/Linux

dmesg -l err
[    3.514566] iTCO_wdt iTCO_wdt: can't request region for resource [mem 0x00c5fffc-0x00c5ffff]

cat /proc/ioports
0000-0cf7 : PCI Bus 0000:00
  0000-001f : dma1
  0020-0021 : pic1
  0040-0043 : timer0
  0050-0053 : timer1
  0060-0060 : keyboard
  0062-0062 : PNP0C09:00
    0062-0062 : EC data
  0064-0064 : keyboard
  0066-0066 : PNP0C09:00
    0066-0066 : EC cmd
  0070-0077 : rtc0
  0080-008f : dma page reg
  00a0-00a1 : pic2
  00c0-00df : dma2
  00f0-00ff : fpu
  02f8-02ff : serial
  03f8-03ff : serial
  0400-041f : iTCO_wdt
  0680-069f : pnp 00:02
  0a00-0a1f : pnp 00:01
  0a20-0a2f : pnp 00:01
  0a30-0a3f : pnp 00:01
  0a40-0a4f : pnp 00:01
  0a50-0a5f : pnp 00:01
  0a60-0a6f : pnp 00:01
0cf8-0cff : PCI conf1
0d00-ffff : PCI Bus 0000:00
  164e-164f : pnp 00:02
  1800-18fe : pnp 00:02
    1800-1803 : ACPI PM1a_EVT_BLK
    1804-1805 : ACPI PM1a_CNT_BLK
    1808-180b : ACPI PM_TMR
    1850-1850 : ACPI PM2_CNT_BLK
    1854-1857 : pnp 00:04
    1880-189f : ACPI GPE0_BLK
  2000-2fff : PCI Bus 0000:01
  f000-f03f : 0000:00:02.0
  f040-f05f : 0000:00:1f.4
    f040-f05f : i801_smbus
  f060-f07f : 0000:00:17.0
    f060-f07f : ahci
  f080-f083 : 0000:00:17.0
    f080-f083 : ahci
  f090-f097 : 0000:00:17.0
    f090-f097 : ahci
  ff00-fffe : pnp 00:07
  ffff-ffff : pnp 00:02
    ffff-ffff : pnp 00:02
      ffff-ffff : pnp 00:02

cat /proc/iomem
00000000-00000fff : Reserved
00001000-00057fff : System RAM
00058000-00058fff : Reserved
00059000-0008dfff : System RAM
0008e000-000fffff : Reserved
  000a0000-000bffff : PCI Bus 0000:00
  000c0000-000c3fff : PCI Bus 0000:00
  000c4000-000c7fff : PCI Bus 0000:00
  000c8000-000cbfff : PCI Bus 0000:00
  000cc000-000cffff : PCI Bus 0000:00
  000d0000-000d3fff : PCI Bus 0000:00
  000d4000-000d7fff : PCI Bus 0000:00
  000d8000-000dbfff : PCI Bus 0000:00
  000dc000-000dffff : PCI Bus 0000:00
  000e0000-000e3fff : PCI Bus 0000:00
  000e4000-000e7fff : PCI Bus 0000:00
  000e8000-000ebfff : PCI Bus 0000:00
  000ec000-000effff : PCI Bus 0000:00
  000f0000-000fffff : PCI Bus 0000:00
    000f0000-000fffff : System ROM
00100000-00dfffff : System RAM
  00c5fffc-00c5ffff : iTCO_wdt
00e00000-00ffffff : Reserved
01000000-10ffffff : System RAM
11000000-13ffffff : RAM buffer
20000000-3fffffff : System RAM
40000000-403fffff : Reserved
  40000000-403fffff : pnp 00:00
40400000-709fefff : System RAM
709ff000-709fffff : ACPI Non-volatile Storage
70a00000-70a00fff : Reserved
70a01000-7a10dfff : System RAM
7a10e000-7a5a2fff : Reserved
7a5a3000-7a5e8fff : ACPI Tables
7a5e9000-7aa05fff : ACPI Non-volatile Storage
7aa06000-7af4dfff : Reserved
7af4e000-7affdfff : Unknown E820 type
7affe000-7affefff : System RAM
7afff000-7fffffff : Reserved
  7c000000-7fffffff : Graphics Stolen Memory
80000000-dfffffff : PCI Bus 0000:00
  90000000-9fffffff : 0000:00:02.0
  a0000000-c1ffffff : PCI Bus 0000:01
  c4000000-da0fffff : PCI Bus 0000:01
  db000000-dbffffff : 0000:00:02.0
  dc000000-dc0fffff : PCI Bus 0000:3b
    dc000000-dc000fff : 0000:3b:00.0
      dc000000-dc000fff : rtsx_pci
  dc100000-dc1fffff : PCI Bus 0000:3a
    dc100000-dc101fff : 0000:3a:00.0
  dc200000-dc21ffff : 0000:00:1f.6
    dc200000-dc21ffff : e1000e
  dc220000-dc22ffff : 0000:00:1f.3
    dc220000-dc22ffff : ICH HD audio
  dc230000-dc23ffff : 0000:00:14.0
    dc230000-dc23ffff : xhci-hcd
      dc238070-dc23846f : intel_xhci_usb_sw
  dc240000-dc243fff : 0000:00:1f.3
    dc240000-dc243fff : ICH HD audio
  dc244000-dc247fff : 0000:00:1f.2
  dc248000-dc249fff : 0000:00:17.0
    dc248000-dc249fff : ahci
  dc24a000-dc24a0ff : 0000:00:1f.4
  dc24b000-dc24b7ff : 0000:00:17.0
    dc24b000-dc24b7ff : ahci
  dc24c000-dc24c0ff : 0000:00:17.0
    dc24c000-dc24c0ff : ahci
  dc24d000-dc24dfff : 0000:00:16.0
    dc24d000-dc24dfff : mei_me
  dc24e000-dc24efff : 0000:00:14.2
  dc24f000-dc24ffff : 0000:00:08.0
  dffe0000-dfffffff : pnp 00:05
e0000000-efffffff : PCI MMCONFIG 0000 [bus 00-ff]
  e0000000-efffffff : Reserved
    e0000000-efffffff : pnp 00:05
fd000000-fe7fffff : PCI Bus 0000:00
  fd000000-fdabffff : pnp 00:06
  fdac0000-fdacffff : pnp 00:08
  fdad0000-fdadffff : pnp 00:06
  fdae0000-fdaeffff : pnp 00:08
  fdaf0000-fdafffff : pnp 00:08
  fdb00000-fdffffff : pnp 00:06
  fe000000-fe010fff : Reserved
  fe028000-fe028fff : pnp 00:08
  fe029000-fe029fff : pnp 00:08
  fe036000-fe03bfff : pnp 00:06
  fe03d000-fe3fffff : pnp 00:06
  fe410000-fe7fffff : pnp 00:06
fec00000-fec00fff : Reserved
  fec00000-fec003ff : IOAPIC 0
fed00000-fed00fff : Reserved
  fed00000-fed003ff : HPET 0
    fed00000-fed003ff : PNP0103:00
fed10000-fed17fff : pnp 00:05
fed20000-fed3ffff : pnp 00:05
fed45000-fed8ffff : pnp 00:05
fed90000-fed93fff : pnp 00:05
fee00000-fee00fff : Local APIC
  fee00000-fee00fff : Reserved
ff000000-ffffffff : Reserved
  ff000000-ffffffff : INT0800:00
    ff000000-ffffffff : pnp 00:05
100000000-870ffffff : System RAM
  850000000-851201170 : Kernel code
  851201171-851dde8bf : Kernel data
  851fa0000-8525fffff : Kernel bss
871000000-87effffff : Reserved

acrn-dm --version
DM version is: 2.4-unstable-94a980c9-dirty (daily tag:acrn-2021w05.5-180000p), build by root@2021-02-02 20:42:05

it's fine for reopen this ticket.

[    0.000000] Linux version 5.4.90-PKT-200203T060100Z-g42a5e48c3e11 (root@demo-acrn-opensuse-labl1) (gcc version 10.2.1 20201202 [revision e563687cf9d3d1278f45aaebd03e0f66531076c9] (SUSE Linux)) #1 SMP PREEMPT Tue Feb 2 20:08:45 EET 2021
[    0.000000] Command line: root=/dev/sda2 console=ttyS0 idle=halt rw rootwait console=tty0 kaslr consoleblank=0 no_timer_check quiet loglevel=3 i915.nuclear_pageflip=1 hvlog=2M@0xe00000 memmap=0x200000$0xe00000 maxcpus=4 root=PARTUUID=d444e5a8-ef96-48da-a81b-d8c450b39b35 hugepagesz=1G hugepages=28
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'

fuzhongl, 2021-02-04:
@ionutnechita 
Which scenario are you using?
Some devices are hidden to sos; in case it will reset RTVM by p2sb.
If you really need it, you can refer follow patch:

```
diff --git a/misc/config_tools/library/board_cfg_lib.py b/misc/config_tools/library/board_cfg_lib.py
index ff2b7e67c..f535084fa 100644
--- a/misc/config_tools/library/board_cfg_lib.py
+++ b/misc/config_tools/library/board_cfg_lib.py
@@ -29,6 +29,7 @@ HEADER_LICENSE = common.open_license() + "\n"
 KNOWN_HIDDEN_PDEVS_BOARD_DB = {
     'apl-mrb':['00:0d:0'],
     'apl-up2':['00:0d:0'],
+    'nuc7i7dnb':['00:1f:1'],
 }

 TSN_DEVS = ["8086:4b30", "8086:4b31", "8086:4b32", "8086:4ba0", "8086:4ba1", "8086:4ba2",
```

Then You will see:
root@acrn-NUC7i7DNHE:/home/acrn# dmesg |grep iTCO
```
[   65.000770] iTCO_wdt: Intel TCO WatchDog Timer Driver v1.11
[   65.000922] iTCO_wdt: Found a Intel PCH TCO device (Version=4, TCOBASE=0x0400)
[   65.001084] iTCO_wdt: initialized. heartbeat=30 sec (nowayout=0)
```
root@acrn-NUC7i7DNHE:/home/acrn#

ionutnechita, 2021-02-04:
Hi @fuzhongl,

// Which scenario are you using?
Use now, industrialy scenario.

ionutnechita, 2021-02-04:
Hi @fuzhongl,

I tested with this modification:
diff --git a/misc/config_tools/library/board_cfg_lib.py b/misc/config_tools/library/board_cfg_lib.py
index ff2b7e67c..f535084fa 100644
--- a/misc/config_tools/library/board_cfg_lib.py
+++ b/misc/config_tools/library/board_cfg_lib.py
@@ -29,6 +29,7 @@ HEADER_LICENSE = common.open_license() + "\n"
 KNOWN_HIDDEN_PDEVS_BOARD_DB = {
     'apl-mrb':['00:0d:0'],
     'apl-up2':['00:0d:0'],
+    'nuc7i7dnb':['00:1f:1'],
 }

 TSN_DEVS = ["8086:4b30", "8086:4b31", "8086:4b32", "8086:4ba0", "8086:4ba1", "8086:4ba2",

And now working.

dmesg -l err | wc -l
0

dmesg | grep iTCO
[    3.527418] iTCO_wdt: Intel TCO WatchDog Timer Driver v1.11
[    3.527509] iTCO_wdt: Found a Intel PCH TCO device (Version=4, TCOBASE=0x0400)
[    3.527615] iTCO_wdt: initialized. heartbeat=30 sec (nowayout=0)

Thanks.