projectacrn/acrn-hypervisor #5052: unknown device "ivshmem"

DeepRed1, 2020-07-19:
Hi,

I'm running ACRN v2.0 on a nuc6cayh with Clear Linux (33050) as Service VM and as two User VMs. Yesterday I tried to add an ivshmem PCI Device my User VM and therefore followed your sample:

https://projectacrn.github.io/latest/developer-guides/hld/ivshmem-hld.html

acrn-dm -A -m $mem_size -s 0:0,hostbridge \
 -s 2,pci-gvt -G "$2" \
 -s 5,virtio-console,@stdio:stdio_port \
 -s 6,virtio-hyper_dmabuf \
 -s 3,virtio-blk,/home/clear/uos/uos1.img \
 -s 4,virtio-net,tap0 \
 -s 6,ivshmem,test,4096 \
 -s 7,virtio-rnd \
 --ovmf /usr/share/acrn/bios/OVMF.fd \
 $vm_name

While booting the VM, I received the message: **pci slot 6:0 already occupied!** 
So I deleted the line "-s 6,virtio-hyper_dmabuf \" in the launch script an tried to boot again. Now the message **pci slot 6:0: unknown device "ivshmem"** appeared. 

I already compiled the ACRN .efi again from the master branch but I'm not sure, if I have the right version (2.0) installed to use the ivshmem device, because I started with v1.6 a few weeks ago. Is there a way to check the installed version in the service VM?

na3em, 2020-07-19:
hi @DeepRed1 can you try those two steps and provide update.

1- type `acrn-dm -v` in SOS command line and check DM version. Although this is just the dm version and not the HV, but normally they build together and have same version if you follow the ACRN build instructions.

2- keep the virtio-hyper_dmabuf in slot 6 and use slot 9 for ivshmem
```
acrn-dm -A -m $mem_size -s 0:0,hostbridge
-s 2,pci-gvt -G "$2"
-s 5,virtio-console,@stdio:stdio_port
-s 6,virtio-hyper_dmabuf
-s 3,virtio-blk,/home/clear/uos/uos1.img
-s 4,virtio-net,tap0
-s 9,ivshmem,test,4096
-s 7,virtio-rnd
--ovmf /usr/share/acrn/bios/OVMF.fd
$vm_name
```

DeepRed1, 2020-07-22:
Thank you, that was the command I was looking for. I typed `acrn-dm -v` in my SOS and the DM version is still v1.6, as expected.

So I tried to manually copy over the acrn-dm file from /build/devicemodel  to /usr/bin on my SOS but I still get v1.6 with `acrn-dm -v`. I also pulled the latest changes via `git pull origin master`.

Do I need to setup the NUC from scratch (Host OS, SOS etc.) or is it possible to update ACRN from v1.6 to v2.0?

fuzhongl, 2020-07-27:
Please update ACRN code to v2.0.

DeepRed1, 2020-08-02:
Could you tell me, how to update ACRN code to v2.0? I'm not very experienced in working with Lunix and Github.

I finally managed to copy over the latest acrn-dm file from /build/devicemodel to my SOS. Now `acrn-dm -v` returns v2.1 (unstable). After adjusting the kernel of the User VMs, I can now boot them with ivshmem device in slot 9.
But I still feel like I need to update the remaining acrn-files to 2.0, right? Is there a command to check the installed acrn version?

fuzhongl, 2020-08-02:
@DeepRed1 https://projectacrn.github.io/latest/getting-started/rt_industry_ubuntu.html

1) Get the ACRN source code:
$ cd /home/acrn/work
$ git clone https://github.com/projectacrn/acrn-hypervisor
$ cd acrn-hypervisor

2) Switch to the v2.0 version:
$ git checkout -b v2.0 remotes/origin/release_2.0

fuzhongl, 2020-08-14:
@DeepRed1 Do you still have issue with v2.0?

DeepRed1, 2020-09-20:
Yes, i managed to update to v2.0 and eventually exchanged messages via shared memory (dm-land). Now I will try to establish a shared memory communication with hv-land ivshmem.

fuzhongl, 2020-11-02:
@DeepRed1 Do you still have issue with v2.2?
Thanks!

fuzhongl, 2021-03-15:
Fixed in v2.3

closed 2021-03-15 by fuzhongl