projectacrn/acrn-hypervisor #5103: Cyclictest not working on NUC6CAYH

hidrol, 2020-07-27:
After entering the following command I get:
```
root@clr-preempt-rt ~ # cyclictest -a 1 -p 80 -m -N -D 1h -q -H 30000 --histfile=test.log
WARN: Couldn't setaffinity in main thread: Invalid argument
# /dev/cpu_dma_latency set to 0us
FATAL: No allowable cpus to run on
pid = 3623

```
I am using ACRN master branch.

hidrol, 2020-07-30:
When assign the task to core 0 with `root@clr-preempt-rt ~ # cyclictest -a 0 -p 80 -m -N -D 1h -q -H 30000 --histfile=test.log`
I get no error.

fuzhongl, 2020-08-14:
@hidrol  How many cpu cores you assign to RTVM?

hidrol, 2020-08-14:
@fuzhongl I have two cores assigned. Although I don't quite understand my setup. I have two User VMs running with two cores assigned each, and the Service VM has one Core. How is this possible with 4 physical cores? Is there CPU sharing involved?

gvancuts, 2020-08-14:
@hidrol , yes, you most likely have CPU sharing enabled in your set-up. How did you install ACRN, did you use one of the pre-defined scenario or did you build it by yourself?

What does `cat /proc/cpuinfo` return? One other thing to check is whether hyperthreading is enabled on the platform.

hidrol, 2020-08-26:
@gvancuts I installed ACRN with industry scenario. I have hyperthreading disabled. How can I check which CPUs are being shared between the VMs?
`cat /proc/cpuinfo` in SOS gave me:

```
processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 92
model name	: Intel(R) Celeron(R) CPU J3455 @ 1.50GHz
stepping	: 9
microcode	: 0x38
cpu MHz		: 798.400
cache size	: 1024 KB
physical id	: 0
siblings	: 1
core id		: 0
cpu cores	: 1
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 22
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc rep_good nopl xtopology tsc_reliable nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq monitor est tm2 ssse3 cx16 xtpr sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave rdrand hypervisor lahf_lm 3dnowprefetch cpuid_fault ibrs ibpb stibp fsgsbase tsc_adjust smep erms rdseed smap clflushopt sha_ni xsaveopt xsavec xgetbv1 xsaves dtherm arat pln pts md_clear arch_capabilities
bugs		: monitor spectre_v1 spectre_v2
bogomips	: 2995.20
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:

in User OS:

processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 92
model name	: Intel(R) Celeron(R) CPU J3455 @ 1.50GHz
stepping	: 9
microcode	: 0x38
cpu MHz		: 798.400
cache size	: 1024 KB
physical id	: 0
siblings	: 2
core id		: 0
cpu cores	: 2
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 22
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc rep_good nopl xtopology tsc_reliable nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq monitor est tm2 ssse3 cx16 xtpr sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave rdrand hypervisor lahf_lm 3dnowprefetch cpuid_fault ibrs ibpb stibp fsgsbase tsc_adjust smep erms rdseed smap clflushopt sha_ni xsaveopt xsavec xgetbv1 xsaves dtherm arat pln pts md_clear arch_capabilities
bugs		: monitor spectre_v1 spectre_v2
bogomips	: 2995.20
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:

processor	: 1
vendor_id	: GenuineIntel
cpu family	: 6
model		: 92
model name	: Intel(R) Celeron(R) CPU J3455 @ 1.50GHz
stepping	: 9
microcode	: 0x38
cpu MHz		: 798.400
cache size	: 1024 KB
physical id	: 0
siblings	: 2
core id		: 1
cpu cores	: 2
apicid		: 1
initial apicid	: 1
fpu		: yes
fpu_exception	: yes
cpuid level	: 22
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc rep_good nopl xtopology tsc_reliable nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq monitor est tm2 ssse3 cx16 xtpr sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave rdrand hypervisor lahf_lm 3dnowprefetch cpuid_fault ibrs ibpb stibp fsgsbase tsc_adjust smep erms rdseed smap clflushopt sha_ni xsaveopt xsavec xgetbv1 xsaves dtherm arat pln pts md_clear arch_capabilities
bugs		: monitor spectre_v1 spectre_v2
bogomips	: 2995.20
clflush size	: 64
cache_alignment	: 64
address sizes	: 39 bits physical, 48 bits virtual
power management:
```

gvancuts, 2020-08-26:
In that scenario the RTVM should be started as VM #2 and it will use cores 2 and 3, all other cores are shared between the Service VM (SOS) and other User VMs. The reason you only see one CPU in the SOS is because we take all cores offline except for core 0 before starting the User VMs.

Do you get the error message regardless of which User VM you start cyclictest in?

fuzhongl, 2020-08-27:
@hidrol Need to check if cpu is offline; please share your launch script of all UOS.
And please tell us which code base you are using.
Thanks!

fuzhongl, 2021-05-26:
Closed since no feedback for long time.

closed 2021-05-26 by fuzhongl