projectacrn/acrn-hypervisor #5106: [KBL] cpu1,2,3 Init BLOCKED; SOS fails to launch; vlapic_update_ppr infinite loop

moto-timo, 2020-07-28:
**Describe the bug**
cpu0 does not seem to fully initialize and cpu1,2,3 end up in Init BLOCKED condition

**Platform**
nuc7i7dnb

**Codebase**
ACRN 1.6.1: https://github.com/moto-timo/meta-acrn/tree/timo/secureboot-acrn-1.6
ACRN 2.0: https://github.com/moto-timo/meta-acrn/tree/timo/secureboot
ACRN 2.1: https://github.com/moto-timo/meta-acrn/tree/timo/secureboot-acrn-2.1

**Scenario**
sdc

**To Reproduce**
Steps to reproduce the behavior:
1. git checkout "master" branch of layers in [bblayers.conf](https://gist.githubusercontent.com/moto-timo/006c594dd0ca5be6d19ad0c240c7c3b2/raw/93d3748782c0b17f5cc2c328edcfb3cf873138cb/bblayers.conf)
(you can ignore the last four layers)
2. Checkout one of the branches of meta-acrn from above (Codebase)
3. source <path/to>/poky/oe-init-buildsetup
4. Setup conf/bblayers.conf similar to link in 1.
5. Setup [conf/local.conf](https://gist.githubusercontent.com/moto-timo/006c594dd0ca5be6d19ad0c240c7c3b2/raw/93d3748782c0b17f5cc2c328edcfb3cf873138cb/local.conf)
6. bitbake acrn-image-verified-boot
7. dd image to USB stick and insert into nuc7i7dnb
8. Set AGP aperture to 512 MB in bios
9. Turn Hyper Threading off in bios
10. On first boot, login as root/root, and run acrn-efi-setup.sh script (or edit in /usr/sbin/acrn-efi-setup.sh as needed)
11. efibootmgr -n 0 (or whichever item is "ACRN (Yocto)"
12. systemctl reboot
13. System boots into ACRN efi app, but not to SOS kernel/VM
13. See output in ACRN serial console

**Expected behavior**
cpu1,2,3 initialize completely, SOS kernel loads and SOS VM boots

**Additional context**
In the following log, the beginning is booting to normal \EFI\BOOT\bootx64.efi and normal kernel mode.
The second boot is the failure to complete Init in acrn mode.
This occurs whether or not Secure Boot is enabled in the BIOS.
Very similar results occur on 1.6.1, 2.0 and 2.1 (tip)

[acrn-nuc7i7dnb-sdc.log](https://github.com/projectacrn/acrn-hypervisor/files/4991365/acrn-nuc7i7dnb-sdc.log)

fuzhongl, 2020-08-11:
@moto-timo 
Please share the command about build ACRN SDC scenario.
Thanks!

moto-timo, 2020-08-19:
The SOS kernel command line was in log already attached to this bug:
`Command line:  rootwait console=ttyS0,115200 console=tty0 hvlog=2M@0xE00000 memmap=2M$0xE00000  i915.enable_gvt=1 i915.nuclear_pageflip=1 i915.avail_planes_per_pipe=0x01010F i915.domain_plane_owners=0x011111110000 i915.domain_scaler_owner=0x011100 printk.time=y root=PARTUUID=12345678-9abc-def0-0fed-cba987654321`

moto-timo, 2020-08-19:
The command that builds the hypervisor:
`NOTE: make RELEASE=0 SYSROOT=/srv/build/ttorling/build-secureboot-verity/master-acrn-sos/work/intel_corei7_64-oe-linux/acrn-hypervisor/2.0-r0/recipe-sysroot O=/srv/build/ttorling/build-secureboot-verity/master-acrn-sos/work/intel_corei7_64-oe-linux/acrn-hypervisor/2.0-r0/build/ DESTDIR=/srv/build/ttorling/build-secureboot-verity/master-acrn-sos/work/intel_corei7_64-oe-linux/acrn-hypervisor/2.0-r0/image prefix=/usr bindir=/usr/bin libdir=/usr/lib datadir=/usr/share includedir=/usr/include systemd_unitdir=/lib/systemd HV_OBJDIR=/srv/build/ttorling/build-secureboot-verity/master-acrn-sos/work/intel_corei7_64-oe-linux/acrn-hypervisor/2.0-r0/build//hypervisor EFI_OBJDIR=/srv/build/ttorling/build-secureboot-verity/master-acrn-sos/work/intel_corei7_64-oe-linux/acrn-hypervisor/2.0-r0/build//efi-stub BOARD=nuc7i7dnb FIRMWARE=uefi SCENARIO=sdc BOARD_FILE=/srv/build/ttorling/build-secureboot-verity/master-acrn-sos/work/intel_corei7_64-oe-linux/acrn-hypervisor/2.0-r0/git//misc/acrn-config/xmls/board-xmls/nuc7i7dnb.xml SCENARIO_FILE=/srv/build/ttorling/build-secureboot-verity/master-acrn-sos/work/intel_corei7_64-oe-linux/acrn-hypervisor/2.0-r0/git//misc/acrn-config/xmls/config-xmls/nuc7i7dnb/sdc.xml -C hypervisor`

fuzhongl, 2020-08-19:
@moto-timo please remote hvlog=2M@0xE00000 memmap=2M$0xE00000 in sos kernel cmdline
or add escape character before  $; lust like memmap=2M\$0xE00000

moto-timo, 2020-08-19:
When we build with grub-efi [1], we escape the $, but I am booting with only systemd-boot. If I escape \$ it does not boot with systemd-boot.

I removed hvlog and memmap arguments and get the exact same behavior. As before, first boot is non-acrn, second boot is with acrn.efi and displays the behavior.

[acrn-image-verified-boot.20200818.1.log](https://github.com/projectacrn/acrn-hypervisor/files/5094000/acrn-image-verified-boot.20200818.1.log)

[1] https://github.com/intel/meta-acrn/blob/8b2bbaa691294dd11cd6442d1d1963e763f4c8a0/conf/distro/acrn-demo-sos.conf#L15

moto-timo, 2020-08-19:
I don't think we are ever getting out of EFI,  if you look at the log, it is caught in an infinite loop.

https://github.com/projectacrn/acrn-hypervisor/blob/28c9e357711c408a274f53c86ff7d67fec3572f1/hypervisor/arch/x86/guest/vlapic.c#L802

fuzhongl, 2020-08-20:
With rebuild acrn.efi; this issue is narrow down to secure boot related.

fuzhongl, 2020-11-02:
@moto-timo Can this ticket be closed?
Thanks!

closed 2021-01-07 by fuzhongl

hongxuzhen, 2021-04-07:
@fuzhongl Would you give the detail explaination about "secure boot related"? I am faced with the same problem,but have no idea. following file is the log.
[20210401.log](https://github.com/projectacrn/acrn-hypervisor/files/6270671/20210401.log)

fuzhongl, 2021-04-08:
@hongxuzhen The original issue is about enabling security boot in BIOS.
Do you also enable security boot in BIOS?
Which platform and codebase are you using?
Could you share `vcpu_list` info under hypervisor console?

hongxuzhen, 2021-04-08:
> @hongxuzhen The original issue is about enabling security boot in BIOS.
> Do you also enable security boot in BIOS?
> Which platform/scenario and codebase are you using?
> Could you share `vcpu_list` info under hypervisor console?

Dear fulongl,
        Sorry，Now I have not found how to enter the hypervisor console. I will continue to study the doc hard. or Can you tell the way?
      the detail as follow.
       Platform: nuc7i7dnb
       Code Base:release v2.3 or release v2.4
       BIOS Setting: HT and security boot disable，VMX and Vt-d enable
       platform/scenario xml file as shown in the attachment 
       The command that builds the hypervisor:
          make BOARD=/home/acrn/hongxu/acrn/v2.4/acrn-hypervisor/nuc7i7dnb.xml SCENARIO=/home/acrn/hongxu/acrn/v2.4/acrn-hypervisor/sdc.xml
       
[nuc7i7dnb.log](https://github.com/projectacrn/acrn-hypervisor/files/6275547/nuc7i7dnb.log)
[sdc.log](https://github.com/projectacrn/acrn-hypervisor/files/6275548/sdc.log)

fuzhongl, 2021-04-08:
@hongxuzhen 
From you shared  20210401.log; 

```
Multiboot2 Bootloader: GRUB 2.04
[0us][cpu=0][(null)][sev=6][seq=1]:mmap addr 0x20c00080 entries 24
[0us][cpu=0][(null)][sev=6][seq=2]:mmap table: 0 type: 0x1
[0us][cpu=0][(null)][sev=6][seq=3]:Base: 0x0000000000000000 length: 0x0000000000058000
```

You already in hypervisor console. 
Just hit `ENTER` key and input `vcpu_list`

hongxuzhen, 2021-04-08:
@fuzhongl 
   Thanks for your help， vpulist as follow

VM ID    PCPU ID    VCPU ID    VCPU ROLE    VCPU STATE    THREAD STATE
=====    =======    =======    =========    ==========    ==========
  0         0          0       PRIMARY      Running          RUNNING
  0         1          1       SECONDARY    Init             BLOCKED
  0         2          2       SECONDARY    Init             BLOCKED
  0         3          3       SECONDARY    Init             BLOCKED

fuzhongl, 2021-04-12:
@hongxuzhen How many memory on you board?
Could you please share the `acrn.bin` for me to reproduce?

Thanks!

hongxuzhen, 2021-04-19:
> @hongxuzhen How many memory on you board?
> Could you please share the `acrn.bin` to me to reproduce?
> 
> Thanks!

Dear fuzhongl,  Sorry for later，memory = 16G, acrn.bin 
[acrn_bin.tar.gz](https://github.com/projectacrn/acrn-hypervisor/files/6334753/acrn_bin.tar.gz)
as follow

fuzhongl, 2021-04-19:
@hongxuzhen  Only 45 byte for your uploaded acrn_v2.4.bin
There must be something wrong,  please re-uploaded again.
Thanks!

hongxuzhen, 2021-04-19:
> @hongxuzhen Only 45 byte for your uploaded acrn_v2.4.bin
> There must be something wrong, please re-uploaded again.
> Thanks!
Sorry,Please check the new one
[acrn_bin.tar.gz](https://github.com/projectacrn/acrn-hypervisor/files/6335044/acrn_bin.tar.gz)

fuzhongl, 2021-04-20:
> > @hongxuzhen Only 45 byte for your uploaded acrn_v2.4.bin
> > There must be something wrong, please re-uploaded again.
> > Thanks!
> > Sorry,Please check the new one
> > [acrn_bin.tar.gz](https://github.com/projectacrn/acrn-hypervisor/files/6335044/acrn_bin.tar.gz)

@hongxuzhen 
You shared acrn.bin can't work on my side. From the  HV version info, it seems like your build isn't the final v2.4 code: unstable info isn't for release branch.

```
ACRN Hypervisor
calibrate_tsc, tsc_khz=2100000
[36030354us][cpu=0][(null)][sev=2][seq=83]:**HV version 2.4-unstable-2021-04-07 19:17:43-3991c4828 DBG** (daily tag:acrn-2021w14.5-180000p) sdc@nuc7i7dnb build by root, start time 35354246us
```

I build with v2.4; it works well. Please check boot log for detail.
sudo make clean && sudo make hypervisor BOARD=nuc7i7dnb SCENARIO=sdc

[Multiboot2 Bootloader: GRUB 2.02-2ubuntu8.17
[0us][cpu=0][(null)][sev=3][seq=1]:vlapic: Basic apicv ops is used!
ACRN Hypervisor
calibrate_tsc, tsc_khz=2100000
[36267987us][cpu=0][(null)][sev=2][seq=2]:**HV version 2.4-2021-04-20 20:17:47-c80508fd7 DBG** (daily tag:acrn-2021w14.5-180000p) sdc@nuc7i7dnb build by root, start time 36256374us

[29178152us][cpu=0][(null)][sev=2][seq=3]:API version 1.0
[29183908us][cpu=0][(null)][sev=2][seq=4]:Detect processor: Intel(R) Core(TM) i7-8650U CPU @ 1.90GHz
[29458116us][cpu=0][(null)][sev=2][seq=29]:Start VM id: 0 name: ACRN SOS VM
[29465578us][cpu=0][vm0:vcpu0][sev=3][seq=30]:VMX ctrl 0x482 not fully enabled: request 0x92220088 but get 0x9621e1fa

ACRN:\>[30135577us][cpu=0][vm0:vcpu0][sev=3][seq=33]:vlapic: Start Secondary VCPU1 for VM[0]...
[30144248us][cpu=1][vm0:vcpu1][sev=3][seq=34]:VMX ctrl 0x482 not fully enabled: request 0x92220088 but get 0x9621e1fa

[30157285us][cpu=0][vm0:vcpu0][sev=3][seq=36]:vlapic: Start Secondary VCPU2 for VM[0]...
[30165943us][cpu=2][vm0:vcpu2][sev=3][seq=37]:VMX ctrl 0x482 not fully enabled: request 0x92220088 but get 0x9621e1fa

[30178864us][cpu=0][vm0:vcpu0][sev=3][seq=39]:vlapic: Start Secondary VCPU3 for VM[0]...
[30187518us][cpu=3][vm0:vcpu3][sev=3][seq=40]:VMX ctrl 0x482 not fully enabled: request 0x92220088 but get 0x9621e1fa

[30578115us][cpu=2][vm0:vcpu2][sev=2][seq=42]:vpci_write_cfg 0:14.1 not found! off: 0x85, val: 0xfe

[33189731us][cpu=0][vm0:vcpu0][sev=2][seq=43]:vpci_write_cfg 0:1f.1 not found! off: 0xe1, val: 0x0

[33199622us][cpu=0][vm0:vcpu0][sev=2][seq=44]:vpci_write_cfg 0:1f.1 not found! off: 0xe1, val: 0xff

ACRN:\>
ACRN:\>
ACRN:\>vcpu_list

VM ID    PCPU ID    VCPU ID    VCPU ROLE    VCPU STATE    THREAD STATE
=====    =======    =======    =========    ==========    ==========
  0         0          0       PRIMARY      Running          BLOCKED
  0         1          1       SECONDARY    Running          BLOCKED
  0         2          2       SECONDARY    Running          BLOCKED
  0         3          3       SECONDARY    Running          BLOCKED
ACRN:\>vm_console

----- Entering VM 0 Shell -----

Ubuntu 18.04.3 LTS acrn-NUC7i7DNKE ttyS0

acrn-NUC7i7DNKE login: acrn
Password:

Last login: Mon Apr 19 19:59:04 EDT 2021 from 10.238.12.223 on pts/0
Welcome to Ubuntu 18.04.3 LTS (GNU/Linux 5.4.90-PKT-200203T060100Z-dirty x86_64)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage

 * Canonical Livepatch is available for installation.
   - Reduce system reboots and improve kernel security. Activate at:
     https://ubuntu.com/livepatch

152 packages can be updated.
0 updates are security updates.

New release '20.04.2 LTS' available.
Run 'do-release-upgrade' to upgrade to it.

Your Hardware Enablement Stack (HWE) is supported until April 2023.
acrn@acrn-NUC7i7DNKE:~$

fuzhongl, 2021-04-20:
@hongxuzhen Could you please check if sos boots up on your board after adding  `nokaslr` in kernel cmdline?
Thanks!

hongxuzhen, 2021-04-21:
> @hongxuzhen
> You shared acrn.bin can't work on my side. From the HV version info, it seems like your build isn't the final v2.4 code: unstable info isn't for release branch.
>
> ```
> ACRN Hypervisor [...]

With rebuild acrn.efi; this issue has be solved. I build acrn.efi with 'git checkout release_v2.4' before that time, may be the root cause is this.
thanks a lot for your help!

fuzhongl, 2021-04-21:
@hongxuzhen Thanks for feedback.