projectacrn/acrn-hypervisor #5112: ACRN debug console help output behavior, line length, and misspellings

dbkinder, 2020-07-29:
**Describe the bug**

In the output of the debug console's ``help`` command, after 10 lines of output there is a line written saying, 
``<*** Hit any key to continue ***>``.  There's no pause in the output waiting for a key to be pressed.

Some output lines are very long and wrap, making it hard to read.

"pass-through" and "Setled" are misspelled words: should be "passthrough" and I'm not sure what "Setled" should be.

**Expected behavior**
The output should pause, waiting for a key to be pressed, and then continue.  (Or just remove that feature since there's really no need to pause after 10 lines.)  Long output lines should be broken so they don't wrap on an 80-character-width console. No misspellings.

**Additional context**

```
ACRN:\>help

Registered Commands:

  help             - Display information about supported hypervisor shell commands
  version          - Display the HV version information
  vm_list          - List all VMs, displaying the VM UUID, ID, name and state
  vcpu_list        - List all vCPUs in all VMs
  vcpu_dumpreg    <vm id, vcpu id> - Dump registers for a specific vCPU
  dump_host_mem   <addr, length> - Dump host memory, starting at a given address(Hex), and for a given length (Dec in bytes)
  dump_guest_mem  <vm_id, addr, length> - Dump guest memory, vm id(Dec), starting at a given address(Hex), and for a given length (Dec in bytes)
  vm_console      <vm id> - Switch to the VM's console. Use [Ctrl+Spacebar] to return to the ACRN shell console
  int              - List interrupt information per CPU
<*** Hit any key to continue ***>
  pt               - Show pass-through device information
  vioapic         <vm id> - Show virtual IOAPIC (vIOAPIC) information for a specific VM
  dump_ioapic      - Show native IOAPIC information
  loglevel        [<console_loglevel> [<mem_loglevel> [npk_loglevel]]] - No argument: get the level of logging for the console, memory and npk. Setel by giving (up to) 3 parameters between 0 and 6 (verbose)
  cpuid           <leaf> [subleaf] - Display the CPUID leaf [subleaf], in hexadecimal
  reboot           - Trigger a system reboot (immediately)
  rdmsr           [-p<pcpu_id>] <msr_index> - Read the MSR at msr_index (in hexadecimal) for CPU ID pcpu_id
  wrmsr           [-p<pcpu_id>] <msr_index> <value> - Write value (in hexadecimal) to the MSR at msr_index (in hexadecimal) for CPU ID pcpu_id

```

gvancuts, 2021-03-15:
Not sure if *any* of the problems mentioned here has been fixed, but at least *some* haven't (e.g. misspelling of `pass-through`).

@fuzhongl , can you please add the commits that may have fixed part of the issues in this report?

dbkinder, 2021-05-25:
@NanlinXie can this be fixed in v2.5 ?

NanlinXie, 2021-06-20:
Not planned in v2.5.

gvancuts, 2021-06-21:
@NanlinXie can you add which release this is planned for? It has been opened for close to a year now, I think it's time we address this. Thanks

liudlong, 2021-09-01:
[image]

@gvancuts How about this output format ? any comments

gvancuts, 2021-09-01:
Looks good to me! 👍

liudlong, 2021-09-01:
OK, I will send the patch to the maillist, please help review.

liudlong, 2021-09-03:
[External_System_ID]ACRN-7386

dbkinder, 2021-09-06:
> Looks good to me! 👍

The output won't look very good on an 80-character terminal (you've consumed 72 characters before you get to the description).  I'm not sure what the supported terminal width requirement should be, but the usual is 80 characters and your output has lines that are more than 200 characters wide.  Line wrapping will make this output confusing to read (DX issue).

dbkinder, 2022-02-23:
I've verified the acrn shell help output looks better now, but there are line breaks at non-word spots that are causing a letter to be missing.  For example the word "given" shows up as "gi"at the end of a line, followed by "en" (missing the v) on the next line:

```
ACRN:\>help

Registered Commands: 

 help
               Display information about supported hypervisor shell commands
 version
               Display the HV version information
 vm_list
               List all VMs, displaying the VM UUID, ID, name and state
 vcpu_list
               List all vCPUs in all VMs
 vcpu_dumpreg   <vm id, vcpu id>
               Dump registers for a specific vCPU
 dump_host_mem  <addr, length>
               Dump host memory, starting at a given address(Hex), and for a gi
               en length (Dec in bytes)
 dump_guest_mem <vm_id, addr, length>
               Dump guest memory, vm id(Dec), starting at a given address(Hex),
               and for a given length (Dec in bytes)
 vm_console     <vm id>
               Switch to the VM's console. Use [Ctrl+Spacebar] to return to the
               ACRN shell console
 int
               List interrupt information per CPU
 pt
               Show pass-through device information
 vioapic        <vm id>
               Show virtual IOAPIC (vIOAPIC) information for a specific VM
 dump_ioapic
               Show native IOAPIC information
 loglevel       [<console_loglevel> [<mem_loglevel> [npk_loglevel]]]
               No argument: get the level of logging for the console, memory an
                npk. Set the level by giving (up to) 3 parameters between 0 and
               6 (verbose)
 cpuid          <leaf> [subleaf]
               Display the CPUID leaf [subleaf], in hexadecimal
 reboot
               Trigger a system reboot (immediately)
 rdmsr          [-p<pcpu_id>]   <msr_index>
               Read the MSR at msr_index (in hexadecimal) for CPU ID pcpu_id
 wrmsr          [-p<pcpu_id>]   <msr_index> <value>
               Write value (in hexadecimal) to the MSR at msr_index (in hexadec
               mal) for CPU ID pcpu_id

ACRN:\>
```