projectacrn/acrn-hypervisor #5151: [WHL][VxWorks] Launch VxWorks fails due to no suitable video mode found error

fuzhongl, 2020-08-07:
**Platform**
WHL

**Codebase**
v2.1

**Scenario**
 Industry

Steps:
1. Launch SOS
2. Use the script to launch Vxworks.

Expected Result:
Launch vxworks success.

Actual result:
Launch vxworks fails due to error detected from the vxworks grub:

fuzhongl, 2020-08-07:
[External_System_ID] ACRN-6557

gvancuts, 2020-11-18:
Error thrown when attempting to lauch VxWorks:
```
WARNING: no console will be available to OS
error: no suitable video mode found.
initialize CPU
start CPU
initialize MMU
initialize base virtual memory support
initialize address space library
initialize global mapping
initialize VxBus
        install bus types:
                vxbHpetBus(Intel Architecture HPET Bus)
                vxbMiiBus(MII bus type)
```

The problem was introduced with this PR #4692

fuzhongl, 2021-03-15:
fixed with v2.2

closed 2021-03-15 by fuzhongl