projectacrn/acrn-hypervisor #5164: Windows as the Guest VM no display

bigchenla, 2020-08-10:
Hi I have an issue about display for installing Windows 10.

I use branch release_2.0 code and follow [Launch Windows as the Guest VM on ACRN](https://projectacrn.github.io/latest/tutorials/using_windows_as_uos.html#using-windows-as-uos) to create a Guest VM.
After I edit install_win.sh and run it, system doesn't show anything.
I can see an UOS created when use ACRN shell console command "vm_list" ( Tera term ).
If I use command "vm_console 1", I can see the UEFI shell. But after input exit, system still can't show Windows install page.
Did I miss any steps for display? (Weston or something)

fuzhongl, 2020-08-14:
@shenfang2019  Please help to check this issue.
Thanks!

shenfang2019, 2020-08-14:
@bigchenla Could you share me your install_win.sh and cmdline in /proc/cmdline? thanks.

bigchenla, 2020-08-14:
acrn@acrn-WhiskeyLake-U:~$ cat /proc/cmdline
pci_devices_ignore=(0:18:1) console=tty0 console=ttyS0 rw rootwait ignore_loglevel no_timer_check consoleblank=0 i915.nuclear_pageflip=1 i915.domain_plane_owners=0x011111110000 i915.enable_gvt=1 i915.enable_guc=0 root=PARTUUID=94b62486-d4ef-431f-b396-005b8a00d4e7 hugepagesz=1G hugepages=4

[install_win.sh.txt](https://github.com/projectacrn/acrn-hypervisor/files/5072460/install_win.sh.txt)
Thanks

shenfang2019, 2020-08-14:
@bigchenla, please change all i915 related cmdline to be i915.nuclear_pageflip=1 i915.avail_planes_per_pipe=0x010101 i915.domain_plane_owners=0x011100001111  i915.enable_gvt=1 i915.enable_guc=0, and have a try again.

bigchenla, 2020-08-17:
@shenfang2019, thanks for your help.
I try to change the parameters as you say, and my screen hang and flash when enter SOS.
I use ACRN shell to run install_win.sh at this time, screen can show UOS display. (The Window installation can't find the disk to install, it is another question.)
But I can't show SOS now, I expect the result is HDMI-1 display SOS, HDMI-2 UOS display.

shenfang2019, 2020-08-17:
@bigchenla, if you need 2 display, please change i915.domain_plane_owners=0x011111110000(your original parameter). About Installation disk, make sure your ISO is right.

shenfang2019, 2020-09-02:
@bigchenla is your issue fixed？V2.2 is coming soon，gvt will be better.

dbkinder, 2021-02-10:
Support questions, not a documentation change request.

closed 2021-02-10 by dbkinder