projectacrn/acrn-hypervisor #5188: Problem with Using Windows as Guest VM on ACRN

SIMING29, 2020-08-18:
Following the instructions of the doc https://projectacrn.github.io/1.6/tutorials/using_windows_as_uos.html , but it did not show windows installation on the HDMI screen.

This is my build steps:
 My board is NUC7i5BNB, Clear linux verson is 32680, I update the hypervisor and kernel to release 2.0 with the boot menu **ACRNGT Service VM**.
```
root@clr-sata~ # uname -a
Linux clr-sata 5.4.43-PKT-200203T060100Z #3 SMP PREEMPT Tue Aug 4 10:24:36 CST 2020 x86_64 GNU/Linux
root@clr-sata~ # acrn-dm -v
DM version is: 2.0-64896ac9 (daily tag:acrn-2020w25.7-180000p), build by acrn@2020-08-04 10:38:21
```

I login the NUC with ths ssh on another pc as the second monitor. The first monitor is the hdmi of the NUC.
There is something wrong when I set the weston desktop.
```
$ useradd cl_uos
$ passwd cl_uos
$ usermod -G wheel -a cl_uos
$ systemctl enable weston@cl_uos
$ systemctl start weston@cl_uos
```

reboot into Clear Linux OS (acrn) --the origin SOS,  the system show the desktop, like the picture1.
reboot into Tht ACRNGT Service VM --the updated  hypervisor and kernel SOS, the system show the login screen but can not input , like the picture2.
Run the install_win.sh in the ACRNGT Service VM,  I think the UOS is running according to the log, but the HDMI show like the picture3.

I'm not sure the set of the weston desktop and use the ssh as the second monitor.
Is there something wrong in the above steps?

my acrngt.conf is different from the instructions of the doc.  
add 
```
+ initrd /EFI/org.clearlinux/freestanding-00-intel-ucode.cpio
+ initrd /EFI/org.clearlinux/freestanding-i915-firmware.cpio.xz
```
otherwise the ACRNGT Service can't bootup.

Thanks
SiMing

[acrngt.conf.txt](https://github.com/projectacrn/acrn-hypervisor/files/5087749/acrngt.conf.txt)

[install_win.sh.txt](https://github.com/projectacrn/acrn-hypervisor/files/5087750/install_win.sh.txt)

picture1:
[image]

picture2:
[image]

picture3:
[image]

```
root@clr-sata~/vm-win # ./install_win.sh 
cpu1 online=1
cpu2 online=1
cpu3 online=1
acrn-dm commond
passed gvt-g optargs low_gm 64, high_gm 448, fence 8
SW_LOAD: get ovmf path /usr/share/acrn/bios/OVMF.fd, size 0x200000
vm_create: win_vm1
VHM api version 1.0
vm_setup_memory: size=0x100000000
open hugetlbfs file /run/hugepage/acrn/huge_lv1/win_vm1/D279543825D611E8864ECB7A18B34643
open hugetlbfs file /run/hugepage/acrn/huge_lv2/win_vm1/D279543825D611E8864ECB7A18B34643
level 0 free/need pages:0/1 page size:0x200000
level 1 free/need pages:0/4 page size:0x40000000
to reserve more free pages:
to reserve pages (+orig 0): echo 4 > /sys/kernel/mm/hugepages/hugepages-1048576kB/nr_hugepages
to reserve pages (+orig 0): echo 1 > /sys/kernel/mm/hugepages/hugepages-2048kB/nr_hugepages
now enough free pages are reserved!

try to setup hugepage with:
level 0 - lowmem 0x0, biosmem 0x200000, highmem 0x0
level 1 - lowmem 0x80000000, biosmem 0x0, highmem 0x80000000
total_size 0x200000000

mmap ptr 0x0x7f7b17e5a000 -> baseaddr 0x0x7f7b40000000
mmap 0x80000000@0x7f7b40000000
touch 2 pages with pagesz 0x40000000
mmap 0x80000000@0x7f7c80000000
touch 2 pages with pagesz 0x40000000
mmap 0x200000@0x7f7c3fe00000
touch 1 pages with pagesz 0x200000

really setup hugepage with:
level 0 - lowmem 0x0, biosmem 0x200000, highmem 0x0
level 1 - lowmem 0x80000000, biosmem 0x0, highmem 0x80000000
vm_init_vdevs
No correct pm notify channel given
pci init hostbridge
pci init lpc
pci init pci-gvt
polling 34...
             Listening 34...
                            pci init virtio-blk
pci init virtio-net
pci init xhci
pci init ahci
pci init ahci
tpm: init_vtpm2:Invalid socket path!
                                    /tmp/dm.XmFQcNt     24:     Device (PCI0)
                                                                             Warning  3073 -        Multiple types ^  (Device object requires either a _HID or _ADR, but not both)

       /tmp/dm.XmFQcNt    827:     Processor (CPU0, 0x00, 0x00000000, 0x00) {}
                                                                              Warning  3168 -                          ^ Legacy Processor() keyword detected. Use Device() keyword instead.

                acrn_sw_load
                            SW_LOAD: partition blob /usr/share/acrn/bios/OVMF.fd size 2097152 copy to guest 0xffe00000
                                                                                                                      SW_LOAD: build e820 9 entries to addr: 0x7f7b400ef008
SW_LOAD: entry[0]: addr 0x0000000000000000, size 0x00000000000a0000,  type 0x1
SW_LOAD: entry[1]: addr 0x00000000000a0000, size 0x0000000000060000,  type 0x2
SW_LOAD: entry[2]: addr 0x0000000000100000, size 0x000000007ff00000,  type 0x1
SW_LOAD: entry[3]: addr 0x0000000080000000, size 0x0000000008000000,  type 0x2
SW_LOAD: entry[4]: addr 0x00000000db000000, size 0x0000000004000000,  type 0x2
SW_LOAD: entry[5]: addr 0x00000000df000000, size 0x0000000001000000,  type 0x2
SW_LOAD: entry[6]: addr 0x00000000e0000000, size 0x0000000020000000,  type 0x2
SW_LOAD: entry[7]: addr 0x0000000100000000, size 0x0000000040000000,  type 0x2
SW_LOAD: entry[8]: addr 0x0000000140000000, size 0x0000000080000000,  type 0x1
SW_LOAD: ovmf_entry 0xfffffff0
                              add_cpu
```

fuzhongl, 2020-08-19:
@SIMING29  Like mentioned in: https://github.com/projectacrn/acrn-hypervisor/issues/5164
please change all i915 related cmdline to be i915.nuclear_pageflip=1 i915.avail_planes_per_pipe=0x010101 i915.domain_plane_owners=0x011100001111 i915.enable_gvt=1 i915.enable_guc=0, and have a try again.

SIMING29, 2020-08-19:
@fuzhongl I haved tried that, but it's the same result.
$ cat /proc/cmdline 
initrd=\EFI\org.clearlinux\freestanding-00-intel-ucode.cpio initrd=\EFI\org.clearlinux\freestanding-i915-firmware.cpio.xz console=tty0 console=ttyS0 root=/dev/sda3 rw rootwait ignore_loglevel no_timer_check consoleblank=0  i915.nuclear_pageflip=1 i915.avail_planes_per_pipe=0x010101 i915.domain_plane_owners=0x011100001111 i915.enable_gvt=1 i915.enable_guc=0 hvlog=2M@0x1FE00000

I think maybe it's the problem about the setting of weston desktop, please help to confim this, thanks. 

when I run install_win.sh, get the errors in the dmesg  
[dmesg.log](https://github.com/projectacrn/acrn-hypervisor/files/5094388/dmesg.log)

```
[    8.349739] acrn-br0: port 2(eno1) entered forwarding state
[   91.520429] smpboot: CPU 1 is now offline
[   92.538749] vhm: try to offline cpu 1 with lapicid 2
[   92.558219] smpboot: CPU 2 is now offline
[   93.563151] vhm: try to offline cpu 2 with lapicid 1
[   93.589498] smpboot: CPU 3 is now offline
[   94.596316] vhm: try to offline cpu 3 with lapicid 3
[   94.660746] vhm_dev_open: opening device node
[   94.663088] vhm-ioreq: init request buffer @ 0000000067438357!
[   94.663347] vhm-ioreq: created ioreq client 1 for ioeventfd-1
[   94.663633] ACRN vhm ioeventfd init done!
[   94.664099] ACRN vhm irqfd init done!
[   94.664266] vhm: VM 1 created
[   95.037525] vhm-ioreq: created ioreq client 2 for ioreq gvt-g
[   95.041399] IPv6: ADDRCONF(NETDEV_CHANGE): tap0: link becomes ready
[   95.041731] acrn-br0: port 1(tap0) entered blocking state
[   95.041982] acrn-br0: port 1(tap0) entered forwarding state
[   95.121274] vhm-ioreq: created ioreq client 3 for acrndm
[   95.277442] [drm] prepare GOP fb: 5625KB for 1600X900@32
[   95.282826] [drm] Set up display w:1600 h:900 for GOP
[   95.283204] gvt: vgpu 1: pvinfo read gop: [7886c:4] = df000000
[   95.283473] gvt: vgpu 1: pvinfo read gop: [78870:4] = 640
[   95.283950] gvt: vgpu 1: pvinfo read gop: [78874:4] = 384
[   95.284201] gvt: vgpu 1: pvinfo read gop: [78878:4] = 640
[   95.284451] gvt: vgpu 1: pvinfo read gop: [7887c:4] = 4
[   95.284949] gvt: vgpu 1: pvinfo read gop: [78880:4] = 57e400
[   95.287817] [drm:gen8_de_irq_handler] *ERROR* Fault errors on pipe A: 0x00000080
[   95.288146] [drm:gen8_de_irq_handler] *ERROR* Fault errors on pipe A: 0x00000080
[   95.304478] [drm:gen8_de_irq_handler] *ERROR* Fault errors on pipe A: 0x00000080
[   95.304822] [drm:gen8_de_irq_handler] *ERROR* Fault errors on pipe A: 0x00000080
[   95.321130] [drm:gen8_de_irq_handler] *ERROR* Fault errors on pipe A: 0x00000080
[   95.321462] [drm:gen8_de_irq_handler] *ERROR* Fault errors on pipe A: 0x00000080
[   95.337801] [drm:gen8_de_irq_handler] *ERROR* Fault errors on pipe A: 0x00000080
........
```

fuzhongl, 2020-08-19:
@shenfang2019 Please help to check this issue.
Thanks!

SIMING29, 2020-08-19:
the error " **[drm:gen8_de_irq_handler] *ERROR* Fault errors on pipe A: 0x00000080**" is gone, when I disable the weston desktop, but the hdmi output show nothing after run install_win.sh

systemctl stop weston@cl_uos
systemctl disable weston@cl_uos

shenfang2019, 2020-08-19:
@SIMING29  To create windows image need not enable weston. And do you use 2 screens, hdmi for sos, dp for uos?

SIMING29, 2020-08-19:
@shenfang2019  Just have the hdmi for the SOS. 
NUC7i5BNB have HDMI and USB Type C to dispaly, but I don't have a Type C interface screen.

--

shenfang2019, 2020-08-19:
@SIMING29 may I ask if the 1.6 release is ok for you to create win10 image?  If it's only 2.0 version's issue, we can follow the latest Doc to create win10 image based on ubuntu.

SIMING29, 2020-08-20:
@shenfang2019 the 1.6 release did not show windows installation on the HDMI screen too. But there is no gvt info in dmesg after run  install_win.sh
[dmesg1.6.log](https://github.com/projectacrn/acrn-hypervisor/files/5100161/dmesg1.6.log)

$ uname -a
Linux clr-sata 4.19.97-104.iot-lts2018-sos #1 SMP PREEMPT Wed Jan 22 14:37:18 PST 2020 x86_64 GNU/Linux
$ acrn-dm -v
DM version is: 1.6-unstable-2020w12.5.140000p_273 (daily tag:acrn-2020w12.5.140000p), build by mockbuild@2020-03-20 01:28:15

shenfang2019, 2020-08-20:
@SIMING29 from your dmesg log, your command line is "initrd=\EFI\org.clearlinux\freestanding-00-intel-ucode.cpio initrd=\EFI\org.clearlinux\freestanding-i915-firmware.cpio.xz root=PARTUUID=d9d05578-6e81-4c79-81c9-c5e542ced1a4 quiet console=tty0 console=ttyS0,115200n8 consoleblank=0 cryptomgr.notests hvlog=2M@0x1FE00000 i915.avail_planes_per_pipe=0x01010F i915.domain_plane_owners=0x011111110000 i915.enable_guc=0 i915.enable_gvt=1 i915.nuclear_pageflip=1 ignore_loglevel init=/usr/lib/systemd/systemd-bootchart intel_iommu=igfx_off memmap=2M$0x1FE00000 no_timer_check no_timer_check noreplace-smp rcu_nocbs=0-64 rcupdate.rcu_expedited=1 rootfstype=ext4,btrfs,xfs rootwait tsc=reliable rw". 

Could you please use our cmdline for V1.6? "console=tty0 console=ttyS0 root=/dev/nvme0n1p3 rw rootwait ignore_loglevel no_timer_check consoleblank=0 i915.nuclear_pageflip=1 i915.avail_planes_per_pipe=0x010101 i915.domain_plane_owners=0x011100001111 i915.enable_gvt=1 i915.enable_conformance_check=0 i915.enable_guc=0 hvlog=2M@0x1FE00000"

We tried V1.6 with CL on KBL, display is ok with only one hdmi screen. WaaG will replace SOS.

But for V2.0, we changed a lot, such as from efi to grub, and not test CL, it's best to set up ubuntu from V2.0.

SIMING29, 2020-08-20:
@shenfang2019 SOS can not output though hdmi after change the cmdline. Run the install_win.sh though ssh login, and same with the previous result.
Should I set something for using WaaG in BIOS? or update the BIOS version?   now  Intel Visual BIOS version 2.2.23

$cat /proc/cmdline 
initrd=\EFI\org.clearlinux\freestanding-00-intel-ucode.cpio initrd=\EFI\org.clearlinux\freestanding-i915-firmware.cpio.xz console=tty0 console=ttyS0 root=/dev/sda3 rw rootwait ignore_loglevel no_timer_check consoleblank=0 i915.nuclear_pageflip=1  i915.avail_planes_per_pipe=0x010101 i915.domain_plane_owners=0x011100001111 i915.enable_gvt=1 i915.enable_conformance_check=0 i915.enable_guc=0 hvlog=2M@0x1FE00000

shenfang2019, 2020-08-20:
@SIMING29 How about your OVMF? Do you use the ovmf of V1.6?

SIMING29, 2020-08-21:
@shenfang2019 I did not changed  it after the system installed. the /usr/share/acrn/bios/changlog_ovmf.txt show the version is release V1.6

BTW:
I builded the V2.0 with ubuntu with another disk, now I can run clr-linux and xenomai UOS, but still can not run windows UOS.
the log show the GVT init failed, and there is no gvt directory in /sys/kernel. I builded the system according to the latest doc, did I miss something?  thanks.
[install.log](https://github.com/projectacrn/acrn-hypervisor/files/5106259/install.log)

```
pci init pci-gvt
GVT: open /sys/kernel/gvt/control/create_gvt_instance failed
                                                            GVT: init failed
                                                                            pci pci-gvt init failed
                                                                                                   mngr_client_new: Failed to accept from fd 34, err: Invalid argument
                                                                                                                                                                      Stop listening 34...
                  Stop polling 34...
                                    No correct pm notify channel given
                                                                      Unable to init vdev (2)
```

shenfang2019, 2020-08-21:
@xiaoguangwu, could you help to check gvt issue?

shenfang2019, 2020-08-24:
@SIMING29 please check your OVMF that allocates in source codes: acrn-hypervisor/devicemodel/bios/OVMF.fd, then copy to your test board:/usr/share/acrn/bios. For v2.0, md5sum of OVMF.f is d58d0a3355c87efbedbfc6bc9f64e7e3f.

xiaoguangwu, 2020-08-24:
@SIMING29  please enter acrn-kernel directory and run command "git log -100 --oneline" and copy the output in the comments, thanks!

SIMING29, 2020-08-24:
@shenfang2019  OVMF.fd is same like the file in source codes.
@xiaoguangwu  
 ```
$git log -100 --oneline 
a414b9241ebd (HEAD -> release2.0, tag: v2.0, tag: acrn-2020w25.7-180000p, origin/release_2.0) Add PCH slow clock wa for EHL GPU flr
8885519db693 (tag: acrn-2020w25.1-180000p) kernel: enable UIO and UIO_PCI_GENERIC configs
80ea97f638a5 (tag: acrn-2020w24.5-180000p, tag: acrn-2020w23.6-180000p, tag: acrn-2020w23.5-180000p) vhm: use 2MB compound pages for ACRN mempool
619bd13b1fba vhm: add vm_configs array to get_platform_info hypercall
2b3a3e2efe65 config: add SOS kernel config required by OpenStack
7b46601d53b5 kernel: enable config to load i915 firmware
c54a5850037a Import the fimrware
a48cf107b223 kernel: add parameter description in mpt.h
6940410f3cec kernel: add config for common intel network card
81eec43d4626 kernel: add kernel config
996c779dc027 Merge branch '5.4/yocto' of https://github.com/intel/linux-intel-lts into yocto_5.4
bda68cfc97bb tcc: parse PTCT table and record pesudo sram ranges
a83e0909a8eb tcc: this is kernel driver to interface to TCC PTCM pesudo SRAM
7882e18f15b8 platform/x86: Add Slim Bootloader firmware update signaling driver
065d5b24be91 Merge tag 'v5.4.43' into lts2019/yocto
e0d81ce76004 Linux 5.4.43
b5100186021a sched/fair: Fix enqueue_task_fair() warning some more
8b13f5657fa8 sched/fair: Fix reordering of enqueue/dequeue_task_fair()
a2ad232aa6a9 sched/fair: Reorder enqueue/dequeue_task_fair path

```

shenfang2019, 2020-08-25:
@SIMING29  1: Please use our cmdline: "console=tty0 console=ttyS0 root=/dev/nvme0n1p3 rw rootwait ignore_loglevel no_timer_check consoleblank=0 i915.nuclear_pageflip=1 i915.avail_planes_per_pipe=0x010101 i915.domain_plane_owners=0x011100001111 i915.enable_gvt=1 i915.enable_conformance_check=0 i915.enable_guc=0 hvlog=2M@0x1FE00000"
2: And make sure your screen support 1920*1080.

shenfang2019, 2020-09-02:
@SIMING29 is your issue fixed？V2.2 is coming soon，gvt will be better.

clefrks, 2020-09-15:
@shenfang2019 in V2.2, what improvement will be done on gvt?

fuzhongl, 2020-11-02:
GVT-g GOP feature is enabled for v2.2.
Please have a try if the issue is fixed on your side.

zhangshuai75, 2021-04-23:
use V2.3 with passthrough gpu works well

closed 2021-04-23 by zhangshuai75