projectacrn/acrn-hypervisor #5197: Remove the de-privileged boot mode support

NanlinXie, 2020-08-20:
De-privileged boot mode was used with Clear Linux SOS, it de-privilege UEFI execution environment from native to Service VM non-root mode, but must ensure any hardware resource touched by UEFI FW before de-privilege must keep the same configuration after boot to service VM. It brings lots of scalability issues when porting ACRN to new platforms. 
Thus ACRN plans to remove the de-privileged mode support and turn to direct boot mode.

ZideChen0, 2020-08-28:
[External_System_ID] ACRN-6487

gvancuts, 2020-10-30:
@NanlinXie , how do we know when this is done (both from a code and documentation perspective)?

Just for the record, here is one paragraph that should update: https://projectacrn.github.io/latest/developer-guides/hld/hld-security.html#verified-boot-sequence-with-uefi

closed 2021-03-15 by fuzhongl