projectacrn/acrn-hypervisor #5215: [UPsquared N3350 board] Solution to Bootloader issue

starmachine24, 2020-08-24:
Hi

We managed to figure out a solution to a problem related to the UEFI not saving the bootorder on the UPsquared board. We found this solution in the BIOS setting for our board which is a N3350, so we don't know if this works on the other UPsquared boards. 

Anyway, the solution was to go into the BIOS for the system and then go over to the Boot section. Then go into the **UEFI harddisk drive BBS priorities** and switch the top most alternative with the Hypervisor boot option. This has worked for us on version acrn-2019w35.1-140000p and 2.0.

gvancuts, 2020-08-24:
I will try this on my board soon as it has affected me too for a while, thanks for sharing the tip!! :+1:

starmachine24, 2020-08-25:
no problem

closed 2020-09-10 by starmachine24