projectacrn/acrn-hypervisor #5233: Boot Acrn failed on Dell-OptiPlex 5040 with Intel i5-6500T

wadext, 2020-08-26:
**Describe the bug**
we have tryed to install sos (ubuntu-18.04) and uos(ubuntu-18.04) on a dell optiplex( i5-6500T 16GB  nvme + sata), generate board and scenario xml using config tools , build acrn hypervisor and kernel , update the efiboot menu and grub config, but when we reboot to the acrn hypervisor and kernel for sos , got stuck neither with uefi nor grub

**Platform**
Dell Optiplex 5040
Intel i5-6500T
4*4GB DDR3
256 GB Nvme SSD
500 GB SATA

**Scenario**
Industry

**Additional context**

- native Linux boot kernel message log
```
[    0.000000] microcode: microcode updated early to revision 0xd6, date = 2019-10-03
[    0.000000] Linux version 5.4.43-PKT-200203T060100Z (root@acrn-OptiPlex-5040) (gcc version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04)) #2 SMP PREEMPT Thu Aug 20 10:51:30 CST 2020
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.4.43-PKT-200203T060100Z root=UUID=8531accc-08cd-46d9-9010-5eced4c84ac1 ro
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000059000-0x000000000009efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009f000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000d3449fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000d344a000-0x00000000d344afff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000d344b000-0x00000000d3474fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000d3475000-0x00000000d34cbfff] usable
[    0.000000] BIOS-e820: [mem 0x00000000d34cc000-0x00000000d3cccfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000d3ccd000-0x00000000db4d3fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000db4d4000-0x00000000db6f7fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000db6f8000-0x00000000db730fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000db731000-0x00000000dbef6fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000dbef7000-0x00000000dc41cfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000dc41d000-0x00000000dc4fefff] type 20
[    0.000000] BIOS-e820: [mem 0x00000000dc4ff000-0x00000000dc4fffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000dc500000-0x00000000dfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000041dffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.40 by American Megatrends
[    0.000000] efi:  ACPI=0xdb703000  ACPI 2.0=0xdb703000  SMBIOS=0xdc2e3000  ESRT=0xdc26e018  SMBIOS 3.0=0xdc2e2000  MPS=0xfcba0  MEMATTR=0xd9429018 
[    0.000000] SMBIOS 3.0.0 present.
[    0.000000] DMI: Dell Inc. OptiPlex 5040/0N5G27, BIOS 1.14.5 07/19/2019
[    0.000000] tsc: Detected 2500.000 MHz processor
[    0.002144] tsc: Detected 2499.952 MHz TSC
[    0.002144] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.002146] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.002154] last_pfn = 0x41e000 max_arch_pfn = 0x400000000
[    0.002158] MTRR default type: write-back
[    0.002159] MTRR fixed ranges enabled:
[    0.002160]   00000-9FFFF write-back
[    0.002161]   A0000-BFFFF uncachable
[    0.002162]   C0000-FFFFF write-protect
[    0.002162] MTRR variable ranges enabled:
[    0.002164]   0 base 00E0000000 mask 7FE0000000 uncachable
[... 755 lines cut ...]
```
- ACRN serial port log (when issues happens)
```
Welcome to minicom 2.7.1

OPTIONS: 
Compiled on Oct  6 2019, 23:16:03.
Port /dev/tty.usbserial-142420, 09:53:36

Press Meta-Z for help on special keys

Detect bootloader: ACRN UEFI loader
ACRN Hypervisor
calibrate_tsc, tsc_khz=2500000
[17818300us][cpu=0][(null)][sev=2][seq=2]:HV version 2.0-2020-08-20 14:37:29-978f8990-dirty DBG (daily tag:acrn-2020w25.7-180000p) build by root, start time 17810488us
[17833681us][cpu=0][(null)][sev=2][seq=3]:API version 1.0
[17839103us][cpu=0][(null)][sev=2][seq=4]:Detect processor: Intel(R) Core(TM) i5-6500T CPU @ 2.50GHz
[17869208us][cpu=0][(null)][sev=3][seq=6]:init_pdev, 5a:11.0 unsupported headed type: 0x28

[17877858us][cpu=0][(null)][sev=3][seq=7]:init_pdev, 5a:13.0 unsupported headed type: 0x28

```

Thanks!
xuteng

fuzhongl, 2020-08-31:
Fixed with hotfix patch; will merge the patch later.

tunicashashi, 2020-09-04:
Is this patch merged?

fuzhongl, 2020-09-04:
Not yet. Patch is under review.

fuzhongl, 2020-09-08:
[External_System_ID]ACRN-6685

fuzhongl, 2020-09-21:
@wadext PR was merged into ACRN v2.2 branch, please help to try it.
Any issue, please reopen this ticket.
Thanks!

closed 2020-09-21 by fuzhongl