projectacrn/acrn-hypervisor #5236: Documentation for the kernel parameters (i915)

gvancuts, 2020-08-26:
Some obsolete parameters have been recently removed in our code (see #5143). However, the PR is incomplete and more is needed to clean the rest of our code and documentation:

1. The following pages still include references to the parameters that have been removed
* https://projectacrn.github.io/latest/faq.html#how-to-modify-the-default-display-output-for-a-user-vm
* https://projectacrn.github.io/latest/tutorials/using_ubuntu_as_sos.html#install-the-service-vm-kernel
* https://projectacrn.github.io/latest/user-guides/kernel-parameters.html#i915-domain-scaler-owner

2. Some of the configs for the `apl-mrb` and `apl-up2` are still using these

3. Other parameters still listed in our documentation are obsolete and should be removed (in both code and doc)
```
i915.gvt_workload_priority
i915.domain_scaler_owner
i915.enable_initial_modeset
```

gvancuts, 2021-02-23:
We have made some progress on this ;-) (see, amongst others, #5759 #5703)

This item is still open though:

>     1. Some of the configs for the `apl-mrb` and `apl-up2` are still using these

wenlingz, 2021-02-24:
@gvancuts  please add JIRA ID, thanks

gvancuts, 2021-02-24:
[External_System_ID] ACRN-6988

gvancuts, 2021-02-24:
> @gvancuts please add JIRA ID, thanks

Done. Thanks.

gvancuts, 2021-03-10:
I believe we have cleaned all these through the various PRs... @dbkinder @fitchbe , let me know if you see something that hasn't been addressed yet. Closing this for now.

closed 2021-03-10 by gvancuts