projectacrn/acrn-hypervisor #5265: Doc: Need update FAQ document for some tips.

NanlinXie, 2020-09-02:
Need update FAQ document for some tips.

dbkinder, 2021-02-10:
@NanlinXie @fuzhongl @gvancuts Where can we get material to update this FAQ?  Alternatively, we should delete it (or just empty out what's there) since it's not very useful or even wrong (e.g, no more ``make menuconfig``).

gvancuts, 2021-02-11:
I would remove this section entirely for now. Half of it is plain outdated and the other half is of limited value in my opinion, the most relevant entry in there in for me is the explanation as to why many things are static in ACRN. So if we want to keep the FAQ, let's use that entry and say a little more than that. It's not just memory size, but also the fact many things are hard-coded such as number of VMs that can be run, etc.

Other good candidates to be put in there would be questions about governance, membership once these things are established.

dbkinder, 2022-01-26:
@NanlinXie @gvancuts If we're not going to keep an FAQ doc updated, I propose we should delete it, right?

closed 2022-02-16 by amyreye