projectacrn/acrn-hypervisor #5282: [acrn-configuration-tool]failure to create a new launch script

szhen11, 2020-09-07:
Environment
BOARD: EHL
SOS IMAGE:http://sit-image.sh.intel.com/acrn_daily_build/YOCTO-2020ww36.4-1/yocto-ehl-crb-b-uefi-industry/build.txt

Steps

git clone https://github.com/projectacrn/acrn-hypervisor.git
cd acrn-hypervisor/misc/acrn-config/config_app;python3 app.py
1. Open Config_app UI by following guide, select board "ehl-crb-b.xml" from select box of "Board Info"; If there is no such board info, click "Import Board Info" to import the board from acrn-hypervisor/misc/xmls/board-xmls/
2. Click menu "Scenario Setting" -> "Load a default sceanrio" -> Select "industry" -> Click "OK"
3. Click menu "Launch Setting" -> " Create a new launch setting " a dialog will pop up to set the new launch name, click "OK"

Expected result
Web work normal.

Actual result
Web crash,
log:
127.0.0.1 - - [07/Sep/2020 13:32:01] "POST /create_setting HTTP/1.1" 200 -
[2020-09-07 13:32:03,579] ERROR in app: Exception on /launch/1599456711636 [GET]
Traceback (most recent call last):
File "/root/RUI/workspace/weekly/cwpqa-acrn-robot/venv/lib/python3.6/site-packages/flask/app.py", line 2446, in wsgi_app
response = self.full_dispatch_request()
File "/root/RUI/workspace/weekly/cwpqa-acrn-robot/venv/lib/python3.6/site-packages/flask/app.py", line 1951, in full_dispatch_request
rv = self.handle_user_exception(e)
File "/root/RUI/workspace/weekly/cwpqa-acrn-robot/venv/lib/python3.6/site-packages/flask/app.py", line 1820, in handle_user_exception
reraise(exc_type, exc_value, tb)
File "/root/RUI/workspace/weekly/cwpqa-acrn-robot/venv/lib/python3.6/site-packages/flask/_compat.py", line 39, in reraise
raise value
File "/root/RUI/workspace/weekly/cwpqa-acrn-robot/venv/lib/python3.6/site-packages/flask/app.py", line 1949, in full_dispatch_request
rv = self.dispatch_request()
File "/root/RUI/workspace/weekly/cwpqa-acrn-robot/venv/lib/python3.6/site-packages/flask/app.py", line 1935, in dispatch_request
return self.view_functions[rule.endpoint](**req.view_args)
File "/root/tdoux/workspace/weekly/cwpqa-acrn-robot/acrn-hypervisor_test/misc/acrn-config/config_app/views.py", line 135, in launch
os.path.join(os.path.dirname(os.path.abspath(_file_)), 'res', board_info + '.xml'), scenario_file)
File "/root/tdoux/workspace/weekly/cwpqa-acrn-robot/acrn-hypervisor_test/misc/acrn-config/config_app/../launch_config/launch_cfg_gen.py", line 55, in get_launch_item_values
launch_cfg_lib.set_shm_regions(launch_item_values, scenario_info)
File "/root/tdoux/workspace/weekly/cwpqa-acrn-robot/acrn-hypervisor_test/misc/acrn-config/config_app/../library/launch_cfg_lib.py", line 571, in set_shm_regions
raw_shmem_regions = common.get_hv_item_tag(scenario_info, "FEATURES", "IVSHMEM", "IVSHMEM_REGION")
File "/root/tdoux/workspace/weekly/cwpqa-acrn-robot/acrn-hypervisor_test/misc/acrn-config/config_app/../library/common.py", line 421, in get_hv_item_tag
root = get_config_root(config_file)
File "/root/tdoux/workspace/weekly/cwpqa-acrn-robot/acrn-hypervisor_test/misc/acrn-config/config_app/../library/common.py", line 233, in get_config_root
tree = ET.parse(config_file)
File "/usr/lib/python3.6/xml/etree/ElementTree.py", line 1196, in parse
tree.parse(source, parser)
File "/usr/lib/python3.6/xml/etree/ElementTree.py", line 586, in parse
source = open(source, "rb")
TypeError: expected str, bytes or os.PathLike object, not NoneType

szhen11, 2020-09-07:
[External_System_ID]ACRN-6681

gvancuts, 2021-01-15:
The patch was merged, can this issue be closed?

szhen11, 2021-01-18:
tested and closed it.

closed 2021-01-18 by szhen11