projectacrn/acrn-hypervisor #5327: Doc: IASL version need to be aligned and clarified in document

NanlinXie, 2020-09-16:
We found that iasl version is referred three times in our doc, these should be aligned and clarified.

1.	https://projectacrn.github.io/latest/getting-started/rt_industry_ubuntu.html?highlight=iasl#install-iasl-in-ubuntu-for-user-vm-launch
Install IASL in Ubuntu for User VM launch
ACRN uses iasl to parse User VM ACPI information. The original iasl in Ubuntu 18.04 is too old to match with acrn-dm; update it using the following steps:
$ sudo -E apt-get install iasl
$ cd /home/acrn/work
$ wget https://acpica.org/sites/acpica/files/acpica-unix-20191018.tar.gz
$ tar zxvf acpica-unix-20191018.tar.gz
$ cd acpica-unix-20191018
$ make clean && make iasl
$ sudo cp ./generate/unix/bin/iasl /usr/sbin/

2.	https://projectacrn.github.io/latest/tutorials/acrn_on_qemu.html?highlight=iasl#bring-up-user-vm-l2-guest
Install latest IASL tool and copy the binary to /usr/sbin/iasl. For this setup, used IASL 20200326 version but anything after 20190215 should be good.

3.	https://projectacrn.github.io/latest/tutorials/running_deb_as_serv_vm.html?highlight=iasl#run-debian-as-the-service-vm
•  Update to the latest iASL (required by the ACRN Device Model):
$ sudo apt update
$ sudo apt install m4 bison flex zlib1g-dev
$ cd ~
$ wget https://acpica.org/sites/acpica/files/acpica-unix-20190816.tar.gz
$ tar zxvf acpica-unix-20190816.tar.gz
$ cd acpica-unix-20190816
$ make clean && make iasl
$ sudo cp ./generate/unix/bin/iasl /usr/sbin/

And pre-launched VM also need iasl version check to build the offline acpi table, currently we verified 20190703+ are OK,
But I think we should have these information aligned in one place in doc.

dnlzm, 2020-10-06:
1. iasl is not needed for 2.1 but for 2.2 and master. Would be nice to include in make.
after manually downloading, make iasl  gives me an error bison not found.
After manually installing it I get
.../acpica-unix-20191018/generate/unix/iasl/obj/aslcompiler.y:1.1: error: syntax error, unexpected end of file
mv: cannot stat 'obj/AslCompiler.WOfT0I/y.tab.h': No such file or directory
Makefile:328: recipe for target 'obj/aslcompiler.y.h' failed
make[1]: *** [obj/aslcompiler.y.h] Error 1

fresh install of Linux 5.4.0-48-generic #52~18.04.1-Ubuntu SMP

gvancuts, 2020-10-07:
> 1. iasl is not needed for 2.1 but for 2.2 and master. Would be nice to include in make.

To be more accurate, `iasl` is a runtime dependency for `acrn-dm`, even in previous versions. It is also used in version >2.2 to pre-compile ACPI tables when build ACRN. Ideally we'd find some way to make this clear in our documentation.

>    after manually downloading, make iasl  gives me an error bison not found.

Did you install dependencies (see `sudo apt install m4 bison flex zlib1g-dev` from our documentation)?

> After manually installing it I get
>    .../acpica-unix-20191018/generate/unix/iasl/obj/aslcompiler.y:1.1: error: syntax error, unexpected end of file
>    mv: cannot stat [...]

Can you check which version is picked up? `which iasl` and `iasl -v`
Installing `iasl` from the Ubuntu repository puts it under `/usr/bin`, the manual installation we recommend in our documentation puts it under `/usr/sbin` so both could potentially be installed on the system.

dbkinder, 2021-02-10:
@fuzhongl @NanlinXie Will this be updated for v2.4?

gvancuts, 2021-02-16:
For what it's worth, I think we should. I'm getting a feeling that unless we force this once for all, it will never happen.

I'm using ACPICA version 20210105 lately and have not run into any problem yet.

I'd like to propose that we use that, update all documents and **if** we cannot verify them all, then treat potential regressions in some of them as bugs. How does that sound?

gvancuts, 2021-03-23:
For the record, see #5856 and #5849

gvancuts, 2021-05-19:
> For the record, see #5856 and #5849

I believe these resolve this issue, closing it.

closed 2021-05-19 by gvancuts