projectacrn/acrn-hypervisor #5385: doc: fix doxygen comments and configuration

dbkinder, 2020-10-06:
**Describe the bug**
An update to doxygen (1.8.17) is more picky about doxygen comment errors than the version currently used (1.8.13).  Let's fix them now to prepare for the new version.

Here are the errors shown when running doxygen 1.8.17) via ``make html``

```
warning: Tag 'PERL_PATH' at line 2172 of file '-' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
warning: Tag 'MSCGEN_PATH' at line 2194 of file '-' has become obsolete.
         To avoid this warning please remove this line from your configuration file or upgrade it using "doxygen -u"
/home/dbkinder/projectacrn/acrn-hypervisor/hypervisor/include/arch/x86/guest/ept.h:163: warning: unbalanced grouping commands
/home/dbkinder/projectacrn/acrn-hypervisor/hypervisor/include/common/ptdev.h:265: warning: return value 'success' of ptirq_activate_entry has multiple documentation sections
/home/dbkinder/projectacrn/acrn-hypervisor/hypervisor/include/common/ptdev.h:265: warning: return value 'success' of ptirq_activate_entry has multiple documentation sections
/home/dbkinder/projectacrn/acrn-hypervisor/hypervisor/include/common/ptdev.h:265: warning: return value 'success' of ptirq_activate_entry has multiple documentation sections
/home/dbkinder/projectacrn/acrn-hypervisor/hypervisor/include/common/ptdev.h:265: warning: return value 'success' of ptirq_activate_entry has multiple documentation sections
/home/dbkinder/projectacrn/acrn-hypervisor/hypervisor/include/common/ptdev.h:265: warning: return value 'success' of ptirq_activate_entry has multiple documentation sections
/home/dbkinder/projectacrn/acrn-hypervisor/hypervisor/include/common/ptdev.h:265: warning: return value 'success' of ptirq_activate_entry has multiple documentation sections
```

wenlingz, 2020-10-26:
[External_System_ID]ACRN-6774

dbkinder, 2021-01-30:
Fixed

closed 2021-01-30 by dbkinder